sudar/Arduino-Makefile #421: 'printDigits' was not declared in this scope

joshuacox, 2016-03-24:
What library is 'printDigits' in?

I added Time to:

```
ARDUINO_LIBS = Wire SoftwareSerial Time DS1307RTC
```

and my compile ends in this:

```
error: 'printDigits' was not declared in this scope
   printDigits(minute());
```

perhaps I need to include another library like #93 ?

sudar, 2016-03-24:
A quick web search says that this function might be available in Time library. But there are various versions of the library floating around. So I am not sure.

If a library depends on another library, then you have to include both.

sej7278, 2016-03-24:
"X was not declared in this scope" usually means you've tried to call a function before declaring it, unlike the IDE which lets you get away with this.

if its your own function, make sure to define it before setup() and loop(), if its a library, then be sure to include it at the top of your sketch e.g.

```
void printDigits(byte digits)
{
    blah.....
}

void setup()
{
    blah....
}

void loop()
{
    printDigits(6);
    blah....
}
```

joshuacox, 2016-03-24:
that's the thing, notice I put the Time library in:

```
ARDUINO_LIBS = Wire SoftwareSerial Time DS1307RTC
```

The Time library is 1.5.0 according the arduino application (which downloaded it in the first place).

```
ls ~/sketchbook/libraries/Time
DateStrings.cpp  examples  keywords.txt  library.json  library.properties  Readme.txt  Time.cpp  Time.h  TimeLib.h
```

  I'm using arch linux here, so the arduino app is 1.6.8 :

```
3 aur/arduino 1:1.6.8-1 [installed] (471) (17.37)
```

here's the full output:
```
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [AUTODETECTED]       ARDUINO_DIR = /usr/share/arduino 
- [COMPUTED]           ARDMK_DIR = /usr/share/arduino (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 168 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/thoth/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/thoth/sketchbook (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /usr/share/arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/share/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/thoth/sketchbook/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [DEFAULT]            BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 9600  (in sketch)
- [DEFAULT]            OPTIMIZAhttps://github.com/joshuacox/DHTavr2serialTION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyUSB0 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 DS1307RTC
- [USER]                 Time
- [PLATFORM]             SoftwareSerial
- [PLATFORM]             Wire
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-uno
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=168 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/standard    -I/usr/share/arduino/hardware/arduino/avr/libraries/SoftwareSerial/src   -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/src   -I/home/thoth/sketchbook/libraries/DS1307RTC   -I/home/thoth/sketchbook/libraries/Time -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  ardushipper.ino -o build-uno/ardushipper.o
ardushipper.ino: In function 'void digitalClockDisplay()':
ardushipper.ino:241:23: error: 'printDigits' was not declared in this scope
   printDigits(minute());
                       ^
/usr/share/arduino/Arduino.mk:1225: recipe for target 'build-uno/ardushipper.o' failed
make: *** [build-uno/ardushipper.o] Error 1

```

repo here:
https://github.com/joshuacox/DHTavr2serial

joshuacox, 2016-03-24:
hrm, a grep inside my `~/sketchbook/libraries` shows printDigits in the examples, but not in Time.h or any other library I have, although it is also referenced in the TimeAlarms docs, but again not in any .h or .c, .cpp, etc.

```
grep -ir printDigits *
Time/examples/TimeRTCSet/TimeRTCSet.ino:  printDigits(minute());
Time/examples/TimeRTCSet/TimeRTCSet.ino:  printDigits(second());
Time/examples/TimeRTCSet/TimeRTCSet.ino:void printDigits(int digits){
Time/examples/TimeRTC/TimeRTC.ino:  printDigits(minute());
Time/examples/TimeRTC/TimeRTC.ino:  printDigits(second());
Time/examples/TimeRTC/TimeRTC.ino:void printDigits(int digits){
Time/examples/TimeSerialDateStrings/TimeSerialDateStrings.ino:  printDigits(minute());
Time/examples/TimeSerialDateStrings/TimeSerialDateStrings.ino:  printDigits(second());
Time/examples/TimeSerialDateStrings/TimeSerialDateStrings.ino:void printDigits(int digits) {
Time/examples/TimeGPS/TimeGPS.ino:  printDigits(minute());
Time/examples/TimeGPS/TimeGPS.ino:  printDigits(second());
Time/examples/TimeGPS/TimeGPS.ino:void printDigits(int digits) {
Time/examples/TimeTeensy3/TimeTeensy3.ino:  printDigits(minute());
Time/examples/TimeTeensy3/TimeTeensy3.ino:  printDigits(second());
Time/examples/TimeTeensy3/TimeTeensy3.ino:void printDigits(int digits){
Time/examples/TimeNTP_ESP8266WiFi/TimeNTP_ESP8266WiFi.ino:  printDigits(minute());
Time/examples/TimeNTP_ESP8266WiFi/TimeNTP_ESP8266WiFi.ino:  printDigits(second());
Time/examples/TimeNTP_ESP8266WiFi/TimeNTP_ESP8266WiFi.ino:void printDigits(int digits){
Time/examples/TimeSerial/TimeSerial.ino:  printDigits(minute());
Time/examples/TimeSerial/TimeSerial.ino:  printDigits(second());
Time/examples/TimeSerial/TimeSerial.ino:void printDigits(int digits){
Time/examples/TimeArduinoDue/TimeArduinoDue.ino:  printDigits(minute());
Time/examples/TimeArduinoDue/TimeArduinoDue.ino:  printDigits(second());
Time/examples/TimeArduinoDue/TimeArduinoDue.ino:void printDigits(int digits){
Time/examples/TimeNTP/TimeNTP.ino:  printDigits(minute());
Time/examples/TimeNTP/TimeNTP.ino:  printDigits(second());
Time/examples/TimeNTP/TimeNTP.ino:void printDigits(int digits){
Time/examples/TimeRTCLog/TimeRTCLog.ino:  printDigits(minute(t));
Time/examples/TimeRTCLog/TimeRTCLog.ino:  printDigits(second(t));
Time/examples/TimeRTCLog/TimeRTCLog.ino:void printDigits(int digits){
TimeAlarms/readme.txt:  printDigits(minute());
TimeAlarms/readme.txt:  printDigits(second());
TimeAlarms/readme.txt:void printDigits(int digits)
TimeAlarms/examples/TimeAlarmExample/TimeAlarmExample.pde:  printDigits(minute());
TimeAlarms/examples/TimeAlarmExample/TimeAlarmExample.pde:  printDigits(second());
TimeAlarms/examples/TimeAlarmExample/TimeAlarmExample.pde:void printDigits(int digits)
```

sej7278, 2016-03-24:
from the looks of your sketch, its just as i said - digitalClockDisplay() is calling printDigits() at line 241 before its been declared at line 252, its nothing to do with the Time library

its basic c++, but the ide lets you get away with it as it munges the methods together, the Makefile doesn't so forces you to write correct c++

move lines 252-257 to somewhere before line 238 and i bet it'll fix your problem

joshuacox, 2016-03-24:
that was it, thanks for that one.  Been working with dynamic languages like perl and python too long, I forgot all about that.

closed 2016-03-24 by joshuacox

sej7278, 2016-03-24:
hah, i'm mr. dynamic, been programming perl for 20 years, so strong typing and method declarations were news to me, which is why this solution stuck in my head ;-)

joshuacox, 2016-03-24:
I'm still laughing that I was grepping around for printDigits everywhere but my own code.  And there it is hanging out where I copied and pasted it over a year ago.  Doh!