sudar/Arduino-Makefile #422: newbie setup help

mark-hahn, 2016-04-02:
In the setup it says:

"Instead of copying the makefile to every sketch folder, you can just place the downloaded makefile separately in a common location and then create a small child makefile for every sketch".

- I installed and package and I cloned the repo and I can't find a makefile anywhere.  Where it it?

- What goes in a "small child makefile"?

mark-hahn, 2016-04-02:
I managed to get past these two problems.  I figured out `Arduino-Makefile/Arduino.mk` was the makefile.  Why wasn't it named `makefile`?  Also, I didn't know how to make the small makefile but I used a symlink and it seems to work.

**New Problem:**

My make is unable to find `Servo.h`.  My `ARDUINO_LIB_PATH` is at `/root/dev/arduino/arduino-1.6.8/libraries`.  The Servo library  is at `/root/dev/arduino/arduino-1.6.8/libraries/Servo`.  Servo.h is at `/root/dev/arduino/arduino-1.6.8/libraries/Servo/src/Servo.h`.  Why can't it find it?

```
roo(~/dev/arduino/robot). make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /root/dev/arduino/arduino-1.6.8 
- [USER]               ARDMK_DIR = /root/dev/arduino/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 168 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [DEFAULT]            ARDUINO_SKETCHBOOK =  
- [USER]               AVR_TOOLS_DIR = /usr 
- [COMPUTED]           ARDUINO_LIB_PATH = /root/dev/arduino/arduino-1.6.8/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /root/dev/arduino/arduino-1.6.8/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /root/dev/arduino/arduino-1.6.8/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /root/dev/arduino/arduino-1.6.8/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [USER]               USER_LIB_PATH = /root/dev/arduino/libraries 
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [DEFAULT]            BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /root/dev/arduino/arduino-1.6.8/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 115200  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [COMPUTED]           DEVICE_PATH = ttyACM0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 Adafruit_CC3000
- [COMPUTED]           BOOTLOADER_PARENT = /root/dev/arduino/arduino-1.6.8/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.5.3 (avr-gcc)
-------------------------
mkdir -p build-uno
/usr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=168 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/root/dev/arduino/arduino-1.6.8/hardware/arduino/avr/cores/arduino -I/root/dev/arduino/arduino-1.6.8/hardware/arduino/avr/variants/standard     -I/root/dev/arduino/libraries/Adafruit_CC3000 -I/root/dev/arduino/libraries/Adafruit_CC3000/utility -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  robot.ino -o build-uno/robot.ino.o
robot.ino:1:19: fatal error: Servo.h: No such file or directory
compilation terminated.
make: *** [build-uno/robot.ino.o] Error 1
```

mark-hahn, 2016-04-02:
Everything is OK.  I was missing the quotes in `export ARDUINO_LIBS="Adafruit_CC3000 Servo LiquidCrystal SPI Wire"`.

I hope someone else having problems as stupid as mine will be helped when they google this.

closed 2016-04-02 by mark-hahn

sudar, 2016-04-03:
Glad to know that you were able to figure out the solution to all your problems.