sudar/Arduino-Makefile #423: Windows/cygwin compile problem

bertjanstroop, 2016-04-11:
compiling sourcecode with make in cygwin fails but works if i try it again.
if i do a make clean and then make the build fails, but when i again do make after the first make the build succeeds

sudar, 2016-04-12:
Support for windows/cygwin is sketchy at best and is not fully supported yet.

Can you give me the full output of your makefile?

bertjanstroop, 2016-04-12:
can't give full output because i can't disclose class names

First compile:
`
collect2.exe: error: ld returned 5 exit status
D:/Dropbox/software/programs/arduinomakefile/Arduino.mk:1396: recipe for target                                                                                                                                   'build-mega-atmega2560/trunk.elf' failed
make: *** [build-mega-atmega2560/trunk.elf] Error 1
`

Second compile:

`mkdir -p build-mega-atmega2560
C:/Arduino/hardware/tools/avr/bin/avr-objcopy -j .eeprom --set-section-flags=.ee                                                                                                                                  prom='alloc,load' \
        --change-section-lma .eeprom=0 -O ihex build-mega-atmega2560/trunk.elf b                                                                                                                                  uild-mega-atmega2560/trunk.eep
C:\Arduino\hardware\tools\avr\bin\avr-objcopy.exe: --change-section-lma .eeprom=                                                                                                                                  0x00000000 never used
C:/Arduino/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom build-mega-atme                                                                                                                                  ga2560/trunk.elf build-mega-atmega2560/trunk.hex

C:/Arduino/hardware/tools/avr/bin/avr-size --mcu=atmega2560 -C --format=avr buil                                                                                                                                  d-mega-atmega2560/trunk.elf
AVR Memory Usage
----------------
Device: atmega2560

Program:   25540 bytes (9.7% Full)
(.text + .data + .bootloader)

Data:       4832 bytes (59.0% Full)
(.data + .bss + .noinit)
`

DancingQuanta, 2016-06-01:
Hi I can comfirm this happening on Cygwin using Blink example with board tag Uno without any connected Arduinos (I did this way to test this awesome Makefile project on Cygwin).

I ran the make command three times consecutively and recorded the output using `script`.

The first make command fails with

```
make: *** [../../Arduino.mk:1441: build-uno/Blink.elf] Error 1
```

due to

```
collect2.exe: error: ld returned 5 exit status
```

The second succeeds and the third stopped at `mkdir -p build-uno` (this last invocation gives correct behaviour as `make clean` removes this directory).

The line in Arduino.mk where the error occured is 1441 which is a command to build elf file

```
$(TARGET_ELF): 	$(LOCAL_OBJS) $(CORE_LIB) $(OTHER_OBJS)
		$(CC) $(LDFLAGS) -o $@ $(LOCAL_OBJS) $(CORE_LIB) $(OTHER_OBJS) -lc -lm $(LINKER_SCRIPTS)
```

On my output the actual command is

```
../../../../../../Arduino/hardware/tools/avr/bin/avr-gcc -mmcu=atmega328p -Wl,--gc-sections -Os -o build-uno/Blink.elf build-uno/Blink.ino.o build-uno/libcore.a  -lc -lm
```

The `LINKER_SCRIPT` is empty and I do not know if this is correct behaviour (I found that chipKIT.mk manipulates this variable and so not used for Arduino Uno).

There appears to be some warnings along the compilation on the first try, will investigate them for any connections to elf fail but first I put my full output below for first and second try for now.

```
$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = WINDOWS 
- [USER]               ARDUINO_DIR = ../../../../../../Arduino 
- [COMPUTED]           ARDMK_DIR = /cygdrive/c/Users/andre/myarduino/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 169 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            VENDOR = arduino 
- [DEFAULT]            ARDUINO_SKETCHBOOK =  
- [BUNDLED]            AVR_TOOLS_DIR = ../../../../../../Arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = ../../../../../../Arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = ../../../../../../Arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = ../../../../../../Arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = ../../../../../../Arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = ../../../../../../Arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [COMPUTED]           DEVICE_PATH = /dev/ttyS2 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = ../../../../../../Arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
[... 72 lines cut ...]
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-uno
../../../../../../Arduino/hardware/tools/avr/bin/avr-objcopy -j .eeprom --set-section-flags=.eeprom='alloc,load' \
	--change-section-lma .eeprom=0 -O ihex build-uno/Blink.elf build-uno/Blink.eep
C:\Arduino\hardware\tools\avr\bin\avr-objcopy.exe: --change-section-lma .eeprom=0x00000000 never used

../../../../../../Arduino/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom build-uno/Blink.elf build-uno/Blink.hex

../../../../../../Arduino/hardware/tools/avr/bin/avr-size --mcu=atmega328p -C --format=avr build-uno/Blink.elf
AVR Memory Usage

----------------

Device: atmega328p

Program:    1072 bytes (3.3% Full)

(.text + .data + .bootloader)

Data:          9 bytes (0.4% Full)

(.data + .bss + .noinit)
```

DancingQuanta, 2016-06-01:
Found out that someone tried the same thing as I did, in #377 and get the same error.

The proposed solution is to replace the id.exe in 1.6.x arduino with the one from 1.0.5 arduino. Will try this next time.

@bertjanstroop have you read #377?

bertjanstroop, 2016-06-08:
@DancingQuanta no not yet, i will try your proposed solution when the need to compile comes again.

rogosher, 2016-07-02:
I also came across this behavior in Win8.1 and IDE version 1.6.9.

The first time you try to compile your project source, you experience an ld.exe error. In my case `ld.exe` returned `collect2.exe: error: ld returned 255 exit status`. If you run `make` a second time the project will finish compiling.

This also happens with the PS instance open as an Administrator. 

I haven't attempted to replace `ld.exe yet`.