sudar/Arduino-Makefile #424: Compile failures under Arduino 1.6.8+ (with fix)

calvinli, 2016-04-13:
I suddenly found that compilation of even an empty sketch would fail either during compilation or linking of the Arduino core libraries under Arduino 1.6.8 (and the 1.6.9 nightlies). Looking at the history of the Arduino repo, it looks like everything back to Arduino 1.6.6 is probably also affected. (FastLED under any version is also affected.)

Adding `CXXFLAGS = --std=g++11 -fno-threadsafe-statics`
was necessary to get anything to build (the Arduino IDE uses these flags, which is how I found them).

Not sure what changed, but it looks like those two flags need to be added.

Example compile errors:
```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=nig -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/leonardo     -I../libraries/FastLED -Wall -ffunction-sections -fdata-sections -Os -DUSB_VID=0x2341 -DUSB_PID=0x8036  -fno-exceptions  /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp -o build-leonardo/core/USBCore.cpp.o
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp: In function 'int USB_RecvControl(void*, int)':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp:431:2: warning: 'auto' changes meaning in C++11; please remove it [-Wc++0x-compat]
  auto length = len;
  ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp:431:7: error: 'length' does not name a type
  auto length = len;
       ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp:432:8: error: 'length' was not declared in this scope
  while(length)
        ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp:436:3: warning: 'auto' changes meaning in C++11; please remove it [-Wc++0x-compat]
   auto recvLength = length;
   ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp:436:8: error: 'recvLength' does not name a type
   auto recvLength = length;
        ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp:437:6: error: 'recvLength' was not declared in this scope
   if(recvLength > 64){
      ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/USBCore.cpp:443:31: error: 'recvLength' was not declared in this scope
   Recv((u8*)d + len - length, recvLength);
                               ^
make: *** [build-leonardo/core/USBCore.cpp.o] Error 1
```

```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -mmcu=atmega32u4 -Wl,--gc-sections -Os -o build-leonardo/neopixel2.elf build-leonardo/neopixel2.ino.o build-leonardo/libcore.a  -lc -lm 
build-leonardo/neopixel2.ino.o: In function `setup':
neopixel2.ino:(.text.setup+0xc): undefined reference to `__cxa_guard_acquire'
neopixel2.ino:(.text.setup+0x9e): undefined reference to `__cxa_guard_release'
build-leonardo/libcore.a(PluggableUSB.cpp.o): In function `PluggableUSB()':
PluggableUSB.cpp:(.text._Z12PluggableUSBv+0xc): undefined reference to `__cxa_guard_acquire'
PluggableUSB.cpp:(.text._Z12PluggableUSBv+0x2c): undefined reference to `__cxa_guard_release'
collect2: error: ld returned 1 exit status
make: *** [build-leonardo/neopixel2.elf] Error 1
```

sudar, 2016-04-20:
Thanks for letting us know about it.

Right now we don't fully support Arduino 1.5 and above, but will try to incorporate this fix.

sej7278, 2016-04-20:
yes i'm not sure how well relying on c++11 would go down with a lot of users. i guess we could check for ide > 1.6

but it certainly works for me as-is, have you tried a non-empty script?

example Makefile that compiles Blink ok (on linux at least):

```make
ARDUINO_DIR = $(HOME)/arduino-1.6.8
BOARD_TAG = pro
BOARD_SUB = 8MHzatmega328
include $(HOME)/Arduino-Makefile/Arduino.mk
```

sej7278, 2016-05-05:
this is quite odd, as for me a blank sketch will never compile under any IDE as it fails with undefined references to setup() and loop(), so we definitely need an example sketch to reproduce this, i suspect one of the libraries you're using is the culprit.

we can't just go with:
```
CXXFLAGS_STD = -std=gnu++11
CXXFLAGS += -fno-threadsafe-statics
```
as it won't work on avr-g++ 4.3 as used in ide 1.0.x, so we'd need to parse ```CC_VERSION``` or ```ARDUINO_VERSION```

https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md

calvinli, 2016-05-05:
> this is quite odd, as for me a blank sketch will never compile under any IDE

I should clarify, I meant a sketch with blank setup() and loop() and no includes. The compile errors listed in the original report originate from code in the Arduino core libraries.

sej7278, 2016-05-05:
i can't reproduce in that case, as that's what i tried and it works fine. maybe its a mac thing?

3cooper, 2016-07-25:
I just ran into this yesterday.  I am on OS X.  No issues on empty loop and setup functions, but if I had a line for FastLED, it would error:

FastLED line in setup():

      FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);

Error messages:

    build-uno/FastLED_DemoReel100.cpp.o: In function `setup':
    FastLED_DemoReel100.cpp:(.text.setup+0x2a): undefined reference to `__cxa_guard_acquire'
    FastLED_DemoReel100.cpp:(.text.setup+0xbc): undefined reference to `__cxa_guard_release'
    collect2: error: ld returned 1 exit status
    make: *** [build-uno/FastLED_DemoReel100.elf] Error 1

If I added the line listed above to my Makefile, it worked.  As far as I have gotten is building.  Have sent anything to the board yet.  Let me know if any additional info would be helpful.

sej7278, 2016-07-25:
I made PR #442 although this seems to be specific to FastLED on OSX

closed 2016-07-25 by sudar

gauteh, 2016-10-28:
Just ran into this with: arduino-mk 1.5.1-1 on Arch Linux with latest arduino and arduino-builder. Some stuff installed using the arduino IDE to get boards.txt and libraries.

sej7278, 2016-10-29:
use arduino-mk from git the arch package is totally broken

gauteh, 2016-10-29:
Alright, thanks.

PR #442 (2016-07-25): Change CXXFLAGS_STD to match upstream
Added ```-std=gnu++11 -fno-threadsafe-statics``` to ```CXXFLAGS_STD``` if we're using Arduino 1.6 to match upstream (without it compilation seems to fail on OSX). Updated docs. Fixes #424 

Also made ```ARDUINO_LIB_PATH``` overloadable although this is a pretty niche use-case, see #441

Probably need to update HISTORY.md with the last couple of commits too.