sudar/Arduino-Makefile #426: avr-g++: error: missing device or architecture after '-mmcu='

joshuacox, 2016-04-28:
I've read through the other threads on this but I have my BOARD_SUB set to atmega328, and the core is set according to the arch wiki:
https://wiki.archlinux.org/index.php/arduino#Arduino-mk  

whcih tells me I need something like this in my Makefile:
BOARD_TAG   = nano
BOARD_SUB   = atmega328
ARDUINO_DIR = /usr/share/arduino
ARDMK_DIR = /usr/share/arduino
AVR_TOOLS_DIR = /usr
ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino
BOARDS_TXT = /usr/share/arduino/hardware/arduino/avr/boards.txt
ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino/avr/variants
BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders
include /usr/share/arduino/Arduino.mk

```
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /usr/share/arduino 
- [USER]               ARDMK_DIR = /usr/share/arduino 
- [AUTODETECTED]       ARDUINO_VERSION = 168 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/thoth/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/thoth/sketchbook (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr 
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/share/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [USER]               ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino/avr/variants 
- [USER]               BOARDS_TXT = /usr/share/arduino/hardware/arduino/avr/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /home/thoth/sketchbook/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = atmega328 
- [USER]               BOARD_TAG = nano 
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-nano-atmega328 (from BOARD_TAG)
- [USER]               ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 Average
- [USER]                 DHT
- [USER]                 Time
- [PLATFORM]             SPI
- [PLATFORM]             Wire
- [USER]               BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders 
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 5.3.0 (avr-gcc)
-------------------------
mkdir -p build-nano-atmega328
/usr/bin/avr-g++ -MMD -c -mmcu= -DF_CPU= -DARDUINO=168 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/    -I/usr/share/arduino/hardware/arduino/avr/libraries/SPI/src   -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/src   -I/home/thoth/sketchbook/libraries/Average   -I/home/thoth/sketchbook/libraries/DHT   -I/home/thoth/sketchbook/libraries/Time -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  DHT.cpp -o build-nano-atmega328/DHT.o
avr-g++: error: missing device or architecture after '-mmcu='
/usr/share/arduino/Arduino.mk:1207: recipe for target 'build-nano-atmega328/DHT.o' failed
make: *** [build-nano-atmega328/DHT.o] Error 1
```

full file here:
https://github.com/Bokbot/hydrobot/blob/master/arduino/hydrobot/Makefile

it was all working till I reinstalled on my machine, so I'm not certain what went wrong

I've got all the usual suspects installed:

```
3 community/avr-binutils 2.26-1 [installed]
    A set of programs to assemble and manipulate binary and object files for the AVR architecture
4 community/avr-gcc 5.3.0-2 [installed]
    The GNU AVR Compiler Collection
5 community/avr-gdb 7.11-1 [installed]
    The GNU Debugger for AVR
6 community/avr-libc 2.0.0-1 [installed]
    The C runtime library for the AVR family of microcontrollers
7 community/avrdude 1:6.3-1 [installed]
    Download/upload/manipulate the ROM and EEPROM contents of AVR microcontrollers

```

arduino:

```
3 aur/arduino 1:1.6.8-1 [installed] (483) (17.87)
    Arduino prototyping platform SDK
```

sej7278, 2016-04-28:
i think your Makefile is overly-complex so something has been set incorrectly, better to let it auto-detect your environment. this works for me:

```
ARDUINO_DIR = $(HOME)/arduino-1.6.8
BOARD_TAG = nano
BOARD_SUB = atmega328
include $(HOME)/Arduino-Makefile/Arduino.mk
```

joshuacox, 2016-04-28:
ya when I mentioned it was working before it was pretty minimal at that point, I added all that other stuff when attempting to fix it myself before reaching out here, I don't have anything in my $(HOME) other than the sketchbook it's all in `/usr/share/arduino`, so my similarly slim makefile is thus:

```
ARDUINO_DIR = /usr/share/arduino
BOARD_TAG   = nano
BOARD_SUB   = atmega328
include /usr/share/arduino/Arduino.mk
```

same result though:

```
avr-g++: error: missing argument to '-mmcu='
```

sej7278, 2016-04-28:
i've no idea really, it looks like it should work. could you try a basic (blink?) script without any libraries.

i assume you've not got 1.0 and 1.6 mixed in there - don't usually see 1.6 installed in /usr/share/arduino

joshuacox, 2016-04-28:
ya same thing

```
/usr/share/arduino/examples/02.Digital/BlinkWithoutDelay
```

sej7278, 2016-04-28:
core and variant are empty too.

any chance you could try the git version, maybe the arch package is messed up?

joshuacox, 2016-04-29:
ugh, I forgot the arduino repo was a full gig!

while it downloads I've been toying with a docker image that does it instead using debian jessie:

https://github.com/joshuacox/mkarduino

`make run` and `make enter` and from inside the container I can even program over USB.  This might be end up being how I program for these things if on arch linux like I am on this machine.

sej7278, 2016-04-29:
i did kinda mean the git version of arduino-mk not arduino lol ;)

sej7278, 2016-04-30:
i've not really got any arch experience, maybe i'll fire up a docker image to play with, but i can only think the problem is the arch package of arduino-mk (or the ide?) as its almost a year out-of-date, we need to see if you get the same problem from the git head of arduino-mk

joshuacox, 2016-04-30:
Sorry about that, I was traveling yesterday.  Sure enough checking the git rep (of Arduihno-Makefile, and not Arduino itself, lol).  And then including it properly e.g.

`include /home/thoth/git/Arduino-Makefile/Arduino.mk`

results in a perfectly fine compile, I'm raising this issue with the [AUR](https://aur.archlinux.org/packages/arduino-mk/) and closing here. @sej7278  I greatly appreciate the second set of eyes on my situation.

PS - the docker thing surprisingly works very well if anyone is stuck inbetween a rock and a hard place.

closed 2016-04-30 by joshuacox

sej7278, 2016-04-30:
not a problem @joshuacox, the AUR probably just needs to update to the 1.5.1 release (git HEAD), i'm quite surprised docker supports serial uploading and stuff too, that's cool.

joshuacox, 2016-04-30:
of note, it is the `--privileged` option that grants this power to the docker container.  Which is complete unmitigated access to `/dev/*` amongst other things (like the ability to overwrite `/etc/shadow` ; ).  But this is me running it on my own machine dedicated to this purpose, so I'm comfortable with the risk.

embed, 2017-08-04:
Currently I am having the same problem. Manjaro (Arch) Linux, MArduino-Makefile from git(here), Arduino 1.8.2 from Arduino.cc. Makefile

    BOARD_TAG ?= pro
    MONITOR_PORT ?= /dev/ttyUSB5
    
    ARDUINO_LIBS = JsonRPCServer \
                                DCServo
    
    include $(ARDMK_DIR)/Arduino.mk

Also I have exactly the same problem if I compile the Blink example for anything other than uno. I need to build for pro. I think I need to choose a sub option.

sej7278, 2017-08-04:
yes if you use arduino 1.5 or later you will generally need a submenu-board, there's no such thing as a "pro"

```
$ ARDUINO_QUIET=1 make show_boards|grep -i pro
pro                 Arduino Pro or Pro Mini

$ ARDUINO_QUIET=1 make show_submenu|grep -i pro
pro        16MHzatmega168  ATmega168 (5V, 16 MHz)
pro        16MHzatmega328  ATmega328 (5V, 16 MHz)
pro        8MHzatmega168   ATmega168 (3.3V, 8 MHz)
pro        8MHzatmega328   ATmega328 (3.3V, 8 MHz)
```

sej7278, 2017-08-05:
@jobegit looks like you have a similar issue - you're using the old 1.0.x style BOARD_TAG without a BOARD_SUB, but you probably also need to specify your ARDUINO_SKETCHBOOK

embed, 2017-08-05:
@sej7278  BOARD_SUB=16MHzatmega328 solved it, thank you. @JobeGIT

mrphysh, 2018-08-30:
I am having the same issue  ..  maybe exactly the same.  i am using a Mega and a simple Makefile.  My Linux Mint distro is new.  The IDE has never worked, but I like the idea of using the cmd line anyway.
I see no path for diagnosing this.  I tried to delete the Arduino altogether, but the system would not let me.  I am committed to learning this system.  Any guidance.?
[make_error_compile_txt.txt](https://github.com/sudar/Arduino-Makefile/files/2337561/make_error_compile_txt.txt)

mrphysh, 2018-08-30:
I am having the same issues.  I just tried [sudo apt autoremove] followed by [sudo apt-get install avrdude make gcc-avr arduino-mk]  still same problem  I see no way to diagnose this!  (The IDE does not work)

sej7278, 2018-08-31:
@mrphysh you're using arduino 1.0.5 from the debian (mint) repo's, so this should work with a mega2560 (its not the same issue at all):

```make
BOARD_TAG = mega2560
include /usr/share/arduino/Arduino.mk
```

i noticed you're Arduino.mk is lowercase, have you broken your install? this should be all you need to fix it:

```
sudo apt-get install --reinstall arduino-core arduino-make
```

mrphysh, 2018-08-31:
thanks,  that helped a lot.

sammilei, 2020-04-24:
I am using maga2560 in Mac
the way I fixed it is to use _SUB instead of _TAG:

`ARDUINO_DIR   = /Applications/Arduino.app/Contents/Java
MONITOR_PORT  = /dev/cu.usbmodem14201

USER_LIB_PATH = /Users/xxx/Documents/Arduino/libraries
BOARD_SUB   = mega2560

include /Users/xxx/Documents/Arduino/Arduino-Makefile/Arduino.mk`

Lazlolozla, 2023-01-07:
Yo tenía el mismo problema con mi clon chino de arduino mega2560. Yo había instalado Arduino.mk hace como 5 años en lubuntu 14.04  y funcionaba perfectamente. Despues cambié de equipo y ahora uso Kubuntu 22.04 pero no me funcionaba. Hasta que llegué aquí y pude solucionarlo con sus aportaciones. Gracias!

 En lubuntu 14.04 lo tenía así: (Ya no recuerdo de donde lo copie el Makefile no es mio) usando el antiguo Arduino.mk

 #make – Sólo compila
 #make upload – Compila y sube
 #make clean – Borra todas las dependencias creadas
#make depends – Actualiza las dependencias
 #make reset – reset the Arduino by tickling DTR on the serial port
 #make raw_upload – upload without first resetting
 #make show_boards – list all the boards defined in boards.txt
 ARDUINO_DIR = /usr/share/arduino
 ARDUINO_PORT = /dev/ttyACM0
 
 USER_LIB_PATH = /home/usuario1/sketchbook/arduinomega2560/libraries
 BOARD_TAG = mega2560
 
 include /usr/share/arduino/Arduino.mk

Ahora en Kubuntu 22.04 quedo así: https://github.com/sudar/Arduino-Makefile/commit/55e003ff6ca8aade08df07ecee34f7b072c06f77

#make – Sólo compila
#make upload – Compila y sube
#make clean – Borra todas las dependencias creadas
#make depends – Actualiza las dependencias
#make reset – reset the Arduino by tickling DTR on the serial port
#make raw_upload – upload without first resetting
#make show_boards – list all the boards defined in boards.txt
ARDUINO_DIR = /usr/share/arduino
 ARDUINO_PORT = /dev/ttyACM0
 
USER_LIB_PATH = /home/usuario1/sketchbook/arduinomega2560/libraries
BOARD_TAG = mega
BOARD_SUB = atmega2560
 
include /usr/share/arduino/Arduino-Makefile/Arduino.mk