sudar/Arduino-Makefile #427: Binary file is greater than the Uno's memory (Windows)

markglavind, 2016-04-28:
I'm having trouble with uploading the sketch to the Arduino, where i am getting the following errors

> AVR Memory Usage
> ----------------
> Device: atmega328p
> 
> Program:    1088 bytes (3.3% Full)
> (.text + .data + .bootloader)

> Data:          9 bytes (0.4% Full)
> (.data + .bss + .noinit)

> /c/cygwin64/bin/awk: /c/cygwin64/bin/awk: cannot execute binary file
> /usr/bin/sh: [: -le: unary operator expected
> \nThe size of the compiled binary file is greater than the uno's flash memory. See http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
> make: *** [verify_size] Error 1

However, the file should be nowhere near the size of maxing out the Arduino? Except for a libcore.a file with a size of 89KB.

This is my Makefile:
``
``

sej7278, 2016-04-28:
i assume line [1261](https://github.com/sudar/Arduino-Makefile/blob/master/Arduino.mk#L1261) is the problem, perhaps with the backticks? maybe cygwin's version of awk is out-of-date?

markglavind, 2016-04-28:
Sorry, but im not sure what you mean with the backticks? My cygwin install is fresh, so i would guess i have the newest version

closed 2016-04-28 by markglavind