sudar/Arduino-Makefile #428: Access denied during upload (windows)

markglavind, 2016-04-28:
When it tries to upload the sketch to the Arduino, it starts the uploading procedure but gives the following output:

> mkdir

>  -p build-uno
> make reset
> Usage: egrep [OPTION]... PATTERN [FILE]...
> Try `egrep --help' for more information.
> Usage: egrep [OPTION]... PATTERN [FILE]...
> Try `egrep --help' for more information.
> make[1]: Entering directory `C:/Users/Mark/Documents/Arduino/Blink'
> C:\Users\Mark\Desktop\Arduino-MakeFile\Arduino-Makefile-master/bin/ard-reset-arduino  com3
> make[1]: Leaving directory `C:/Users/Mark/Documents/Arduino/Blink'
> make do_upload
> Usage: egrep [OPTION]... PATTERN [FILE]...
> Try `egrep --help' for more information.
> Usage: egrep [OPTION]... PATTERN [FILE]...
> Try `egrep --help' for more information.
> make[1]: Entering directory `C:/Users/Mark/Documents/Arduino/Blink'
> C:/WinAVR-20100110/bin -q -V -p atmega328p -C C:/Users/Mark/Desktop/Arduino-MakeFile /Arduino-Makefile-master/AvrFiles/../avrdude.conf -D -c arduino -b 115200 -P com3 \
                        -U flash:w:build-uno/Blink.hex:i
> process_begin: CreateProcess(NULL, C:/WinAVR-20100110/bin -q -V -p atmega328p -C C:/Users/Mark/Desktop/Arduino-MakeFile/Arduino-Makefile-master/AvrFiles/../avrdude.conf -D -c arduino -b 115200 -P com3 -U flash:w:build-uno/Blink.hex:i, ...) failed.
> make (e=5): Access denied.
> make[1]: *** [do_upload] Error 5
> make[1]: Leaving directory `C:/Users/Mark/Documents/Arduino/Blink'
> make: *** [upload] Error 2

Since it says access denied i thought it was an access problem, and i changed the security settings for the destinations printed in the Process_Begin line to full controle, but that did not help. Anyone has any idea what it could be?

sej7278, 2016-04-28:
its not permissions, if you look at the error message, egrep is failing, again something wrong with youe cygwin setup i'd say

markglavind, 2016-04-29:
Is it the line with 

> Usage: egrep [OPTION]... PATTERN [FILE]...
> Try egrep --help' for more information. Usage: egrep [OPTION]... PATTERN [FILE]... Tryegrep --help' for more information.

That makes you think it is a cygwin problem? or where do you see the error? I'm not really sure what to do about cygwin, except looking for extra packages that could have something to do with egrep. Is there some sort of setup guide available?