sudar/Arduino-Makefile #431: Teensyduino 1.28 places boards.txt in different directory (OSX)

tomsmalley, 2016-06-06:
BOARDS_TXT as defined in Teensy.mk:38 is
```
BOARDS_TXT          = $(ARDUINO_DIR)/hardware/$(ARDMK_VENDOR)/boards.txt
```
however with Arduino 1.6.8 and Teensyduino 1.28, boards.txt is actually in
```
$(ARDUINO_DIR)/hardware/teensy/avr/boards.txt
```
Additionally, boards.txt does not define the architecture which is later used determining other directories, this also cannot be overridden in the makefile.

sej7278, 2016-06-07:
surely this fixes it? should work for 1.0.x and 1.6.x (as ARCH defaults to avr):

```
BOARDS_TXT          = $(ARDUINO_DIR)/hardware/$(ARDMK_VENDOR)/$(ARCHITECTURE)/boards.txt
```

tomsmalley, 2016-06-08:
Indeed that is what I added, had additional trouble with information missing from boards.txt itself though.

sej7278, 2016-06-08:
so is that a teensy bug or what is the trouble you had?

adding ```$(ARCHITECTURE)/``` we could do as it won't break anything, but any more would need info

tomsmalley, 2016-06-08:
boards.txt does not actually contain the architecture which Teensy.mk looks for, so ARCHITECTURE=avr is never set even if you user define it. With the above fix, there is a circular reference somewhere in BOARDS_TXT. But adding an ifdef around ARCHITECTURE so it can be user defined, the files can be found with no problem. I don't know if this would be classed as a teensy bug, is it just a change with the new version?

Anyway, the problem then becomes ```arm-none-eabi-g++: error: unrecognized argument in option '-mcpu=mk20dx256'``` because the makefile can't find the mcu from boards.txt (it looks at build.cpu when it is build.flags.cpu).

Fixing that gave me ```arm-none-eabi-g++: error trying to exec 'cc1plus': execvp: No such file or directory``` which would seem to be my problem rather than this makefiles.

sej7278, 2016-11-10:
Fixed with commit 2c1155e156ac0c

closed 2016-11-10 by sej7278