sudar/Arduino-Makefile #433: Documentation: Python or perl

az-z, 2016-06-18:
hello,
in the guide http://hardwarefun.com/tutorials/compiling-arduino-sketches-using-makefile you state : "
Install dependencies

The Makefile delegates resetting the board to a short Perl program. You’ll need to install Device::SerialPort and YAML library."

in the README () you state "
pySerial
 The Makefile also delegates resetting the board to a short Python program. You'll need to install pySerial to use it though."

sudar, 2016-06-18:
You have to install pySerial. The documentation in github is the current.

The guide at  http://hardwarefun.com/tutorials/compiling-arduino-sketches-using-makefile is out of date. I will update it.

sej7278, 2016-06-18:
that's just an old blog post, its not the guide.

the guide is https://github.com/sudar/Arduino-Makefile#requirements and states its python not perl

az-z, 2016-06-18:
thank you gents. Then, if i may suggest,  update the home page (https://github.com/sudar/Arduino-Makefile) and either remove the reference to the blog or (better) just state that documentation supersedes the blog. The way i read it is - "go to the blog for the detailed how-to instructions".
Thank you

az-z, 2016-06-18:
let me figure out what i need to do and i will submit the pull request.

closed 2016-06-18 by az-z