sudar/Arduino-Makefile #439: building plain C file with arduino.h and other libraries

tizziano, 2016-07-16:
Hi,

I'd like to build a C program that allows me to include the arduino core library. So here is my program:

````
/*
* blink-main.c
*/

#include <Arduino.h>

int main(void)
{
//
// setup
//
init(); // starts arduino library
pinMode(13,OUTPUT);

//
// loop
//
while(1){
  digitalWrite(13,HIGH);
  delay(1000);
  digitalWrite(13,LOW);
  delay(1000);
}
return 0;
}
````
and here his makefile:
````
# directory make di sistema
MAKE_DIR=$(HOME)/.local/lib/arduino_mk-151
MAKE_FILE=Arduino.mk

#
# Settings for Arduino-R3 board in C but using arduino.h
#
ARDUINO_CORE_PATH = $(HOME)/.local/share/arduino-1.6.4/hardware/arduino/avr/cores/arduino
ARDUINO_VAR_PATH = $(ARDUINO_DIR)/hardware/arduino/avr/variants/standard/

BOARD_TAG = AVR-gcc
MCU = atmega328p
F_CPU = 16000000L
HEX_MAXIMUM_SIZE = 32256
#
# avrdude settings for Arduino-R3 board
# flashed via bootloader
#
AVRDUDE_ARD_PROGRAMMER=arduino
AVRDUDE_ARD_BAUDRATE=115200

include $(MAKE_DIR)/$(MAKE_FILE)
````
This works fine and all applicable targets are working correctly.

But, now I would try to include some other library, so:

````
/*
* blink-main.c
*/
#include <Arduino.h>
/*
* Test from standard libraries:
*/
#include <LiquidCrystal.h> // here I get unknown type name 'class' error
/*
* Test from user libraries
*/
#include <blinklib.h> // here I get unknown type name 'class' error
#include <blinklibc.h> // this is a C library, so no problem
````
So, if the library is in C++ I get always an error of this kind:

`/home/tiziano/lavori/arduino/libraries/blinklib/blinklib.h:17:1: error: unknown type name 'class'`

There is a way to solve this problem? Am I missing something?

thanks for your kind advice

electrokean, 2016-07-17:
I'd guess this is because the file extension of .c will invoke the C compiler, instead of the C++ compiler, so the C++ keyworks in blinklib.h (e.g. class) aren't understood.

What if you rename your blink-main.c to blink-main.cpp or better blink-main.ino ?

tizziano, 2016-07-17:
You are right. If I rename the main program from blink-main.c to blink-main.cpp the build proceeds regularly as expected (upload included). 

Now I have to figure out if a usage pattern of this kind has some advantages against building a regular arduino .ino sketch. 

Thank you!

closed 2016-09-11 by tizziano