sudar/Arduino-Makefile #440: avrdude not found even though IDE and OS versions both exist (Linux)

celem, 2016-07-19:
1. My Linux version = Linux Mint 17.3 Rosa, 3.16.0-38-generic #52~14.04.1-Ubuntu SMP Fri May 8 09:43:57 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
2. avrdude was Arduino IDE installed at "/home/ecomer/arduino-1.6.7/hardware/tools/avr/bin/avrdude" and "/home/ecomer/arduino-1.6.7/hardware/tools/avr/etc/avrdude.conf"
3. The Linux OS also installed avrdude at "/usr/bin/avrdude" and "/etc/avrdude.conf"
4. My system has /home/ecomer/atmel installed but it does not contain avrdude files
5. My Arduino IDE is version 1.6.7
6. AVRDUDE and AVRDUDE_CONF are not currently exported to the environment - I'm trying to see why Arduino.mk fails to find avrdude
7. Arduino.mk fails to find either set of avrdude files, the IDE installs or the OS installs. Here is the console output:
...
/home/ecomer/atmel/bin/avrdude -q -V -p atmega328p -C /home/ecomer/atmel/etc/avrdude.conf -D -c arduino -b 115200 -P /dev/ttyACM0 \
			-U flash:w:build-uno/Blink.hex:i
make[1]: /home/ecomer/atmel/bin/avrdude: Command not found
...
It appears that the existence of a BUNDLED_AVR_TOOLS_DIR that lacks avrdude is confusing Arduino.mk's current logic.

sej7278, 2016-07-20:
can you show us the arduino.mk output and also your $PATH ?

it usually looks in ```$(ARDUINO_DIR)/hardware/tools/avr/bin``` or the same directory as avr-gcc which is usually /usr/bin but it seems its finding it in ```$(HOME)/atmel/bin``` so i suspect you've set your ```$PATH``` in an odd way, as ```$(BUNDLED_AVR_TOOLS_DIR)``` should be in your arduino directory not the atmel one.

celem, 2016-07-20:
My arduino.mk is version 1.5.1 at https://github.com/sudar/Arduino-Makefile/blob/master/Arduino.mk without modification.

$ echo $PATH
/home/ecomer/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/jvm/java-8-oracle/bin:/usr/lib/jvm/java-8-oracle/db/bin:/usr/lib/jvm/java-8-oracle/jre/bin

BTW - I edited .bashrc, adding the following and I can successfully utilize avrdude. I knew that this would work and my reason for opening this issue is because the arduino.mk logic, in my opinion, should have found the avrdude installed by the Arduino IDE in "/home/ecomer/arduino-1.6.7/hardware/tools/avr", which it did not do.
...
export AVRDUDE=/home/ecomer/arduino-1.6.7/hardware/tools/avr/bin/avrdude
export AVRDUDE_CONF=/home/ecomer/arduino-1.6.7/hardware/tools/avr/etc/avrdude.conf

Here is what my $PATH finds:
$ whereis avrdude
avrdude: /usr/bin/avrdude /etc/avrdude.conf /usr/bin/X11/avrdude /usr/share/man/man1/avrdude.1.gz

I suspect that "/home/ecomer/atmel" was installed in 2012 when I was working directly on Atmel chips without use of the Arduino tools, and did an install from Atmel - meaning that "/home/ecomer/atmel" was not installed as part of, or at the same time as, the Arduino software.

sej7278, 2016-07-20:
could you should us the arduino.mk output, that starts:
```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
```

and also your Makefile?

as i have no idea how/why it even looked in the atmel directory as its not even in your $PATH, unless its symlinked somewhere?

usually i just set ```ARDUINO_DIR = /path/to/arduino-1.6.9``` and it works, even though i have /usr/bin/avrdude and /etc/avrdude.conf like yourself (on debian)

if you remove (or rename/move) the atmel directory, does it work then?

celem, 2016-07-21:
[[[[[[CONSOLE WITH EXPLICIT AVRDUDE PATHS NOT IN .BASHRC]]]]]]

```
ecomer@bergamont ~/Arduino/Arduino-Makefile/examples/Blink $ make upload
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/ecomer/arduino-1.6.7 
- [USER]               ARDMK_DIR = /home/ecomer/Arduino/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 167 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/ecomer/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/ecomer/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /home/ecomer/atmel 
- [COMPUTED]           ARDUINO_LIB_PATH = /home/ecomer/arduino-1.6.7/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/ecomer/arduino-1.6.7/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/ecomer/arduino-1.6.7/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /home/ecomer/arduino-1.6.7/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/ecomer/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /home/ecomer/arduino-1.6.7/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyACM0 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /home/ecomer/arduino-1.6.7/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
-------------------------
mkdir -p build-uno
make reset
make[1]: Entering directory `/home/ecomer/Arduino/Arduino-Makefile/examples/Blink'
/usr/bin/ard-reset-arduino  /dev/ttyACM0
make[1]: Leaving directory `/home/ecomer/Arduino/Arduino-Makefile/examples/Blink'
make do_upload
make[1]: Entering directory `/home/ecomer/Arduino/Arduino-Makefile/examples/Blink'
/home/ecomer/atmel/bin/avrdude -q -V -p atmega328p -C /home/ecomer/atmel/etc/avrdude.conf -D -c arduino -b 115200 -P /dev/ttyACM0 \
			-U flash:w:build-uno/Blink.hex:i
make[1]: /home/ecomer/atmel/bin/avrdude: Command not found
make[1]: *** [do_upload] Error 127
make[1]: Leaving directory `/home/ecomer/Arduino/Arduino-Makefile/examples/Blink'
make: *** [upload] Error 2
```

[[[[[NEXT, I PUT THE AVRDUDE PATHS BACK INTO .BASHRC - THIS MAKES IT WORK]]]]]

```
ecomer@bergamont ~/Arduino/Arduino-Makefile/examples/Blink $ make upload
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/ecomer/arduino-1.6.7 
- [USER]               ARDMK_DIR = /home/ecomer/Arduino/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 167 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/ecomer/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/ecomer/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /home/ecomer/atmel 
- [COMPUTED]           ARDUINO_LIB_PATH = /home/ecomer/arduino-1.6.7/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/ecomer/arduino-1.6.7/hardware/arduino/avr/libraries (from ARDUINO_DIR)
[... 47 lines cut ...]
BOARD_TAG    = uno
include ../../../Arduino-Makefile/Arduino.mk
```

[[[[[My .bashrc (with avrdude paths present)]]]]]

```
export ARDUINO_DIR=/home/ecomer/arduino-1.6.7
export ARDMK_DIR=/home/ecomer/Arduino/Arduino-Makefile
export AVR_TOOLS_DIR=/home/ecomer/atmel
export AVRDUDE=/home/ecomer/arduino-1.6.7/hardware/tools/avr/bin/avrdude
export AVRDUDE_CONF=/home/ecomer/arduino-1.6.7/hardware/tools/avr/etc/avrdude.conf
```

[[[[[Does renaming the atmel make it work? - No, here is the output]]]]]
```
...
make[1]: /home/ecomer/atmel/bin/avr-gcc: Command not found
/home/ecomer/atmel/bin/avrdude -q -V -p atmega328p -C /home/ecomer/atmel/etc/avrdude.conf -D -c arduino -b 115200 -P /dev/ttyACM0 \
			-U flash:w:build-uno/Blink.hex:i
make[1]: /home/ecomer/atmel/bin/avrdude: Command not found
make[1]: *** [do_upload] Error 127
make[1]: Leaving directory `/home/ecomer/Arduino/Arduino-Makefile/examples/Blink'
make: *** [upload] Error 2
```

celem, 2016-07-21:
UPDATE: after I commented out the AVRDUDE entries in the .bashec AND renamed the atmel folder, that /home/ecomer/atmel/bin/avr-gcc was being referenced by arduino.mk. I combed through the arduino.mk file and saw that that path was built if AVR_TOOLS_DIR was in the environment. Sure enough, it was being set in the .bashrc (see previous post). I also commented out  AVR_TOOLS_DIR in .bashrc did a make clean, then make and then make upload. ALL WORKED CORRECTLY.

I did not create that AVR_TOOLS_DIR entry in the .bashrc, or I don't recall doing so. I initially installed Arduino-Makefile via the OS's apt-get - so I wonder if that is how the AVR_TOOLS_DIR entry got created in in .bashrc?

In any case, with my .bashrc as shown below, everything works correctly - compilation works and avrdude upload works. I'm sorry if this issue wasted too much of your time.

MY .bashrc file that WORKS!
```
export ARDUINO_DIR=/home/ecomer/arduino-1.6.7
export ARDMK_DIR=/home/ecomer/Arduino/Arduino-Makefile
```

sudar, 2016-07-21:
Glad to know that you got it working. Closing it now.

@sej7278 As always thanks for your help :)

closed 2016-07-21 by sudar

sej7278, 2016-07-21:
i reckon that atmel thing put it it .bashrc as there was no way the atmel tools would have been found otherwise as it wasn't in the $PATH, certainly the debian package wouldn't do that. anyway great you got it working now.