sudar/Arduino-Makefile #441: Q: advice on using MegaCore with Arduino-Makefile?

beasfall, 2016-07-23:
Hi -

We've been working on using https://github.com/MCUdude/MegaCore with Makefile for a while now, and have the chip programming and running with low-level code, but cannot get the standard Arduino libraries to work.

Any advice on the "correct" way to use the MegaCore package above?

much thanks!

sej7278, 2016-07-23:
from a quick look you'd probably have to make a 1.0.x style boards.txt, we don't support ```menu.clock```

beasfall, 2016-07-23:
thanks for the quick response! In their boards.txt file, we already added:

```
128.build.f_cpu=16000000L
```

and that seemed to at least parse the file correctly when `BOARD_TAG=128` and `BOARDS_TXT` is explicitly set. However we're having trouble figuring out how to reference the rest of the files, and how they interact merge with our existing configuration.

Right now we install all these packages via apt-get:

```
package  "avr-libc"       #   - Standard C library for Atmel AVR development
package  "avrdude"        #   - software for programming Atmel AVR microcontrollers
package  "binutils-avr"   #   - Binary utilities supporting Atmel's AVR targets
package  "gcc-avr"        #   - The GNU C compiler (cross compiler for avr)
package  "gdb-avr"        #   - The GNU Debugger for avr

package  "arduino-core"
package  "arduino-mk"

# dependencies for Arduino-Makefile
package  "libdevice-serialport-perl"
package  "libyaml-perl"
```

I'm guessing we need to put the MegaCore files somewhere and then update Makefile vars like `USER_LIB_PATH` and `ARDUINO_DIR`, but it's not clear to me how those map to their files.

Sorry for the long message, and thanks for a great project!

beasfall, 2016-07-23:
I'm realizing that one point of confusion for us is that we are using the `arduino-core` package rather than installing the Arduino IDE, so the installation process is different... typically for us,

```
ARDUINO_DIR = /usr/share/arduino
```

sej7278, 2016-07-24:
i had to tweak Arduino.mk to allow the user to overload ARDUINO_LIB_PATH:

```diff
diff --git a/Arduino.mk b/Arduino.mk
index b522e64..66830a1 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -472,8 +472,12 @@ ifndef AVR_TOOLS_PATH
     AVR_TOOLS_PATH    = $(AVR_TOOLS_DIR)/bin
 endif
 
-ARDUINO_LIB_PATH  = $(ARDUINO_DIR)/libraries
-$(call show_config_variable,ARDUINO_LIB_PATH,[COMPUTED],(from ARDUINO_DIR))
+ifndef ARDUINO_LIB_PATH
+    ARDUINO_LIB_PATH = $(ARDUINO_DIR)/libraries
+    $(call show_config_variable,ARDUINO_LIB_PATH,[COMPUTED],(from ARDUINO_DIR))
+else
+    $(call show_config_variable,ARDUINO_LIB_PATH,[USER])
+endif
 
 # 1.5.x platform dependent libs path
 ifndef ARDUINO_PLATFORM_LIB_PATH
```

beasfall, 2016-07-24:
Very informative, thanks a lot! We'll be working on it again tomorrow and will look at your advice carefully then.

It looks like we're also on an older version of `Arduino-Makefile` which predates the `ALTERNATE_CORE` var, so that's helpful new info also.

sej7278, 2016-07-24:
1.5-2 is in 15.10 onwards:  https://launchpad.net/ubuntu/+source/arduino-mk

or you could check out the git version.

sej7278, 2016-07-24:
i might give this a try, i've got a mega1284 and 644 although i found the [mighty-1284p](https://github.com/wijnen/mighty-1284p) and [sanguino](https://github.com/Lauszus/sanguino/tree/master) cores very straightforward, what's new in this one?

beasfall, 2016-07-24:
Basically we've got a lot of code and a toolchain (described above) that works great on the ATmega328. Now we have an application that needs a lot more I/O (30 digital and 8 analog), and we went looking for a chip with more pins that could run the same code... The alternative we're considering is a bunch of I2C port expanders or something like that.

We're not too familiar with the AVR/Arduino ecosystem so the pointers are appreciated!

sej7278, 2016-07-24:
if you literally don't want to change your existing code and toolchain from 328p to 1284p, i'd say forget the megacore and just:

```sudo apt-get install arduino-mk arduino-mighty-1284p arduino-core```

and be done with it (on a newish version of ubuntu/debian). the 1284 core installs into /usr/share/arduino/hardware/ so no need to do anything with libraries etc; just a simple Makefile:

```make
BOARD_TAG = mighty_opt
ALTERNATE_CORE_PATH = /usr/share/arduino/hardware/mighty-1284p
include /usr/share/arduino/Arduino.mk
```

beasfall, 2016-07-24:
Looks promising, but still looks like it leaves us 6 digital pins short factoring in 8 analog ... though maybe we can cut back :)

sej7278, 2016-07-24:
ah i didn't realise the atmega128 was a specific chip, though it meant 128* (as in 1284p)

anyway if you use IDE 1.6.9 i can get megacore to use its own libraries (e.g. Wire.h) without messing with other paths using:

Makefile:
```make
ARDUINO_DIR = /path/to/arduino-1.6.9
BOARD_TAG = 128
ALTERNATE_CORE = megacore
F_CPU = 16000000L
ARDUINO_PLATFORM_LIB_PATH  = $(HOME)/arduino16/hardware/megacore/avr/libraries/
include /path/to/Arduino.mk
```

arduino16 sketchbook - note avr subdirectory:
```
hardware/
└── megacore
    └── avr
        ├── avrdude.conf
        ├── boards.txt
        ├── bootloaders
        ├── cores
        ├── externalprogrammers.txt
        ├── libraries
        ├── platform.txt
        └── variants
```

beasfall, 2016-07-25:
Hey - just wanted to follow-up and share our experience:

With your help we were able to build and program the atmega64 (not 128 as I previously indicated).

Unfortunately, we continued to encounter issues with running code from the MegaCore libraries - in particular a simple `digitalWrite` consistently caused the chip to crash and restart. We investigated for some time but couldn't identify the issue and decided to cut our losses and move on.

Thanks again for your help!

closed 2016-07-25 by beasfall

sej7278, 2016-07-25:
that's a shame. megacore did seem a bit of a hack. i'd say try mighty-1284p or i guess the regular 328p with a couple of mcp23017 port expanders.

beasfall, 2016-07-26:
It's all good - worth a try! We've decided to use port expanders with the 328 because the 1284 would require an expander anyway for the I/O we need.

thanks for the heads-up on the packages - still loving this project