sudar/Arduino-Makefile #447: user defined AVR_TOOLS_DIR and AVRDUDE_CONF

tizziano, 2016-08-06:
If I define AVR_TOOLS_DIR=/usr in order to use the avr compiler from the package installed in my XUbuntu 16.04 system, I can't upload my sketch because AVRDUDE_CONF is not calculated correctly.

Here is the error I get from avrdude:  
`avrdude: can't open config file "/usr/etc/avrdude.conf": No such file or directory`
`avrdude: error reading system wide configuration file "/usr/etc/avrdude.conf"`

I have solved defining also AVRDUDE_CONF=/etc/avrdude.conf and all works fine.

sej7278, 2016-08-06:
you shouldn't need to define any of that manually (and you're setting them wrong too), this should be all you need:

```make
BOARD_TAG = leonardo
MONITOR_PORT = /dev/ttyACM0
include /usr/share/arduino/Arduino.mk
```

tizziano, 2016-08-07:
Sorry, I had to explain better what I was doing. I need to compile a no-core C source file. This is my makefile and I was trying to bypass the avr tools of the Arduino IDE 1.0.5 and use the more recent avr-tools installed from the xubuntu repository.  

Now I understand why I don't need to define them manually, since the Arduino IDE tools (that, in the system I'm testing on, are installed from the XUbuntu package) are links to the real avr-tools. 

Thanks.
```
#
# Arduino Make file. Refer to https://github.com/sudar/Arduino-Makefile
#
#  Copyright 2015 tiziano <tiziano.detogni@gmail.com>
#  
#
#  Settings for ATMega328 in plain C
#
AVR_TOOLS_DIR =/usr
AVRDUDE_CONF=/etc/avrdude.conf

NO_CORE    = Yes
BOARD_TAG  = AVR-gcc
MCU        = atmega328p
F_CPU      = 16000000L
HEX_MAXIMUM_SIZE = 32256

# here I have some C libraries to simplify GPIO
USER_LIB_PATH = $(HOME)/avr/libraries

#
#  Settings for avrdude 
#   and MCU with bootloader
#
AVRDUDE_ARD_PROGRAMMER=arduino
AVRDUDE_ARD_BAUDRATE=115200

#
# Settings for using another arduino as ISP programmer
#
ISP_PORT     = $(DEVICE_PATH)
ISP_PROG   = stk500v1
AVRDUDE_ISP_BAUDRATE = 19200

# fuses  --> atmega328@16MHz
ISP_LOCK_FUSE_PRE  = 0x3f
ISP_LOCK_FUSE_POST = 0x0f
ISP_HIGH_FUSE      = 0xde
ISP_LOW_FUSE       = 0xff
ISP_EXT_FUSE       = 0x05

include $(HOME)/.local/lib/arduino_mk-151/Arduino.mk
```

sej7278, 2016-08-08:
so is this sorted now, can it be closed?

tizziano, 2016-08-11:
yes, thans

closed 2016-08-11 by tizziano