sudar/Arduino-Makefile #450: unexpected EOF while looking for matching `}'

Babbleshack, 2016-08-28:
Cannot compile simple Blink example due to a missing '}', however I cannot find the refered to file. The Makefile output refers to line 1226 in Ardunio.mk, the refereed to line holds dependency declaration.

```

sej7278, 2016-08-29:
where did you get the arduino-mk package from - is it debian, ubuntu, arch, fedora.... and which version is the package? as i recall the aur package is out-of-date and probably broken, have you tried a git checkout?

it seems to be failing at the arduino version detection part, but we don't use a variable called ```${ver```

did you edit the Makefile that came with the example (other than ```BOARD_TAG=mega2560```)?

it looks like you've installed the arduino.cc 1.6.x tarball to /usr/share/arduino which is really odd and quite likely to break things.

Babbleshack, 2016-08-30:
I got it from the arch repo (so its likely out of date :p), however I believe I also tried a clone from the repo.

I tried several incantations of the makefile before posting this! 
Mainly around the example variables, as shown in my previous post.

The arch package install arduino to `/usr/share/arduino`, has this dir moved?

I can try with the latest version of the repo when I get home from work today.

sej7278, 2016-08-30:
the arch repo i think installs arduino 1.6 to /usr/share/arduino which may cause problems - i've only ever tried 1.0 in /usr/share/arduino with 1.6 in a user-writable directory such as $HOME.

this works for me for 1.6:
```make
ARDUINO_DIR = /home/user/arduino-1.6.10
BOARD_TAG = mega
BOARD_SUB = atmega2560
include /home/user/Arduino-Makefile/Arduino.mk
```

or with 1.0:
```make
BOARD_TAG = mega2560
include /usr/share/arduino/Arduino.mk
```

Babbleshack, 2016-08-30:
I cloned repo, and the example compiled, 

My bad! Ill see about updating the aur package.

closed 2016-08-30 by Babbleshack