sudar/Arduino-Makefile #454: Add support for ESP8266 Boards for Arduino IDE

xetra11, 2016-09-14:
Hey guys,

I'm sure somebody mentioned this already in a comment or something.
At the moment I'm using this sweet thingie here: https://github.com/esp8266/Arduino
However I love to use the Arduino Makefile here with my Emacs and really don't want to switch over to the Arduino IDE just for flashing the ESP.

Is there already a documented way to get those ESP Boards to run with this Makefile here?

sej7278, 2016-09-14:
i think someone asked within the SAM support issue, same reply really, its a bit too complicated with the entire toolchain and custom arduino libraries, platform etc.

i'd love to see a PR, but its beyond me. i tend to use arduino-builder for esp8266+arduino, but its not as nice as this makefile

xetra11, 2016-09-14:
Ah ok - meh I'd love to help but its beyond my skill-level.
@sej7278 you mentioned arduino-builder - are you talking about this makefile: https://github.com/plerup/makeEspArduino

sej7278, 2016-09-14:
no i mean [arduino-builder](https://github.com/arduino/arduino-builder) that comes with the ide

xetra11, 2016-09-14:
Ahhh ok - but this is only for compiling? No Uploading right?

sej7278, 2016-09-15:
i use this - for uploading and building, basically copied what the ide throws out:

```make
ARDUINO_PATH = $(HOME)/arduino-1.6.11
SKETCHBOOK   = $(HOME)/arduino16
ESPTOOL		 = $(SKETCHBOOK)/hardware/esp8266com/esp8266/tools/esptool/esptool
SKETCH 		 = $(notdir $(CURDIR)).ino
TARGET_DIR   = $(CURDIR)/build-esp8266
MONITOR_PORT = /dev/ttyUSB0

all:
	@ mkdir -p $(TARGET_DIR)

	$(ARDUINO_PATH)/arduino-builder -compile -logger=machine \
	-hardware "$(ARDUINO_PATH)/hardware" \
	-hardware "$(SKETCHBOOK)/hardware" \
	-tools "$(ARDUINO_PATH)/tools-builder" \
	-tools "$(ARDUINO_PATH)/hardware/tools/avr" \
	-built-in-libraries "$(ARDUINO_PATH)/libraries" \
	-libraries "$(SKETCHBOOK)/libraries" \
	-fqbn=esp8266com:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M \
	-ide-version=10611 \
	-build-path "$(TARGET_DIR)" \
	-warnings=none \
	-prefs=build.warn_data_percentage=75 \
	-verbose "$(SKETCH)"

flash:
	$(ESPTOOL) -v -cd nodemcu -cb 115200 -cp $(MONITOR_PORT) -ca 0x00000 -cf $(TARGET_DIR)/$(SKETCH).bin

upload: all flash

clean:
	rm -rf $(TARGET_DIR)

monitor:
	screen $(MONITOR_PORT) 9600
```

xetra11, 2016-09-17:
@sej7278 Gonna try this out in a few!

xetra11, 2016-09-17:
Works like a charm! Thank you a lot bro!

closed 2016-09-17 by xetra11