sudar/Arduino-Makefile #455: SoftwareSerial.h does not "make upload"

xetra11, 2016-09-14:
Sorry to bother you guys again but maybe you can help me out here.
I get Arduino Makefile for every library to work but for "SoftwareSerial".
I could not figure out what the problem was - also updated the Makefile to the newest release.

Here's the makelog:

`~/Workspace/Development/Arduino/ sudo make upload
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [AUTODETECTED]       ARDUINO_DIR = /usr/share/arduino 
- [COMPUTED]           ARDMK_DIR = /home/xetra11/Workspace/Tools/Arduino-Makefile-1.5 (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 105 
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            VENDOR = arduino 
- [DEFAULT]            ARDUINO_SKETCHBOOK =  
- [BUNDLED]            AVR_TOOLS_DIR = /usr/share/arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [USER]               ARDUINO_VAR_PATH = /home/xetra11/Workspace/Tools/arduino-1.6.8/hardware/arduino/avr/variants 
- [USER]               BOARDS_TXT = /home/xetra11/Workspace/Tools/arduino-1.6.8/hardware/arduino/avr/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [USER]               ARDUINO_CORE_PATH = /home/xetra11/Workspace/Tools/arduino-1.6.8/hardware/arduino/avr/cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD =  
- [DEFAULT]            CXXFLAGS_STD =  
- [COMPUTED]           DEVICE_PATH = /dev/ttyACM0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [SYSTEM]               SoftwareSerial
- [USER]               BOOTLOADER_PARENT = /home/xetra11/Workspace/Tools/arduino-1.6.8/hardware/arduino/avr/bootloaders 
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.2 (avr-gcc)
-------------------------
mkdir -p build-uno
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/home/xetra11/Workspace/Tools/arduino-1.6.8/hardware/arduino/avr/cores/arduino -I/home/xetra11/Workspace/Tools/arduino-1.6.8/hardware/arduino/avr/variants/standard    -I/usr/share/arduino/libraries/SoftwareSerial   -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  test.ino -o build-uno/test.o
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/home/xetra11/Workspace/Tools/arduino-1.6.8/hardware/arduino/avr/cores/arduino -I/home/xetra11/Workspace/Tools/arduino-1.6.8/hardware/arduino/avr/variants/standard    -I/usr/share/arduino/libraries/SoftwareSerial   -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  /usr/share/arduino/libraries/SoftwareSerial/SoftwareSerial.cpp -o build-uno/libs/SoftwareSerial/SoftwareSerial.o
/usr/share/arduino/libraries/SoftwareSerial/SoftwareSerial.cpp:27:18: fatal error: gpio.h: No such file or directory
 #include "gpio.h"
                  ^
compilation terminated.
make: *** [build-uno/libs/SoftwareSerial/SoftwareSerial.o] Error 1
`

Other libraries seem to have similar problems with other dependencies missing. While trying stuff out the "WiFi.h" library seems to be the only one working properly.

Using Arduino IDE yields no problems at all using the same sketches.

No Sketch Info here because the sketches only contain empty loop and setup. Tested if the source code does change the error message by any means = negative.

xetra11, 2016-09-15:
updated code format

sej7278, 2016-09-15:
can we see your Makefile, as ARDUINO_VERSION is being detected as 105 not 168, you've mixed your system-installed 1.0.5 in /usr/share with the 1.6.8 you've downloaded to your home directory.

you need to set ```ARDUINO_DIR=/home/xetra11/Workspace/Tools/arduino-1.6.8``` 

ARDUINO_VAR_PATH, BOARDS_TXT, BOOTLOADER_PARENT and ARDUINO_CORE_PATH shouldn't need to be set then

xetra11, 2016-09-15:
@sej7278 

Thanks for pointing that out!
Actually the ARDUINO_DIR was defined in my .zshrc as environment variable - I thought that would be ok. Adding it to the Makefile solved the issue.

I think I have to clean up my Arduino stuff because there are folders with contents everywhere...I'm a very bad linux user :(

closed 2016-09-15 by xetra11