sudar/Arduino-Makefile #456: program has been built without plugin support

SirRibeiro, 2016-09-18:
I get this error when a try to execute make ( sorry - this program has been built without plugin support ) :

`/git/Arduino-Makefile/examples/Blink$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/code/arduino 
- [USER]               ARDMK_DIR = /home/code/git/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 1710 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/code/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/code/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /home/code/arduino/hardware/tools/avr 
- [COMPUTED]           ARDUINO_LIB_PATH = /home/code/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/code/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/code/arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /home/code/arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/code/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /home/code/arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [COMPUTED]           DEVICE_PATH = /dev/ttyACM0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /home/code/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p build-uno
/home/code/arduino/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto Blink.ino -o build-uno/Blink.ino.o
/home/code/arduino/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects /home/code/arduino/hardware/arduino/avr/cores/arduino/hooks.c -o build-uno/core/hooks.c.o
/home/code/arduino/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects /home/code/arduino/hardware/arduino/avr/cores/arduino/WInterrupts.c -o build-uno/core/WInterrupts.c.o
/home/code/arduino/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects /home/code/arduino/hardware/arduino/avr/cores/arduino/wiring_analog.c -o build-uno/core/wiring_analog.c.o
/home/code/arduino/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects /home/code/arduino/hardware/arduino/avr/cores/arduino/wiring.c -o build-uno/core/wiring.c.o
/home/code/arduino/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects /home/code/arduino/hardware/arduino/avr/cores/arduino/wiring_digital.c -o build-uno/core/wiring_digital.c.o
/home/code/arduino/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects /home/code/arduino/hardware/arduino/avr/cores/arduino/wiring_pulse.c -o build-uno/core/wiring_pulse.c.o
/home/code/arduino/hardware/tools/avr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects /home/code/arduino/hardware/arduino/avr/cores/arduino/wiring_shift.c -o build-uno/core/wiring_shift.c.o
/home/code/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto /home/code/arduino/hardware/arduino/avr/cores/arduino/abi.cpp -o build-uno/core/abi.cpp.o
/home/code/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto /home/code/arduino/hardware/arduino/avr/cores/arduino/CDC.cpp -o build-uno/core/CDC.cpp.o
/home/code/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto /home/code/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp -o build-uno/core/HardwareSerial0.cpp.o
/home/code/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto /home/code/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial1.cpp -o build-uno/core/HardwareSerial1.cpp.o
/home/code/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1710 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/code/arduino/hardware/arduino/avr/cores/arduino -I/home/code/arduino/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto /home/code/arduino/hardware/arduino/avr/cores/arduino/HardwareSerial2.cpp -o  [... cut ...] d-uno/core/HardwareSerial2.cpp.o  build-uno/core/HardwareSerial3.cpp.o  build-uno/core/HardwareSerial.cpp.o  build-uno/core/HID.cpp.o  build-uno/core/IPAddress.cpp.o  build-uno/core/main.cpp.o  build-uno/core/new.cpp.o  build-uno/core/Print.cpp.o  build-uno/core/Stream.cpp.o  build-uno/core/Tone.cpp.o  build-uno/core/USBCore.cpp.o  build-uno/core/WMath.cpp.o  build-uno/core/WString.cpp.o         
sorry - this program has been built without plugin support
make: *** [build-uno/libcore.a] Error 1
`

sej7278, 2016-09-18:
we don't support the arduino.org (1.7+) IDE you're using there.

i suspect this is the recent LTO support causing the error. try removing the LTO options from CXX_FLAGS_STD and CFLAGS_STD or unsetting the variables or changing them to one from [here](https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#compilerexecutable-variables)

sej7278, 2016-09-25:
did you get anywhere with this?

gpancot, 2016-10-11:
same thing for me with Arduino 1.6.8
it works setting in Makefile:

CFLAGS_STD = -std=gnu11 -flto
AR_NAME      = avr-ar

sej7278, 2016-10-11:
@gpancot - they are the default settings in the latest Arduino.mk, so you don't need to set it in your Makefile.

is your avr-gcc newer than 4.8.0 ?

have you tried with ide 1.6.12 ?

gpancot, 2016-10-11:
```
# ### -------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /usr/share/arduino 
- [COMPUTED]           ARDMK_DIR = /home/zero/src/Arduino-Makefile-master (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 168 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/zero/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/zero/src/arduino (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /usr/share/arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/share/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/zero/src/arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = atmega328 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno-atmega328 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [USER]               CFLAGS_STD = -std=gnu11 -flto 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyUSB0 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------

```

sej7278, 2016-10-11:
can you unset CFLAGS_STD and see what that gives you? as you have avr-gcc 4.8.1 it should work even with ide 1.6.8

```
- [AUTODETECTED]       ARDUINO_VERSION = 168 
- [USER]               CFLAGS_STD = -std=gnu11 -flto 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
```

gpancot, 2016-10-11:
fail with:
```
/usr/share/arduino/hardware/tools/avr/bin/avr-ar rcs build-uno-atmega328/libcore.a  build-uno-atmega328/core/wiring.c.o  build-uno-atmega328/core/wiring_digital.c.o  build-uno-atmega328/core/WInterrupts.c.o  build-uno-atmega328/core/hooks.c.o  build-uno-atmega328/core/wiring_pulse.c.o  build-uno-atmega328/core/wiring_analog.c.o  build-uno-atmega328/core/wiring_shift.c.o  build-uno-atmega328/core/IPAddress.cpp.o  build-uno-atmega328/core/HardwareSerial0.cpp.o  build-uno-atmega328/core/WString.cpp.o  build-uno-atmega328/core/Tone.cpp.o  build-uno-atmega328/core/HardwareSerial1.cpp.o  build-uno-atmega328/core/WMath.cpp.o  build-uno-atmega328/core/Print.cpp.o  build-uno-atmega328/core/Stream.cpp.o  build-uno-atmega328/core/HardwareSerial2.cpp.o  build-uno-atmega328/core/USBCore.cpp.o  build-uno-atmega328/core/abi.cpp.o  build-uno-atmega328/core/new.cpp.o  build-uno-atmega328/core/PluggableUSB.cpp.o  build-uno-atmega328/core/main.cpp.o  build-uno-atmega328/core/HardwareSerial.cpp.o  build-uno-atmega328/core/CDC.cpp.o  build-uno-atmega328/core/HardwareSerial3.cpp.o  build-uno-atmega328/core/wiring_pulse.S.o
/usr/share/arduino/hardware/tools/avr/bin/avr-gcc -mmcu=atmega328p -Wl,--gc-sections -Os -flto -fuse-linker-plugin -o build-uno-atmega328/Blink.elf build-uno-atmega328/Blink.ino.o build-uno-atmega328/libcore.a  -lc -lm
/tmp/cc57bOjM.ltrans0.ltrans.o: In function `main':
cc57bOjM.ltrans0.o:(.text.startup+0x0): undefined reference to `init'
cc57bOjM.ltrans0.o:(.text.startup+0x8): undefined reference to `pinMode'
cc57bOjM.ltrans0.o:(.text.startup+0x14): undefined reference to `digitalWrite'
cc57bOjM.ltrans0.o:(.text.startup+0x20): undefined reference to `delay'
cc57bOjM.ltrans0.o:(.text.startup+0x28): undefined reference to `digitalWrite'
cc57bOjM.ltrans0.o:(.text.startup+0x34): undefined reference to `delay'
collect2: error: ld returned 1 exit status
../../Arduino.mk:1442: recipe for target 'build-uno-atmega328/Blink.elf' failed
make: *** [build-uno-atmega328/Blink.elf] Error 1         

complete make log attached
[make-log.zip](https://github.com/sudar/Arduino-Makefile/files/521783/make-log.zip)

```

sej7278, 2016-10-11:
ok so it is setting CFLAGS_STD properly without you having to set it.

you're also setting AR_NAME manually and incorrectly - with avr-gcc >4.8.0 you should be using avr-gcc-ar for the LTO support to work

so try again without setting either of those variables manually. why did you think that would fix things, when its actually breaking things?

gpancot, 2016-10-11:
make works only with:
```
CFLAGS_STD = -std=gnu11 -flto
AR_NAME = avr-ar
```
without it, fails with:
```
/usr/share/arduino/hardware/tools/avr/bin/avr-gcc-ar rcs build-uno-atmega328/libcore.a  build-uno-atmega328/core/wiring.c.o  build-uno-atmega328/core/wiring_digital.c.o  build-uno-atmega328/core/WInterrupts.c.o  build-uno-atmega328/core/hooks.c.o  build-uno-atmega328/core/wiring_pulse.c.o  build-uno-atmega328/core/wiring_analog.c.o  build-uno-atmega328/core/wiring_shift.c.o  build-uno-atmega328/core/IPAddress.cpp.o  build-uno-atmega328/core/HardwareSerial0.cpp.o  build-uno-atmega328/core/WString.cpp.o  build-uno-atmega328/core/Tone.cpp.o  build-uno-atmega328/core/HardwareSerial1.cpp.o  build-uno-atmega328/core/WMath.cpp.o  build-uno-atmega328/core/Print.cpp.o  build-uno-atmega328/core/Stream.cpp.o  build-uno-atmega328/core/HardwareSerial2.cpp.o  build-uno-atmega328/core/USBCore.cpp.o  build-uno-atmega328/core/abi.cpp.o  build-uno-atmega328/core/new.cpp.o  build-uno-atmega328/core/PluggableUSB.cpp.o  build-uno-atmega328/core/main.cpp.o  build-uno-atmega328/core/HardwareSerial.cpp.o  build-uno-atmega328/core/CDC.cpp.o  build-uno-atmega328/core/HardwareSerial3.cpp.o  build-uno-atmega328/core/wiring_pulse.S.o         
sorry - this program has been built without plugin support
../../Arduino.mk:1445: recipe for target 'build-uno-atmega328/libcore.a' failed
make: *** [build-uno-atmega328/libcore.a] Error 1
```

complete make log attached.
[make-log.zip](https://github.com/sudar/Arduino-Makefile/files/521858/make-log.zip)

sej7278, 2016-10-11:
no idea then, those two variables are set incorrectly, so should prevent it from building, not help it build.

where are you getting the IDE and arduino-mk from?

gpancot, 2016-10-11:
IDE come from www.arduino.cc
arduino-mk from https://github.com/sudar/Arduino-Makefile

thanks anyway ;)

sej7278, 2016-10-11:
ok i've reproduced it now - ide 1.6.8 seems to be the problem, perhaps there is more than just avr-gcc 4.8 required for LTO support

sej7278, 2016-10-11:
Seems to be a known issue with pre-1.6.10 IDE: https://github.com/arduino/Arduino/issues/5168

the problem is that on some linux distro's (or if you compiled it yourself) the toolchain supports LTO even if you use IDE 1.0.5, so we need to find a way to check if its supported using a better method than just checking if avr-gcc is newer than 4.8.0 but without simply checking the IDE version

sej7278, 2016-10-11:
my fix is to replace all occurrences of 480 with 490, so only avr-gcc 4.9.2 (not 4.8.1) will use LTO support:

```diff
-ifeq ($(shell expr $(CC_VERNUM) '>' 480), 1)
+ifeq ($(shell expr $(CC_VERNUM) '>' 490), 1)
```

gpancot, 2016-10-11:
yes, LTO is ok with IDE 1.6.12
with IDE 1.6.8 your fix is ok.

thank you so much

sej7278, 2016-10-11:
@gpancot thanks for testing, sorry about that, not sure how it slipped past me when I made the LTO additions.

I've made a PR #464 which i'll merge in a minute which fixes it for 1.6.8, 1.6.12 and 1.0.5 (etc.)

closed 2016-10-11 by sej7278

LRDPRDX, 2021-05-19:
I had the same issue on CentOS 7 and IDE 1.8.15. In order to fix this I had to set the `AVR_TOOLS_DIR` to the one in the IDE's dir : `ARDUINO_DIR/hardware/tools/avr`.