sudar/Arduino-Makefile #457: error "not declared in this scope" when building the Bridge sketch

artynet, 2016-09-21:
Hello folks, any idea about this error ?

```
Bridge.ino: In function 'void loop()':
Bridge.ino:53:19: error: 'process' was not declared in this scope
     process(client);
                   ^
Bridge.ino: In function 'void process(YunClient)':
Bridge.ino:68:26: error: 'digitalCommand' was not declared in this scope
     digitalCommand(client);
                          ^
Bridge.ino:73:25: error: 'analogCommand' was not declared in this scope
     analogCommand(client);
                         ^
Bridge.ino:78:23: error: 'modeCommand' was not declared in this scope
     modeCommand(client);
                       ^
make: *** [build-yun/Bridge.ino.o] Errore 1
```

I get it when I try buld the Build sketch for my Arduino Yun. Should I include other search paths or add specific options to the compilers ? Thank you in advance.

sej7278, 2016-09-21:
the yun is the intel box isn't it? i'd be surprised if it worked for with this makefile which is mainly for AVR chips.

generally the scope declaration issues are caused by you calling functions from setup() or loop() that aren't declared **above** setup() or loop() in the .ino file. the ide lets you get away with this bad c++ but not the makefile.

basically you need this sort of order:

```cpp
void myfunction() {}

void setup(void) {}

void loop() {
    myfunction();
}
```

sej7278, 2016-09-25:
did you get this working?

kishalmi, 2017-01-25:
and just in case you run into a scenario where two functions need to call each other, so there is no obvious order to put them in, you need a so called forward decleration. looks like this:
`
// forward declaration of f2:
f2();

// two interlinked functions:
void f1() {
	f2();
}
void f2() {
	f1();
}
`

this - in its simplicity - might be a silly example, 
but think of ISRs where you'd attach a listener to a rising edge in the listener of the falling one and vice versa.

sej7278, 2017-03-19:
did forward declarations/re-ordering fix the issue, would be interested to know if the yun even worked.

closed 2017-04-13 by sej7278