sudar/Arduino-Makefile #458: Delay causes Arduino Ethernet (Board) to hang

nathansizemore, 2016-09-21:
**main.cpp**

``` cpp
#include <Arduino.h>
int main(void)
{
    Serial.println("asdf");
    delay(1);
    Serial.println("asdf");
    return 0;
}
```

**Makefile**

``` makefile
### DISCLAIMER
### This is an example Makefile and it MUST be configured to suit your needs.
### For detailed explanations about all of the available options, please refer
### to https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md
### Original project where this Makefile comes from: https://github.com/WeAreLeka/Bare-Arduino-Project

### PROJECT_DIR
### This is the path to where you have created/cloned your project
PROJECT_DIR       = ${HOME}/development/mtsi-19391-sensors

### ARDMK_DIR
### Path to the Arduino-Makefile directory.
ARDMK_DIR         = $(HOME)/development/arduino-mk

### ARDUINO_DIR
### Path to the Arduino application and resources directory.
ARDUINO_DIR       = /Applications/Arduino.app/Contents/Java

### USER_LIB_PATH
### Path to where the your project's libraries are stored.
USER_LIB_PATH    :=  $(PROJECT_DIR)/lib

### BOARD_TAG
### It must be set to the board you are currently using. (i.e uno, mega2560, etc.)
BOARD_TAG         = ethernet

### MCU_FLAG_NAME
### MCU description
MCU_FLAG_NAME     = mmcu

### MONITOR_BAUDRATE
### It must be set to Serial baudrate value you are using.
MONITOR_BAUDRATE  = 9600

### MONITOR_PORT
### The port your board is connected to. Using an '*' tries all the ports and finds the right one.
MONITOR_PORT	  = /dev/cu.usbserial-AL029Z8N

### FORCE_MONITOR_PORT
### I dunno what this does, maybe fixes things?
FORCE_MONITOR_PORT = true

### AVR_TOOLS_DIR
### Path to the AVR tools directory such as avr-gcc, avr-g++, etc.
### On OS X with `homebrew`:
AVR_TOOLS_DIR     = ${HOME}/development/tool-chains/gnu/avr

### AVRDUDE
### Path to avrdude directory.
### On OS X with `homebrew`:
AVRDUDE          = /usr/local/bin/avrdude

### CFLAGS_STD
### Set the C standard to be used during compilation.
CFLAGS_STD        = -std=c11

### CXXFLAGS_STD
### Set the C++ standard to be used during compilation.
CXXFLAGS_STD      = -std=c++14

### CXXFLAGS
### Flags you might want to set for debugging purpose. Comment to stop.
CXXFLAGS         += -pedantic -Wextra -Werror
CXXFLAGS	 += -Wno-error=deprecated-declarations
CXXFLAGS	 += -Wno-error=sized-deallocation

### CURRENT_DIR
### Do not touch - used for binaries path
CURRENT_DIR       = $(shell basename $(CURDIR))

### OBJDIR
### This is where you put the binaries you just compile using 'make'
OBJDIR            = $(PROJECT_DIR)/out/bin/$(BOARD_TAG)/$(CURRENT_DIR)

### Do not touch - the path to Arduino.mk, inside the ARDMK_DIR
include $(ARDMK_DIR)/Arduino.mk
```

```
$ make
-----------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Java
- [USER]               ARDMK_DIR = /Users/nathan/development/arduino-mk
- [AUTODETECTED]       ARDUINO_VERSION = 1611
- [DEFAULT]            ARCHITECTURE = avr
- [DEFAULT]            ARDMK_VENDOR = arduino
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/nathan/Library/Arduino15/preferences.txt
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/nathan/Documents/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /Users/nathan/development/tool-chains/gnu/avr
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [USER]               USER_LIB_PATH = /Users/nathan/development/mtsi-19391-sensors/lib
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = ethernet
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = ethernet (from build.variant)
- [USER]               OBJDIR = /Users/nathan/development/mtsi-19391-sensors/out/bin/ethernet/src
- [COMPUTED]           ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
- [USER]               MONITOR_BAUDRATE = 9600
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [USER]               MCU_FLAG_NAME = mmcu
- [USER]               CFLAGS_STD = -std=c11
- [USER]               CXXFLAGS_STD = -std=c++14
- [COMPUTED]           DEVICE_PATH = /dev/cu.usbserial-AL029Z8N (from MONITOR_PORT)
- [USER]               FORCE_MONITOR_PORT = true
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5
- [COMPUTED]           CC_VERSION = 6.1.0 (avr-gcc)
```

nathansizemore, 2016-09-22:
Digging in a little bit further, Serial also does not work as expected. When I mention works, and doesn't work below, it is regarding both delay and Serial.print.

I have three AVR toochains installed. In these examples, we'll refer to them as `arduino-avr`, `avr-4.9` and `avr-6.1`. The number represent the version of GCC compiled. 4.9 and 6.1 were built by myself, and `arduino-4.9` is the one packaged in the Arduino IDE download.

Some flags will need changed when jumping from compiler version, so probably best to just disable all flags in makefile when testing.

**`avr-6.1`**
`main.cpp` and `avr-6.1` combination does not work.

`main.ino` and `avr-6.1` with setup/loop functions works.

`main.cpp` and `avr-6.1` with setup/loop functions works.

**`avr-4.9`**
`main.cpp` and `avr-4.9` combination does not work.

`main.ino` and `avr-4.9` with setup/loop functions works.

`main.cpp` and `avr-4.9` with setup/loop functions works.

**`arduino-avr`**
`main.cpp` and `arduino-avr` combination does not work. Link time error: missing setup/loop functions.

`main.ino` and `arduino-avr` with setup/loop functions works.

`main.cpp` and `arduino-avr` with setup/loop functions works.

nathansizemore, 2016-09-22:
Found out through #459 that this problem is probably more due to [Bare-Arduino-Project](https://github.com/ladislas/Bare-Arduino-Project) variables than the auto selection used with this makefile.

sej7278, 2016-09-25:
does this work with a simpler makefile then? can we close the issue or move it to bare-arduino-project?

nathansizemore, 2016-09-27:
Sorry, yes.

closed 2016-09-27 by nathansizemore

ladislas, 2016-09-28:
Okay, I found the error.

First off, you need to `init()` when just using `main()`. Second, your delay was too short between the past print and the return.

So here you go:

```cpp
#include <Arduino.h>

int main(void) {

	init();

	Serial.begin(115200);
	delay(100);

	Serial.println("asdf1");
	delay(10);
	Serial.println("asdf2");
	delay(10);

	return 0;
}
```

sej7278, 2016-09-28:
oh yes if you're not using setup() or loop() you need to call init() and a for loop, see https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/main.cpp#L33

nathansizemore, 2016-09-28:
Might be worth mentioning that some where in the docs here

sej7278, 2016-09-28:
its not hugely relevant to arduino-makefile, perhaps moreso to bare-arduino-project though