sudar/Arduino-Makefile #459: Undefined reference to __cxa_guard_acquire and __cxa_guard_release

jonblack, 2016-09-22:
Using the JsonArduino library with Arduino-Makefile, I get the following error during the linker phase when building one of the libraries examples:

```
build-mega-atmega2560/example.ino.o: In function `ArduinoJson::JsonArray::invalid()':
example.ino:(.text._ZN11ArduinoJson9JsonArray7invalidEv[ArduinoJson::JsonArray::invalid()]+0xc): undefined reference to `__cxa_guard_acquire'
example.ino:(.text._ZN11ArduinoJson9JsonArray7invalidEv[ArduinoJson::JsonArray::invalid()]+0x28): undefined reference to `__cxa_guard_release'
build-mega-atmega2560/example.ino.o: In function `ArduinoJson::JsonObject::invalid()':
example.ino:(.text._ZN11ArduinoJson10JsonObject7invalidEv[ArduinoJson::JsonObject::invalid()]+0xc): undefined reference to `__cxa_guard_acquire'
example.ino:(.text._ZN11ArduinoJson10JsonObject7invalidEv[ArduinoJson::JsonObject::invalid()]+0x28): undefined reference to `__cxa_guard_release'
collect2: error: ld returned 1 exit status
/home/jon/.arduino_mk/Arduino.mk:1416: recipe for target 'build-mega-atmega2560/arduino-example.elf' failed
make: *** [build-mega-atmega2560/arduino-example.elf] Error 1
```

Using the arduino ide I don't get this problem. I first raised the issue with ArduinoJson here: https://github.com/bblanchon/ArduinoJson/issues/356.

sej7278, 2016-09-22:
works fine for me with 1.0.5 and 1.6.11 with these simple Makefiles

```make
ARDUINO_DIR = $(HOME)/arduino-1.6.11
BOARD_TAG = mega
BOARD_SUB = atmega2560
MONITOR_PORT = /dev/ttyACM0
include /usr/share/arduino/Arduino.mk
```

```
BOARD_TAG = mega2560
MONITOR_PORT = /dev/ttyACM0
include /usr/share/arduino/Arduino.mk
```

need more info from you - your Makefile, platform, Arduino.mk Configuration output etc.

nathansizemore, 2016-09-22:
I just received the same error, but for Ethernet lib usage.

**Makefile**

``` makefile
OS		 := $(shell uname)

### PROJECT_DIR
### This is the path to where you have created/cloned your project
PROJECT_DIR       = ${HOME}/development/mtsi-19391-sensors

### ARDMK_DIR
### Path to the Arduino-Makefile directory.
ARDMK_DIR         = $(HOME)/development/arduino-mk

### ARDUINO_DIR
### Path to the Arduino application and resources directory.
ARDUINO_DIR       = /Applications/Arduino.app/Contents/Java
ifeq ($(OS),Linux)
ARDUINO_DIR       = $(HOME)/arduino-1.6.12
endif

### USER_LIB_PATH
### Path to where the your project's libraries are stored.
USER_LIB_PATH    :=  $(PROJECT_DIR)/lib

### BOARD_TAG
### It must be set to the board you are currently using. (i.e uno, mega2560, etc.)
BOARD_TAG         = ethernet

### MCU_FLAG_NAME
### MCU description
MCU_FLAG_NAME     = mmcu

### MONITOR_BAUDRATE
### It must be set to Serial baudrate value you are using.
MONITOR_BAUDRATE  = 9600

### MONITOR_PORT
### The port your board is connected to. Using an '*' tries all the ports and finds the right one.
MONITOR_PORT	  = /dev/cu.usbserial-AL029Z8N

### AVR_TOOLS_DIR
### Path to the AVR tools directory such as avr-gcc, avr-g++, etc.
AVR_TOOLS_DIR     = /usr/local

### AVRDUDE
### Path to avrdude directory.
### On OS X with `homebrew`:
AVRDUDE          = /usr/local/bin/avrdude
ifeq ($(OS),Linux)
AVRDUDE          = /usr/bin/avrdude
endif

### CFLAGS_STD
### Set the C standard to be used during compilation.
CFLAGS_STD        = -std=c11

### CXXFLAGS_STD
### Set the C++ standard to be used during compilation.
CXXFLAGS_STD      = -std=c++14

### CURRENT_DIR
### Do not touch - used for binaries path
CURRENT_DIR       = $(shell basename $(CURDIR))

### OBJDIR
### This is where you put the binaries you just compile using 'make'
OBJDIR            = $(PROJECT_DIR)/out/bin/$(BOARD_TAG)/$(CURRENT_DIR)

### Do not touch - the path to Arduino.mk, inside the ARDMK_DIR
include $(ARDMK_DIR)/Arduino.mk

```

**Error**
```
libcore.a(Ethernet.cpp.o): In function `EthernetClass::begin(unsigned char*, unsigned long, unsigned long)':
Ethernet.cpp:(.text._ZN13EthernetClass5beginEPhmm+0x46): undefined reference to `__cxa_guard_acquire'
Ethernet.cpp:(.text._ZN13EthernetClass5beginEPhmm+0x5a): undefined reference to `__cxa_guard_release'
collect2: error: ld returned 1 exit status
```

nathansizemore, 2016-09-22:
Seems to be related to static initialization and what boards support it where.

sej7278, 2016-09-22:
these look like [bare-arduino-project](https://github.com/sudar/Arduino-Makefile#bare-arduinoproject) makefiles, so you should raise the issue there, its not up-to-date with arduino-mk at all.

nathansizemore, 2016-09-22:
The makefile I posted above came out of this repo [here](https://github.com/sudar/Arduino-Makefile/blob/master/examples/MakefileExample/Makefile-example.mk).

sej7278, 2016-09-22:
ah yes that makefile is way overcomplicated. in most cases you won't need more than:

```make
ARDUINO_DIR = /path/to/arduino-1.6.11
BOARD_TAG = mega
BOARD_SUB = atmega2560
MONITOR_PORT = /dev/ttyACM0
include /path/to/Arduino.mk
```

jonblack, 2016-09-23:
I think I've figured it out. I have the following in my Makefile:

    CXXFLAGS_STD += -std=gnu++11

Removing this gets rid of the linker error. @nathansizemore, this is also true for the Ethernet error as well.

sej7278, 2016-09-23:
```CXXFLAGS_STD      = -std=c++14``` and ```CXXFLAGS_STD += -std=gnu++11``` is going to be a mess yes.

but have you tried a much simpler makefile like i posted, most of the other stuff should be auto-detected, and setting it manually you may be setting it incorrectly.

certainly Ethernet (and SPI) and Json works fine for me with the makefile i posted above.

jonblack, 2016-09-25:
Sorry my message wasn't very clear. By removing the `CXXFLAGS_STD` line I had, my Makefile is the same as yours.

sej7278, 2016-09-25:
so can we close this then?

closed 2016-09-27 by jonblack