sudar/Arduino-Makefile #460: undefined reference errors while building elf object

devjagadish, 2016-09-26:
I have a project in which I am trying to use avrnacl lib, which was complied as a object archive file. If I name my file as .c all works fine, after I changed it to .cpp file for Serial i/o I get the following errors. I wonder how linker was able to find my archived lib when it has to deal with c file but not with cpp file. Please help me in understanding what's going wrong here, thanks.

build-uno/print.c.o
build-uno/randombytes.c.o
build-uno/Blink.cpp.o
(build-uno/libcore.a)wiring.c.o
(build-uno/libcore.a)wiring_digital.c.o
(build-uno/libcore.a)HardwareSerial.cpp.o
(build-uno/libcore.a)HardwareSerial0.cpp.o
(build-uno/libcore.a)Print.cpp.o
(build-uno/libcore.a)abi.cpp.o
(build-uno/libcore.a)main.cpp.o
(build-uno/libcore.a)hooks.c.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/../../../../avr/lib/avr5/libc.a)exit.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/avr5/libgcc.a)_exit.o
/var/folders/wj/k_j90nx15nx7fql_pb8fg1gr0000gr/T//cc0OGXFD.ltrans0.ltrans.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/../../../../avr/lib/avr5/libc.a)calloc.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/../../../../avr/lib/avr5/libc.a)malloc.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/../../../../avr/lib/avr5/libc.a)abort.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/../../../../avr/lib/avr5/libc.a)memset.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/avr5/libgcc.a)_copy_data.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/avr5/libgcc.a)_clear_bss.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/avr5/libgcc.a)_ctors.o
(/usr/local/Cellar/avr-gcc/4.9.3/lib/gcc/avr/4.9.3/avr5/libgcc.a)_tablejump2.o
/var/folders/wj/k_j90nx15nx7fql_pb8fg1gr0000gr/T//cc0OGXFD.ltrans0.ltrans.o: In function HardwareSerial::_tx_udr_empty_irq()': /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:95: undefined reference torandombytes(unsigned char, unsigned long long)'
/var/folders/wj/k_j90nx15nx7fql_pb8fg1gr0000gr/T//cc0OGXFD.ltrans0.ltrans.o: In function HardwareSerial::write(unsigned char)': /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:214: undefined reference torandombytes(unsigned char, unsigned long long)'
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:225: undefined reference to `crypto_secretbox_xsalsa20poly1305(unsigned char, unsigned char const, unsigned int, unsigned char const, unsigned char const)'

sej7278, 2016-09-26:
not sure what you mean there - you can't just rename a .c file to a .cpp file, also do you mean you have a prebuild .a binary file to link into?

need to see your Makefile and the arduino-mk output really.

devjagadish, 2016-09-27:
Hi sej7278,
Yes I have a prebuilt .a binary.

Here is my Makefile content

ARDUINO_DIR   = /Applications/Arduino.app/Contents/Java
ARDMK_DIR     = /Users/jagdish/Development/arduino/Arduino-Makefile
AVR_TOOLS_DIR = /usr/local
MONITOR_PORT = /dev/tty.usbmodem1411
BOARD_TAG = uno
ISP_PROG	   = arduino
ISP_PORT	   = /dev/tty.usbmodem1411
include ../Arduino.mk

And the complete output of arduino-mk

-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Java 
- [USER]               ARDMK_DIR = /Users/jagdish/Development/arduino/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 1611 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/jagdish/Library/Arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/jagdish/Documents/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr/local 
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /Users/jagdish/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [COMPUTED]           DEVICE_PATH = /dev/tty.usbmodem1411 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.9.3 (avr-gcc)
-------------------------
mkdir -p build-uno
/usr/local/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1611 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects print.c -o build-uno/print.c.o
/usr/local/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1611 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects randombytes.c -o build-uno/randombytes.c.o
/usr/local/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1611 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto Blink.cpp -o build-uno/Blink.cpp.o
/usr/local/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1611 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c -o build-uno/core/WInterrupts.c.o
/usr/local/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1611 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 -flto -fno-fat-lto-objects /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/hooks.c -o build-uno/core/hooks.c.o
/usr/local/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=1611 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduin [... cut ...]
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:214: undefined reference to `randombytes(unsigned char*, unsigned long long)'
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp:225: undefined reference to `crypto_secretbox_xsalsa20poly1305(unsigned char*, unsigned char const*, unsigned int, unsigned char const*, unsigned char const*)'
collect2: error: ld returned 1 exit status
make: *** [build-uno/GrayLock.elf] Error 1

sej7278, 2016-09-27:
i've no idea really, but i don't think it magically finds your .a binary, you'd have to tell it to link it somehow, its not even picking it up as an arduino lib there.

maybe try renaming Blink.cpp to Blink.ino, although i doubt that will do much, i suspect it was just silently failing when it was Blink.c

devjagadish, 2016-09-27:
Sorry for not mentioning that I have added libnacl.a to the CC command in Aurduino.mk file as shown below.

$(TARGET_ELF): 	$(LOCAL_OBJS) $(CORE_LIB) $(OTHER_OBJS)
		$(CC) $(LDFLAGS) -o $@ $(LOCAL_OBJS) $(CORE_LIB) $(OTHER_OBJS) $(USER_LIB_PATH)/libnacl.a -lc -lm $(LINKER_SCRIPTS)

When my file was a .c, it got compiled and flashed to the board successfully and on board LED blinks as programmed. I am new to Arduino and Makefiles, not able to understand whats going wrong.

sej7278, 2016-09-27:
well it seems your sketch is just a c file then, and i think what you could do is just add this to your Makefile instead of arduino.mk

```OTHER_OBJS = $(USER_LIB_PATH)/libnacl.a```

devjagadish, 2016-09-27:
Hi sej7278,

It didn't help, my project folder structure looks like this

├── Blink.cpp
├── Makefile
├── avrnacl.h
├── libraries
│   └── libnacl.a
├── print.c
├── print.h
├── randombytes.c
└── randombytes.h

I have attached my sample project, you can see that simply changing the file name from .cpp to .c it works.
[Blinker.zip](https://github.com/sudar/Arduino-Makefile/files/495775/Blinker.zip)

sej7278, 2016-09-27:
ok sure, renaming to .c works, so what's the problem, it **is** a C file not a C++ one, you're getting errors when trying to compile it as C++ as you'd need to use iostream or the string library for 'char' in C++, neither of which arduino has. you get the same error from the IDE if you rename it to Blink.ino

This Makefile builds Blink.c for me:

```make
ARDUINO_DIR = $(HOME)/arduino-1.6.11
BOARD_TAG = uno
USER_LIB_PATH = ./libraries
ARDUINO_LIBS = SoftwareSerial
OTHER_OBJS = $(USER_LIB_PATH)/libnacl.a

include /usr/share/arduino/Arduino.mk
```

devjagadish, 2016-09-27:
Thanks for your time, I wanted to use Serial class to print some messages from the board. Serial is Arduino core class. Without adding the nacl library .ino or .cpp file with Serial usage works fine. Not sure how can it cause problem when using external libraries. Can you please suggest any other way to solve this problem?

sej7278, 2016-09-27:
ah i understand now, yes HardwareSerial.h is a C++ class so can't be used from C (you don't want SoftwareSerial, but that wouldn't work either).

sorry i've got no solution to that, you'd have to use ```extern "C"``` and write a bunch of interfaces between the c++ methods and c, yuk! or find a c++ version of your library....

devjagadish, 2016-09-27:
Thanks, will check if a c++ version of the library exists.

closed 2016-09-28 by devjagadish