sudar/Arduino-Makefile #461: boards.txt file is not parsed properly

imranmunshi, 2016-10-03:
I was trying to build the edtracker 9150 code using their custom core. The boards.txt has edtracker9150.build.variant=edtracker:edtracker2-9150

But that is not parsed properly by the makefile due to which I have to add VARIANT manually in the project Makefile.

sej7278, 2016-10-03:
looks like the edtracker is using a subdirectory for some reason

imranmunshi, 2016-10-04:
Arduino IDE 1.6.11-2 (arduino-bin PKGBUILD from AUR, Arch Linux) parses it properly and gives no issues.

sej7278, 2016-10-11:
can you give us a link as to where you got the edtracker core, also the directory structure within your hardware directory would be useful i.e. is it ~/sketchbook/hardware/edtracker/edtracker2-9150/ ?

the only core i could find uses ~/sketchbook/hardware/edtracker/avr/variants/edtracker2/ which is totally wrong

sej7278, 2016-10-11:
this seems to work for me - basically if there is a colon, set $VARIANT to the value on the right of the colon. doesn't break cores without colons e.g. arduino 1.0.5 with an uno:

```diff
diff --git a/Arduino.mk b/Arduino.mk
index 133a9a2..54f4920 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -587,9 +587,9 @@ ifeq ($(strip $(NO_CORE)),)
 
     # Which variant ? This affects the include path
     ifndef VARIANT
-        VARIANT := $(call PARSE_BOARD,$(BOARD_TAG),menu.(chip|cpu).$(BOARD_SUB).build.variant)
+        VARIANT := $(shell echo $(call PARSE_BOARD,$(BOARD_TAG),menu.(chip|cpu).$(BOARD_SUB).build.variant) | cut -d : -f 2)
         ifndef VARIANT
-            VARIANT := $(call PARSE_BOARD,$(BOARD_TAG),build.variant)
+            VARIANT = $(shell echo $(call PARSE_BOARD,$(BOARD_TAG),build.variant) | cut -d : -f 2)
         endif
         $(call show_config_variable,VARIANT,[COMPUTED],(from build.variant))
     else
```

sej7278, 2016-10-11:
I think changing PARSE_BOARD is better:

```diff
diff --git a/Arduino.mk b/Arduino.mk
index 133a9a2..1c0debb 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -568,7 +568,7 @@ endif
 
 ifndef PARSE_BOARD
     # result = $(call READ_BOARD_TXT, 'boardname', 'parameter')
-    PARSE_BOARD = $(shell grep -Ev '^\#' $(BOARDS_TXT) | grep -E "^[ \t]*$(1).$(2)=" | cut -d = -f 2)
+    PARSE_BOARD = $(shell grep -Ev '^\#' $(BOARDS_TXT) | grep -E "^[ \t]*$(1).$(2)=" | cut -d = -f 2 | cut -d : -f 2)
 endif
 
 # If NO_CORE is set, then we don't have to parse boards.txt file
@@ -579,7 +579,7 @@ ifeq ($(strip $(NO_CORE)),)
     # 'robot', but can also hold 'tiny', for example, if using
     # https://code.google.com/p/arduino-tiny alternate core.
     ifndef CORE
-        CORE = $(shell echo $(call PARSE_BOARD,$(BOARD_TAG),build.core) | cut -d : -f 2)
+        CORE = $(call PARSE_BOARD,$(BOARD_TAG),build.core)
         $(call show_config_variable,CORE,[COMPUTED],(from build.core))
     else
         $(call show_config_variable,CORE,[USER])
```

sej7278, 2016-10-11:
still doesn't seem to fix weird cases like the adafruit metro board that uses its own core but the default arduino variants e.g. 

```
metro.build.variant=arduino:standard
metro.build.core=arduino:arduino

- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
```

it computes it properly, but looks in the wrong place (in user hardware dir not system arduino dir) although it could be a bug with the adafruit cores.

closed 2016-10-11 by sej7278