sudar/Arduino-Makefile #462: AVR_TOOLS_DIR variable is ignored

imranmunshi, 2016-10-03:
I am on Arch Linux and using arduino-mk from AUR.

I have installed avr-gcc, avr-binutils and avr-libc (ver. 6.20) and to use that I set AVR_TOOLS_DIR to /usr but it still shows /usr/share/arduino/..... (the default calculated value) in the make output and the same is used. I want to use the latest avr-gcc toolchain

ladislas, 2016-10-03:
can we see the makefile? maybe you added AVR_TOOLS_DIR after the include at the end.

imranmunshi, 2016-10-03:
Here it is:

ALTERNATE_CORE = edtracker
BOARD_TAG = edtracker9150
VARIANT = edtracker2-9150
ARDUINO_PLATFORM_LIB_PATH = ${HOME}/Arduino/hardware/edtracker/avr/libraries
ARDUINO_LIBS = Wire EEPROM I2Cdev InvensenseMotionDriver Adafruit-PWM-Servo-Driver-Library
AVR_TOOLS_DIR = /usr
include /usr/share/arduino/Arduino.mk
ALTERNATE_CORE = edtracker
BOARD_TAG = edtracker9150
VARIANT = edtracker2-9150
ARDUINO_PLATFORM_LIB_PATH = ${HOME}/Arduino/hardware/edtracker/avr/libraries
ARDUINO_LIBS = Wire EEPROM I2Cdev InvensenseMotionDriver Adafruit-PWM-Servo-Driver-Library
AVR_TOOLS_DIR = /usr
include /usr/share/arduino/Arduino.mk

ladislas, 2016-10-03:
remove the first  include /usr/share/arduino/Arduino.mk

imranmunshi, 2016-10-03:
Already tried that. Here is what I get
 [USER]               AVR_TOOLS_DIR = /usr/share/arduino/hardware/tools/avr

ladislas, 2016-10-03:
are you sure that what you posted is your **actual** Makefile? also why do you want to use avr-gcc 6.2?

try this:

```
ALTERNATE_CORE = edtracker
BOARD_TAG = edtracker9150
VARIANT = edtracker2-9150
ARDUINO_PLATFORM_LIB_PATH = ${HOME}/Arduino/hardware/edtracker/avr/libraries
ARDUINO_LIBS = Wire EEPROM I2Cdev InvensenseMotionDriver Adafruit-PWM-Servo-Driver-Library
AVR_TOOLS_DIR = /usr/local
include /usr/share/arduino/Arduino.mk
```

imranmunshi, 2016-10-03:
Yes, that is exactly what I have.

I just want to try avr-gcc-6.2.0. I have build failure when using standard board tags like uno or micro (I will report it in an other issue)

imranmunshi, 2016-10-03:
Does it have something to do with ARDMK_VERSION? Mine is 1.5

ladislas, 2016-10-03:
try compiling a blink example with avr-gcc 6.2 and let me know if it works

imranmunshi, 2016-10-03:
I tried with the latest pull from github and it works. It detects the AVR_TOOLS_DIR and uses avr-gcc-6.2.0 properly, although avr-gcc segfaults (lto1: internal compiler error: Seg fault)

imranmunshi, 2016-10-03:
I guess this issue can be closed

closed 2016-10-03 by imranmunshi