sudar/Arduino-Makefile #463: ARDUINO_PLATFORM_LIB_PATH not set properly for custom cores (ALTERNATE_CORE)

imranmunshi, 2016-10-04:
When setting ALTERNATE_CORE to something that is present in $ARDUINO_SKETCHBOOK/hardware, the ARDUINO_PLATFORM_LIB_PATH should be set to $ALTERNATE_CORE_PATH/libraries but it gets set (computed) to $ARDUINO_DIR/hardware/$VENDOR/$ARCHITECTURE/libraries, which is incorrect.

sej7278, 2016-10-11:
do you have a link to documentation that backs that up?

imranmunshi, 2016-10-17:
The documentation for this available here [https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification](url)

imranmunshi, 2016-10-17:
Moreover, if the core is placed in `${ARDUINO_SKETCHBOOK}/hardware` i.e. `ALTERNATE_CORE_PATH=${ARDUINO_SKETCHBOOK}/hardware/${VENDOR}/${ARCHITECTURE}` then it is obvious that the related platform include libraries would be in `${ALTERNATE_CORE_PATH}/libraries`

sej7278, 2016-10-17:
i'm not convinced that ARDUINO_PLATFORM_LIB_PATH is what you think it is, i believe its the libraries for the platform/arch (avr, sam21 etc.) not the core (pinout etc.). the documentation doesn't really cover it, which is why i asked where you got your idea from.

imranmunshi, 2016-10-17:
Yes, you're correct that it is not documented somewhere, but having a look at the third party hardwares and custom cores, they do have libraries in that location where I pointed out.

We can scan that location first

sej7278, 2016-10-17:
yes it seems that's what the esp8266 arduino core does with the libraries it requires to override the system ones, not sure how to implement things though, probably a new variable or some sort of duplicate checking between user hardware libs and system libs

imranmunshi, 2016-10-20:
Does it still need more details? Because I guess if a core already has it's own libraries folder then ARDUINO_PLATFORM_LIB_PATH should be set to that, if not then scan in $ARDUINO_DIR

sej7278, 2016-10-20:
i don't know really, if you want to do PR we can have a look and figure out some test cases, or do you have a test that works if you manually override ```ARDUINO_PLATFORM_LIB_PATH``` in your Makefile?

i've never really come across a core that has its own libraries other than esp8266 which doesn't work with arduino-mk anyway. also the variable is specific to ide 1.6

imranmunshi, 2016-10-20:
That makes this reported issue more of a 'feature request' than a bug

sej7278, 2016-10-20:
well the label's not important. do you have a core that works or doesn't work based on ```ARDUINO_PLATFORM_LIB_PATH``` so we can test a fix?

imranmunshi, 2016-10-20:
Yes, the edtracker core does not work now. Only way to make it work is to set ARDUINO_PLATFORM_LIB_PATH according to what I said

sej7278, 2016-10-20:
ah, i've figured out what ```ARDUINO_PLATFORM_LIB_PATH``` does when not using an alternate core - it provides libs like ```$(ARDUINO_DIR)/hardware/arduino/avr/libraries/SPI/src/SPI.h```

the problem is, if you set ```ARDUINO_PLATFORM_LIB_PATH``` to ```$(SKETCHBOOK)/hardware/edtracker/avr/libraries``` then SPI.h never gets found

so any workaround here needs more logic than just changing ```ARDUINO_PLATFORM_LIB_PATH = $(ARDUINO_DIR)/hardware/$(ARDMK_VENDOR)/$(ARCHITECTURE)/libraries``` it needs to look in both places and filter out duplicates in favour of the user ones instead of the system ones.

celem, 2017-02-01:
I think that I have stumbled on the same bug when trying to compile the attiny85 code at the link below with Makefile. Arduino IDE works fine. The makefile fails to find the avr includes.
https://arduinoelectronics.wordpress.com/2014/01/06/ultra-low-power-led-flasher-with-attiny/

sej7278, 2017-02-01:
no that's just bad code @celem - you need to move setup_watchdog() and system_sleep() before setup() and it builds fine.

celem, 2017-02-01:
Thanks sej7278 - I neglected to consider compilation order - the IDE numbs the brain - use it too much and you get lazy and forgetful, which is one reason that Makefiles are important -closer to the metal.

sej7278, 2017-02-01:
"not defined in this scope" is the key message, it basically means you're calling a function that another function doesn't even know about yet.