sudar/Arduino-Makefile #466: avr-gcc 6.2.0

tuna-f1sh, 2016-10-14:
Has Arduino-Makefile been tested with avr-gcc 6.2.0 or is anyone successfully using? 

I'm on macOS and using Homebrew and recently updated to 6.2.0. I'm not sure of the changes but seems to be a double major version jump! The build fails to find stdlib.h. A quick look into the 6.2.0 directory and the structure is completely different to 4.9.3 and 'avr' was not in the root directory, nor could I find 'stdlib.h'. However, a symbolic link still exists in my '/usr/local/include' to '/usr/local/Cellar/avr-gcc/4.9.3/avr'. This might be more an issue for the avr-gcc tap but I know @ladislas contributes to both so might be able to shed some light.

I've had to roll back to 4.9.3 for now. Build output below:

```
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Java
- [USER]               ARDMK_DIR = /usr/local/Cellar/arduino-mk/1.5.1
- [AUTODETECTED]       ARDUINO_VERSION = 167
- [DEFAULT]            ARCHITECTURE = avr
- [DEFAULT]            ARDMK_VENDOR = arduino
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/John/Library/Arduino15/preferences.txt
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/John/Programming/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr/local
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /Users/John/Programming/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = uno
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 9600  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [DEFAULT]            CFLAGS_STD =
- [DEFAULT]            CXXFLAGS_STD =
- [AUTODETECTED]       DEVICE_PATH =
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 Adafruit_NeoPixel
- [USER]                 DS3232RTC
- [USER]                 Time
- [PLATFORM]             Wire
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5
- [COMPUTED]           CC_VERSION = 6.2.0 (avr-gcc)
-------------------------
mkdir -p build-uno
/usr/local/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega328p -DF_CPU=16000000 -DARDUINO=167 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/utility   -I/Users/John/Programming/Arduino/libraries/Adafruit_NeoPixel   -I/Users/John/Programming/Arduino/libraries/DS3232RTC   -I/Users/John/Programming/Arduino/libraries/Time -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions  bClock.ino -o build-uno/bClock.ino.o
In file included from <command-line>:0:0:
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:23:20: fatal error: stdlib.h: No such file or directory
 #include <stdlib.h>
                    ^
compilation terminated.
make: *** [build-uno/bClock.ino.o] Error 1
```

Makefile
```
BOARD_TAG    = uno
ISP_PROG = usbasp

MCU = atmega328p
F_CPU = 16000000
AVRDUDE_ARD_BAUDRATE = 57600

ARDUINO_LIBS = Time DS3232RTC Wire Adafruit_NeoPixel
AVRDUDE_OPTS = -V

# ARDUINO_QUIET = 1

include ${ARDMK_DIR}/Arduino.mk
```

ladislas, 2016-10-14:
tested both arduino-makefile standalone and bare-arduino-project with avr-gcc 6.2 installed from scratch with homebrew (remove all and install again using `brew install avr-libc`) --> works well.

also using neopixel, no issues whatsoever.

ladislas, 2016-10-14:
another example using using the autolib detection from my fork

```
# ladislas @ LadBookPro in ~/dev/weareleka/leka-prototype-arduino/src/Leka (master●●)
→ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Java
- [USER]               ARDMK_DIR = /Users/ladislas/dev/weareleka/leka-prototype-arduino/Arduino-Makefile
- [AUTODETECTED]       ARDUINO_VERSION = 1612
- [DEFAULT]            ARCHITECTURE = avr
- [DEFAULT]            ARDMK_VENDOR = arduino
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/ladislas/Library/Arduino15/preferences.txt
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/ladislas/Documents/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr/local
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [USER]               USER_LIB_PATH = /Users/ladislas/dev/weareleka/leka-prototype-arduino/lib
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_SUB = atmega2560
- [USER]               BOARD_TAG = mega
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = mega (from build.variant)
- [USER]               OBJDIR = /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega
- [COMPUTED]           ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
- [USER]               MONITOR_BAUDRATE = 115200
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [USER]               CFLAGS_STD = -std=gnu11
- [USER]               CXXFLAGS_STD = -std=gnu++1z
- [COMPUTED]           DEVICE_PATH = /dev/tty.usbmodem* (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      SKETCH_LIBS =
- [USER]                 Motor
- [USER]                 NeoPixel
-
-                      SKETCH_LIBS_DEPS =
- [USER]                 IMotor
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5
- [COMPUTED]           CC_VERSION = 6.2.0 (avr-gcc)
-------------------------
mkdir -p /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega
/usr/local/bin/avr-g++ -MMD -c -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=1612 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/mega     -I/Users/ladislas/dev/weareleka/leka-prototype-arduino/lib/Motor   -I/Users/ladislas/dev/weareleka/leka-prototype-arduino/lib/NeoPixel   -I/Users/ladislas/dev/weareleka/leka-prototype-arduino/lib/Motor   -I/Users/ladislas/dev/weareleka/leka-prototype-arduino/lib/NeoPixel   -I/Users/ladislas/dev/weareleka/leka-prototype-arduino/lib/IMotor -Wall -ffunction-sections -fdata-sections -Os -pedantic -Wall -Wextra -fdiagnostics-color -fpermissive -fno-exceptions -std=gnu++1z main.cpp -o /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega/main.cpp.o
/usr/local/bin/avr-gcc -mmcu=atmega2560 -Wl,--gc-sections -Os -flto -fuse-linker-plugin -o /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega/Leka.elf /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega/main.cpp.o /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega/libcore.a  -lc -lm
/usr/local/bin/avr-objcopy -j .eeprom --set-section-flags=.eeprom='alloc,load' \
		--change-section-lma .eeprom=0 -O ihex /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega/Leka.elf /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega/Leka.eep
/usr/local/bin/avr-objcopy: --change-section-lma .eeprom=0x0000000000000000 never used
/usr/local/bin/avr-objcopy -O ihex -R .eeprom /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega/Leka.elf /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega/Leka.hex

/usr/local/bin/avr-size --mcu=atmega2560 -C --format=avr /Users/ladislas/dev/weareleka/leka-prototype-arduino/bin/Leka/mega/Leka.elf
AVR Memory Usage
----------------
Device: atmega2560

Program:    6174 bytes (2.4% Full)
(.text + .data + .bootloader)

Data:        355 bytes (4.3% Full)
(.data + .bss + .noinit)
```

ladislas, 2016-10-14:
so i'm pretty sure you only updated avr-gcc without recompiling avr-libc :)

tuna-f1sh, 2016-10-14:
Thanks for the quick reply. I just compiling a single file avr source and it couldn't find avr/io so it appears to be a problem with my includes sourcing rather than Arduino-Makefile - probably should have tried that first!

libc just finished re-installing as you suggested and it's working now...thanks for the help. Enjoying the colourised warning/errors of gcc 6 👍

ladislas, 2016-10-14:
glad i could help :)

closed 2017-03-19 by sej7278