sudar/Arduino-Makefile #467: Wire.h not found

heitorPB, 2016-11-02:
Hello,

Could not compile example master_reader. Here is output of make:
```
$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/h/projects/Arduino 
- [USER]               ARDMK_DIR = /home/h/projects/Bare-Arduino-Project/Arduino-Makefile 
- [DEFAULT]            ARDUINO_VERSION = 100 
- [USER]               ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [DEFAULT]            ARDUINO_SKETCHBOOK =  
- [USER]               AVR_TOOLS_DIR = /usr 
- [COMPUTED]           ARDUINO_LIB_PATH = /home/h/projects/Arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/h/projects/Arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /home/h/projects/Arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /home/h/projects/Arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 9600  (in sketch)
../../Arduino.mk:971: *** The following libraries specified in ARDUINO_LIBS could not be found (searched USER_LIB_PATH and ARDUINO_LIB_PATH): Wire.  Stop.
```

I have not installed arduino-ide, just cloned the github repo. I have avr-gcc 6.2.

I can compile and upload the Blink example.

Also, auto-detection of the architecture did not work for me, I had to specify it in the Makefile:
```
$ cat Makefile 
# Arduino Make file. Refer to https://github.com/sudar/Arduino-Makefile

ARCHITECTURE = avr
BOARD_TAG    = uno
ARDUINO_LIBS = Wire

include ../../Arduino.mk
```

sej7278, 2016-11-02:
please report this to [BareArduinoProject](https://github.com/ladislas/Bare-Arduino-Project/issues) that uses a rather modified fork of this makefile.

also without the ide installed (or at least arduino-core, not just avr-gcc) it won't work, it won't have the libraries like Wire.h or Arduino.h or avrdude etc.

heitorPB, 2016-11-02:
There is no need to install arduino ide. I have not installed. I have avrdude, avr-gcc, avr-libc and the arduino repo cloned.

I fixed this problem by creating a symbolic link in Arduino/libraries/Wire -> ../hardware/arduino/avr/libraries/Wire/ (this is in the arduino ide files)

My arduino codes compile and i am able to upload to the board, but there was this ugly fix.

closed 2017-08-15 by sej7278