sudar/Arduino-Makefile #470: ard-reset-arduino resets Leonardo device under OSX but not under Ubuntu

rdpoor, 2016-12-08:
This is less likely a bug with `ard-reset-arduino` and more likely something with the serial or usb drivers, but the following command succeeds in resetting our USB-connected Leonardo-like device (an ATmega32U4) under OSX but not under Ubuntu:

    $ ard-reset-arduino --verbose --caterina /dev/$TTY

where $TTY is `cu.usbmodem1411` in OSX and `ttyACM0` in Ubuntu.

Under Ubuntu, I observe /dev/ttyACM0 vanishing and being replaced by /dev/ttyACM1 for a handful of seconds before /dev/ttyACM0 reappears.  However, watching the LEDs on our Leonardo board tells me that its sketch is not interrupted (which is why I know it is not entering the boot loader).  

The same sketch, in fact the same board, when driven by `ard-reset-arduino` under OSX _does_ get interrupted and the boot loader runs for about 8 seconds before the sketch resumes.

I do not know what tools to bring to bear to help debug this.  (I have an ancient analog scope but no logic analyzer, so I can't see the exact traffic on the USB port.)  Any suggestions what would account for the difference?

rdpoor, 2016-12-08:
In case it's helpful, here's the result of:

`strace -e trace=file,ioctl,close /boot/ard-reset-arduino --verbose --caterina /dev/ttyACM0`

```Forcing reset using 1200bps open/close on port /dev/ttyACM0
open("/dev/ttyACM0", O_RDWR|O_NOCTTY|O_NONBLOCK) = 3
ioctl(3, TCGETS, {B9600 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B9600 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B9600 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_START or TCSETS, {B57600 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B9600 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCFLSH, TCIFLUSH)              = 0
close(3)                                = 0
open("/dev/ttyACM0", O_RDWR|O_NOCTTY|O_NONBLOCK) = 3
ioctl(3, TCGETS, {B9600 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B9600 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B9600 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_START or TCSETS, {B1200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B1200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCFLSH, TCIFLUSH)              = 0
ioctl(3, TIOCMBIS, [TIOCM_RTS])         = 0
ioctl(3, TIOCMBIC, [TIOCM_DTR])         = 0
close(3)                                = 0
stat("/dev/ttyACM0", {st_mode=S_IFCHR|0660, st_rdev=makedev(166, 0), ...}) = 0
/dev/ttyACM0 has come back after reset
+++ exited with 0 +++
```

sej7278, 2016-12-08:
it works on debian, just tried it with my pro micro.

what version of arduino-mk are you using - git or ubuntu package (and which distro)?

md5sum is: c01bf83210d21f0430f32ee66e458bce  ard-reset-arduino

i suspect its either something in your ubuntu environment, or something to do with your hardware - the 32u4 is an odd beast, especially with the sparkfun firmware that bypasses the 8sec delay but only on hard boot, not soft reboot. on windows the 32u4 causes all sorts of problems with that silly disappearing usb port.

reset.py
```
open("/dev/ttyACM0", O_RDWR|O_NOCTTY|O_NONBLOCK) = 3
ioctl(3, TCGETS, {B1200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B1200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B1200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_START or TCSETS, {B1200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B1200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TIOCMBIS, [TIOCM_DTR])         = 0
ioctl(3, TIOCMBIS, [TIOCM_RTS])         = 0
ioctl(3, TCFLSH, TCIFLUSH)              = 0
close(3)                                = 0
close(5)                                = 0
close(4)                                = 0
close(7)                                = 0
close(6)                                = 0
+++ exited with 0 +++
```

rdpoor, 2016-12-09:
I'm embarrassed to admit what the problem was: another process was grabbing ttyACM0.  I finally figured that out when I plugged in a second Leonardo and had no trouble resetting the device on ttyACM1.  

So I'll leave this here as form of atonement for my error, and perhaps as a help to the next person who makes the same error.

closed 2016-12-09 by rdpoor

rdpoor, 2016-12-10:
As further atonement, here's a tiny low-level C program that appears to work under OSX and Ubuntu for flipping an ATmega32U4 into boot mode.  This might be handy for people that don't want to pull a full Python environment simply to reset their Arduino:

```
/**
 * Reset a Leonardo-style Arduino chip (e.g. atmega32u4).  This is done by
 * setting the speed to 1200 baud and then disconnecting.
 */
#include <fcntl.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <termios.h>
#include <errno.h>

void errtest(int v, char *s) {
  if (v < 0) {
    perror(s);
    exit(v);
  }
}

int main(int argc, char **argv) {
  struct termios t;

  int fd = open(argv[1], O_RDWR | O_NOCTTY | O_NONBLOCK);
  errtest(fd, "opening tty");
  errtest(tcgetattr(fd, &t), "tcgetattr");
  errtest(cfsetspeed(&t, B1200), "cfsetspeed");
  errtest(tcsetattr(fd, TCSANOW, &t), "tcsetattr");
  errtest(close(fd), "close");
}
```