sudar/Arduino-Makefile #474: [ATtiny85] pins_arduino.h: No such file or directory

JelmerT, 2017-01-16:
I'm trying to compile for ATtiny85 (arduino trinket)

Working with arduino 1.8.1 on Mac with `attiny` by Mellis and `Adafruit AVR Boards` by Adafruit installed via Board Manager. Both these alternate cores work in arduino itself.
When trying to compile for both alternate cores via makefile I get following error:

```
In file included from <command-line>:0:0:
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:257:26: fatal error: pins_arduino.h: No such file or directory
 #include "pins_arduino.h"
```

My config for attiny by melis:
> BOARD_TAG = ATtinyX5
> BOARD_SUB = attiny85
> ALTERNATE_CORE_PATH = ~/Library/Arduino15/packages/attiny/hardware/avr/1.0.2

```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [AUTODETECTED]       ARDUINO_DIR = /Applications/Arduino.app/Contents/Java 
- [COMPUTED]           ARDMK_DIR = /Users/jelmer/Workspace/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 181 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/jelmer/Library/Arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/jelmer/Documents/Arduino (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /Applications/Arduino.app/Contents/Java/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [USER]               ALTERNATE_CORE_PATH = ~/Library/Arduino15/packages/attiny/hardware/avr/1.0.2 
- [COMPUTED]           ARDUINO_VAR_PATH = ~/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = ~/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/boards.txt (from ALTERNATE_CORE_PATH)
- [DEFAULT]            USER_LIB_PATH = /Users/jelmer/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = attiny85 
- [USER]               BOARD_TAG = ATtinyX5 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = tiny8 (from build.variant)
- [COMPUTED]           OBJDIR = build-ATtinyX5-attiny85 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
```

My config for the Adafruit core:
> BOARD_TAG = trinket3
> ALTERNATE_CORE_PATH = ~/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9

```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [AUTODETECTED]       ARDUINO_DIR = /Applications/Arduino.app/Contents/Java 
- [COMPUTED]           ARDMK_DIR = /Users/jelmer/Workspace/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 181 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/jelmer/Library/Arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/jelmer/Documents/Arduino (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /Applications/Arduino.app/Contents/Java/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [USER]               ALTERNATE_CORE_PATH = ~/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9 
- [COMPUTED]           ARDUINO_VAR_PATH = ~/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = ~/Library/Arduino15/packages/adafruit/hardware/avr/1.4.9/boards.txt (from ALTERNATE_CORE_PATH)
- [DEFAULT]            USER_LIB_PATH = /Users/jelmer/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = trinket3 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = tiny8 (from build.variant)
- [COMPUTED]           OBJDIR = build-trinket3 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
```

I tried defining CORE and ARDUINO_CORE_PATH to no avail.

What am I missing?

sej7278, 2017-01-16:
This works for me on Linux with 1.8.1 for damellis attiny85:

```make
ARDUINO_DIR = $(HOME)/arduino-1.8.1
BOARD_TAG = attiny
BOARD_SUB = attiny85
ALTERNATE_CORE = attiny
F_CPU = 16000000L
include /usr/share/arduino/Arduino.mk
```

And this works for trinket3:

```make
ARDUINO_DIR = $(HOME)/arduino-1.8.1
ALTERNATE_CORE = adafruit
BOARD_TAG = bluefruitmicro
BOARD_TAG = trinket3
include /usr/share/arduino/Arduino.mk
```

my ~/arduino16/hardware directory looks like this, i'm not using that awful boards mangler though:

```
adafruit/
└── avr
    ├── boards.txt
    ├── bootloaders
    │   ├── caterina
    │   └── optiboot
    ├── platform.txt
    └── variants
        ├── bluefruitmicro
        ├── flora
        └── tiny8
attiny/
└── avr
    ├── boards.txt
    ├── platform.txt
    └── variants
        ├── tiny14
        └── tiny8
```

JelmerT, 2017-01-16:
Thanks @sej7278 for checking.

Your configs all give 
```
avr-g++: error: missing device or architecture after '-mmcu='
```
aka can't find the board file at all. The board manager makes my directory tree look completely different:

```
jelmer ~/Library/Arduino15$ tree -L 5
.
├── library_index.json
├── package_adafruit_index.json
├── package_adafruit_index.json.sig.tmp
├── package_damellis_attiny_index.json
├── package_damellis_attiny_index.json.sig.tmp
├── package_esp8266com_index.json
├── package_esp8266com_index.json.sig.tmp
├── package_index.json
├── package_index.json.sig
├── packages
│   ├── adafruit
│   │   └── hardware
│   │       └── avr
│   │           └── 1.4.9
│   ├── arduino
│   │   ├── hardware
│   │   │   └── avr
│   │   │       └── 1.6.17
│   │   └── tools
│   │       ├── arduinoOTA
│   │       │   └── 1.0.0
│   │       ├── avr-gcc
│   │       │   └── 4.9.2-atmel3.5.3-arduino2
│   │       └── avrdude
│   │           └── 6.3.0-arduino8
│   ├── attiny
│   │   └── hardware
│   │       └── avr
│   │           └── 1.0.2
│   └── esp8266
│       ├── hardware
│       │   └── esp8266
│       │       └── 2.3.0
│       └── tools
│           ├── esptool
│           │   └── 0.4.9
│           ├── mkspiffs
│           │   └── 0.1.2
│           └── xtensa-lx106-elf-gcc
│               └── 1.20.0-26-gb404fb9-2
├── preferences.txt
.
.
.
```
If I define the the `ALTERNATE_CORE_PATHS` I can find the boards and submenus though.
```
$ make show_boards
ATtinyX4  ATtiny24/44/84
ATtinyX5  ATtiny25/45/85

$ make show_submenu
ATtinyX4.menu.cpu.attiny24  ATtiny24
ATtinyX4.menu.cpu.attiny44  ATtiny44
ATtinyX4.menu.cpu.attiny84  ATtiny84
ATtinyX5.menu.cpu.attiny25  ATtiny25
ATtinyX5.menu.cpu.attiny45  ATtiny45
ATtinyX5.menu.cpu.attiny85  ATtiny85
```
and
```
$ make show_boards
adafruit32u4        Adafruit 32u4 Breakout
bluefruitmicro      Adafruit Bluefruit Micro
circuitplay32u4cat  Adafruit Circuit Playground
feather32u4         Adafruit Feather 32u4
flora8              Adafruit Flora
gemma               Adafruit Gemma 8MHz
metro               Adafruit Metro
protrinket3         Pro Trinket 3V/12MHz (USB)
protrinket3ftdi     Pro Trinket 3V/12MHz (FTDI)
protrinket5         Pro Trinket 5V/16MHz (USB)
protrinket5ftdi     Pro Trinket 5V/16MHz (FTDI)
trinket3            Adafruit Trinket 8MHz
trinket5            Adafruit Trinket 16MHz
```

So I have the feeling I'm close, especially since the location of the missing header file is included in the compile command generated by the makefile:

>mkdir -p build-ATtinyX5-attiny85
>/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -x c++ 
>-include Arduino.h -MMD -c -mmcu=attiny85 -DF_CPU= -DARDUINO=181 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ 
>-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino 
>**-I~/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8**    
>-Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto test.ino 
>-o build-ATtinyX5-attiny85/test.ino.o

Seems like the 
```
ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
```
is also different from the one that the arduino IDE itself uses (when compiling for damellis attiny85)
```
Using board 'ATtinyX5' from platform in folder: /Users/jelmer/Library/Arduino15/packages/attiny/hardware/avr/1.0.2
Using core 'arduino' from platform in folder: /Users/jelmer/Library/Arduino15/packages/arduino/hardware/avr/1.6.17
```

I guess the makefiles are just not compatible with the board manager thing? (I do like the self update feature of the board manager)

JelmerT, 2017-01-16:
Gave up on trying to make the makefile play nice with the board manager, added the boards manually and then things work.

For future reference: How to compile ATtiny85 core (Trinket 3v) on Mac with makefile
**Adafruit core:**
-download https://adafruit.github.io/arduino-board-index/boards/adafruit-avr-1.4.9.tar.bz2 (or newer)
-unzip in `~/Documents/Arduino/hardware/`
-rename folder to `adafruit`
-add folder named `avr` in previous folder
-move everything into `avr` folder
-use following makefile
```
ALTERNATE_CORE = adafruit
BOARD_TAG = trinket3
F_CPU = 8000000L
```

**Damellis core:**
-download https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json (or newer)
-unzip in `~/Documents/Arduino/hardware/`
-rename folder to `attiny`
-add folder named `avr` in previous folder
-move everything into `avr` folder
-use following makefile
```
BOARD_TAG = ATtinyX5
BOARD_SUB = attiny85
ALTERNATE_CORE = attiny
F_CPU = 8000000L
```

Your folders should look like this
```
~/Documents/Arduino/hardware$ tree -L 3
.
├── adafruit
│   └── avr
│       ├── README.md
│       ├── boards.txt
│       ├── bootloaders
│       ├── drivers
│       ├── libraries
│       ├── platform.txt
│       └── variants
└── attiny
    └── avr
        ├── README.md
        ├── boards.txt
        ├── platform.txt
        └── variants
```

This issue can be closed, although making makefile compatible with the board manager structure (even if it's awful) might be a good thing going forward.

sej7278, 2017-01-16:
i can't see we can ever make it compatible with boards mangler, its directory structure and multiple versioned directory structure seems almost random and not predictable.

if you want updates, then its best to use the cores from git and update that way, rather than downloading the zip files.

glad you got it working for you anyway.

closed 2017-01-16 by sej7278