sudar/Arduino-Makefile #481: Unable to upload Sketch to Nano "not in sync" in Linux but works with Arduino Studio Win 10

Floyddotnet, 2017-02-18:
I try to upload sketches via my Raspbian (Debian / Raspberry Pi) and Windows 10 to my Nano 328.

Arduino Studio 1.6.7  tells me:

Board: Arduino Nano
Processor: ATmega 328
Programmer: AVRISP mkII

Upload the Sketch via Arduino Studio on Windows 10 works fine.
Upload the same Sketch vias Raspbian fails with the following error:

```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX
- [AUTODETECTED]       ARDUINO_DIR = /usr/share/arduino
- [COMPUTED]           ARDMK_DIR = /source/Ambilight/ArduinoMk (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 105
- [DEFAULT]            ARCHITECTURE =
- [DEFAULT]            ARDMK_VENDOR = arduino
- [DEFAULT]            ARDUINO_SKETCHBOOK =
- [BUNDLED]            AVR_TOOLS_DIR = /usr/share/arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino//variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino//boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = nano328
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = eightanaloginputs (from build.variant)
- [COMPUTED]           OBJDIR = build-nano328 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino//cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 115200  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [DEFAULT]            CFLAGS_STD =
- [DEFAULT]            CXXFLAGS_STD =
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyUSB0
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [SYSTEM]               Adafruit_NeoPixel
- [SYSTEM]               SPI
- [SYSTEM]               arduino_uip
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino//bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------

....

AVR Memory Usage
----------------
Device: atmega328p

Program:   24362 bytes (74.3% Full)
(.text + .data + .bootloader)

Data:       1472 bytes (71.9% Full)
(.data + .bss + .noinit)

make reset
make[1]: Entering directory '/source/Ambilight/AmbilightSketch'
/usr/bin/ard-reset-arduino  /dev/ttyUSB0
make[1]: Leaving directory '/source/Ambilight/AmbilightSketch'
make do_upload
make[1]: Entering directory '/source/Ambilight/AmbilightSketch'
/usr/share/arduino/hardware/tools/avr/../avrdude -q -V -p atmega328p -C /usr/share/arduino/hardware/tools/avr/../avrdude.conf -D -c arduino -b 57600 -P /dev/ttyUSB0 \
                -U flash:w:build-nano328/AmbilightSketch.hex:i
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xbf
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x80
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x48
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x88
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x38
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xce
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x0e
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xd6
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x0a
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x67

avrdude done.  Thank you.

/source/Ambilight/ArduinoMk/Arduino.mk:1462: recipe for target 'do_upload' failed
make[1]: *** [do_upload] Error 1
make[1]: Leaving directory '/source/Ambilight/AmbilightSketch'
/source/Ambilight/ArduinoMk/Arduino.mk:1454: recipe for target 'upload' failed
make: *** [upload] Error 2
```

I try it with Arduino-Makefile release 1.5 and the current git version 1.5.2+ (last commit: 08b4f441ff5e608b67d43491f414f43b1c4c5384)

Floyddotnet, 2017-02-18:
Its a Nano v3 from Robodyn with CH340G - USB-UART interface

http://robotdyn.com/catalog/boards/nano_v3_ch340g_compatible_with_arduino_nano_v3/

Floyddotnet, 2017-02-18:
Output from Arduino Studio in Windows 10:

```
avrdude: Version 6.3, compiled on Dec 16 2016 at 13:33:19
...
         Using Port                    : COM6
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions
...
```

Floyddotnet, 2017-02-18:
I upgrade my avrdude on my raspberry pi also to version 6.3 .. but change nothing

sej7278, 2017-02-19:
that's the most common arduino error ever, its got dozens of causes which doesn't help much.

first i'd say try setting ```ARDUINO_SKETCHBOOK```, as i assume most of those paths aren't supposed to be from the root dir and those libraries aren't really system libs e.g. /source/Ambilight/ArduinoMk and /libraries. i'm surprised it even compiles with that setup.

also is you user in the dialout group, otherwise it won't have permissions to do the serial upload.

Floyddotnet, 2017-02-19:
> first i'd say try setting ARDUINO_SKETCHBOOK, as i assume most of those paths aren't supposed to be from the root dir and those libraries aren't really system libs e.g. /source/Ambilight/ArduinoMk and /libraries. i'm surprised it even compiles with that setup.

I can try this but compiling isn't the problem.

> also is you user in the dialout group, otherwise it won't have permissions to do the serial upload.

Yes

```
root@pi:~# groups
root dialout video
```

> what are the permissions of /dev/ttyUSB0
`screen /dev/ttyUSB0 115200 ` works

> finally as is usually the problem with pi's - do you have enough power to power the usb nano - a 500mA phone charger is no good

The Nano is directly powerd from a good 5V 12A Power supply. Also the Pi.

Note: if i unbox a new nano, the first upload works

Floyddotnet, 2017-02-19:
A manuell call of avrdude 6.3

```
root@pi:~# avrdude -q -V -p atmega328p -C /etc/avrdude.conf -D -c arduino -b 57600 -P /dev/ttyUSB0 -U flash:w:build-nano328/AmbilightSketch.hex:i
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xbd
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xfc
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x53
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x0a
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x9f
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xad
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xbd
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xdd
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x0a
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x0a

avrdude done.  Thank you.

```

Floyddotnet, 2017-02-19:
Change the programmer to `avrispmkII` didn't help. But get different error.

```
root@pi:~# avrdude -q -V -p atmega328p -C /etc/avrdude.conf -D -c avrispmkII  -b 57600 -P /dev/ttyUSB0 -U flash:w:build-nano328/AmbilightSketch.hex:i
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
...

```

sej7278, 2017-02-19:
you shouldn't be running it as root. permissions are irrelevant as root. try adding the pi (or whatever user you are) to the dialout group. but as its not working as root i guess that's not the problem.

is this raspbian jessie, as that has some weird systemctl stuff that interferes with the serial ports (getty console etc.)

i'm just wondering if you're messing up the fuses or bootloader and that's why it works the first time. shouldn't be able to do that over serial though, that's iscp only. try avrdude without the -D option, that'll force a chip erase e.g. in your Makefile:

```
BOARD_TAG = nano328
MONITOR_PORT = /dev/ttyUSB0
AVRDUDE_OPTS = -v
include /usr/share/arduino/Arduino.mk
```

i assume your nano is a 16MHz 328p version and not the older 168p or 20MHz version? flashing a 20MHz crystal as a 16MHz board (or vice versa) can definitely cause the problem.

Floyddotnet, 2017-02-22:
Perfect .. After adding "MONITOR_PORT" it works
Thanks.

closed 2017-02-22 by Floyddotnet

fluentart, 2018-06-18:
I found that if you choose Tools>Processor>**ATmega328P (Old Bootloader)** then uploading works on ubuntu linux. Perhaps that is useful to someone reading this.

derfaq, 2018-09-02:
thanks @fluentart it worked for me

acalzada, 2022-05-06:
Some new arduino nano boards when uploading sketches to them the Make file needs weird configuration parameters.

For example setting BOARD_TAG = uno, instead of nano.

Check this post in case the above solutions do not work.

[https://forum.arduino.cc/t/arduino-makefile-reqs-different-setup-for-apparently-otherwise-same-boards/632931](https://forum.arduino.cc/t/arduino-makefile-reqs-different-setup-for-apparently-otherwise-same-boards/632931)