sudar/Arduino-Makefile #482: Make clean doesn't work

dHutchings, 2017-03-03:
This may no longer be applicable, since this was on version 1.5, but still:

OS: Ubuntu 14.04
Arduino IDE: 1.6.12
Snippet of Our wrapper makefile at the bottom.

Running "make clean" produces the following error message: "/Arduino-Makefile/Arduino.mk:772: *** At least one source file (*.ino, *.pde, *.cpp, *c, *cc, *.S) is needed.  Stop"

Make clean doesn't actually remove the bin files generated by the previous compilation.

Snippet:

''' PROJECT_DIR       = $(PWD)

### AVR_GCC_VERSION
### Check if the version is equal or higher than 4.9
AVR_GCC_VERSION  := $(shell expr `avr-gcc -dumpversion | cut -f1` \>= 4.9)

### ARDMK_DIR
### Path to the Arduino-Makefile directory.
ARDMK_DIR         = $(PROJECT_DIR)/Arduino-Makefile

### ARDUINO_DIR
### Path to the Arduino application and ressources directory.
ARDUINO_DIR       = /usr/share/arduino

ARDUINO_LIBS := Wire Servo

### USER_LIB_PATH
### Path to where the your project's libraries are stored.
SKETCH_LIBS = hibike Adafruit_TCS34725 EEPROM SevenSeg TimerOne TimerThree FlexiTimer2 PID_v1 Encoder
USER_LIB_PATH     :=  $(realpath $(PROJECT_DIR)/lib)

### BOARD_TAG & BOARD_SUB
### For Arduino IDE 1.0.x
### Only BOARD_TAG is needed. It must be set to the board you are currently using. (i.e uno, mega2560, etc.)
# BOARD_TAG         = mega2560
### For Arduino IDE 1.6.x
### Both BOARD_TAG and BOARD_SUB are needed. They must be set to the board you are currently using. (i.e BOARD_TAG = uno, mega, etc. & BOARD_SUB = atmega2560, etc.)
### Note: for the Arduino Uno, only BOARD_TAG is mandatory and BOARD_SUB can be equal to anything
BOARD_TAG         = micro
# BOARD_SUB         = 

### MONITOR_BAUDRATE
### It must be set to Serial baudrate value you are using.
MONITOR_BAUDRATE  = 115200

### AVR_TOOLS_DIR
### Path to the AVR tools directory such as avr-gcc, avr-g++, etc.
AVR_TOOLS_DIR     = /usr

### AVRDDUDE
### Path to avrdude directory.
AVRDDUDE          = /usr/bin/avrdude

### CFLAGS_STD
CFLAGS_STD        = -std=gnu11

### CXXFLAGS_STD
CXXFLAGS_STD      = -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -fno-devirtualize -fno-use-cxa-atexit

### CPPFLAGS
### Flags you might want to set for debugging purpose. Comment to stop.
CXXFLAGS         = -pedantic -Wall -Wextra

### If avr-gcc -v is higher than 4.9, activate coloring of the output
ifeq "$(AVR_GCC_VERSION)" "1"
    CXXFLAGS += -fdiagnostics-color
endif
### random uid
RANDOM := 0x$(shell head -c 8 /dev/urandom | xxd -p)
CXXFLAGS += -D UID_RANDOM=$(RANDOM)

CXXFLAGS += -D DESCRIPTOR=$(shell python escape_json.py $(PROJECT_DIR)/devices/$(DEVICE)/*.json)

LOCAL_CPP_SRCS := $(wildcard $(PROJECT_DIR)/devices/$(DEVICE)/*.cpp)

### MONITOR_PORT
### The port your board is connected to. Using an '*' tries all the ports and finds the right one.
MONITOR_PORT      = /dev/ttyACM*

### don't touch this
CURRENT_DIR       = $(shell basename $(CURDIR))

### OBJDIR
### This is were you put the binaries you just compile using 'make'
CURRENT_DIR       = $(shell basename $(CURDIR))
OBJDIR            = $(PROJECT_DIR)/bin/$(CURRENT_DIR)/$(BOARD_TAG)

### path to Arduino.mk, inside the ARDMK_DIR, don't touch.
include $(ARDMK_DIR)/Arduino.mk
'''

dHutchings, 2017-03-03:
Again, thanks for making this!  It's really helped us out, and for now, we're working around the lack of make clean by having a " rm -rf " in a script.

sej7278, 2017-03-03:
lol i've no idea how "make clean" wouldn't work, all it does is ```rm -rf $(OBJDIR)``` so its something weird you're doing in your wrapper.

that error message doesn't even make sense for "make clean", i suspect you've knackered the OBJDIR variable, does it even compile?

closed 2017-08-15 by sej7278