sudar/Arduino-Makefile #483: Unable to locate BOARD_SUB

geosmall, 2017-03-04:
Operating system: Ubuntu 16.04
Arduino IDE version: 1.8.1

Using this core: https://github.com/MCUdude/MightyCore.  Core builds correctly in 1.8.1 IDE.

make show_boards correctly identifies boards.
make show_submenu is not able to detect variants.

------Makefile and build terminal logs below----------------------------------------------------

Snippet of Makefile:

--- mighty 1284p
BOARD_TAG         = 1284
BOARD_SUB         = standard
BOARDS_TXT        = $(HOME)/Arduino/hardware/MightyCore/avr/boards.txt
BOOTLOADER_PARENT = $(HOME)/Arduino/hardware/MightyCore/avr/bootloaders
BOOTLOADER_PATH   = optiboot
BOOTLOADER_FILE   = optiboot_atmega1284p.hex
ISP_PROG     	   = usbasp
AVRDUDE_OPTS 	   = -v
include /home/geo/Arduino/Arduino-Makefile/Arduino.mk

geo@geo-Latitude-D630:~/Arduino/Blink$ **make show_boards**
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /opt/arduino-1.8.1 
- [USER]               ARDMK_DIR = /home/geo/Arduino/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 181 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/geo/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/geo/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /opt/arduino-1.8.1/hardware/tools/avr 
- [COMPUTED]           ARDUINO_LIB_PATH = /opt/arduino-1.8.1/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /opt/arduino-1.8.1/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /opt/arduino-1.8.1/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [USER]               BOARDS_TXT = /home/geo/Arduino/hardware/MightyCore/avr/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /home/geo/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = standard 
- [USER]               BOARD_TAG = 1284 
- [COMPUTED]           CORE = MightyCore (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-1284-standard (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /opt/arduino-1.8.1/hardware/arduino/avr/cores/MightyCore (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [USER]               BOOTLOADER_PARENT = /home/geo/Arduino/hardware/MightyCore/avr/bootloaders 
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
-------------------------
1284  ATmega1284
164   ATmega164
16    ATmega16
324   ATmega324
32    ATmega32
644   ATmega644
8535  ATmega8535

geo@geo-Latitude-D630:~/Arduino/Blink$ make show_submenu
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /opt/arduino-1.8.1 
- [USER]               ARDMK_DIR = /home/geo/Arduino/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 181 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/geo/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/geo/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /opt/arduino-1.8.1/hardware/tools/avr 
- [COMPUTED]           ARDUINO_LIB_PATH = /opt/arduino-1.8.1/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /opt/arduino-1.8.1/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /opt/arduino-1.8.1/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [USER]               BOARDS_TXT = /home/geo/Arduino/hardware/MightyCore/avr/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /home/geo/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = standard 
- [USER]               BOARD_TAG = 1284 
- [COMPUTED]           CORE = MightyCore (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-1284-standard (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /opt/arduino-1.8.1/hardware/arduino/avr/cores/MightyCore (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [USER]               BOOTLOADER_PARENT = /home/geo/Arduino/hardware/MightyCore/avr/bootloaders 
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
-------------------------

sej7278, 2017-03-04:
Probably as the core has gone a bit overboard with the submenu's!:

```
# Menu options
menu.clock=Clock
menu.BOD=BOD
menu.pinout=Pinout
menu.variant=Variant
menu.LTO=Compiler LTO
```

We only support chip/cpu variants (and only one!), even if you search'n'replace ```(chip|cpu)``` with ```(chip|cpu|clock)``` in Arduino.mk, there's is no ```1284.menu.clock.12MHz_external.build.mcu``` value as the mcu is yet another submenu variant!

```make
ALTERNATE_CORE = MightyCore
BOARD_TAG = 1284
BOARD_SUB = 12MHz_external
```

The mighty-1284p and sanguino cores work well with ide 1.0

geosmall, 2017-03-05:
Ok thanks.  I agree the core has extreme menuing.  That said, it is indeed valid within IDE.

Feel free to close this issue if it is in fact unsupported.

Best Regards,
George

sej7278, 2017-03-05:
yes unfortunately it probably needs the board processing rewritten as its own utility to recursively figure out all of the random submenu's cores are using now, and this one has submenu's within submenu's it seems! also i think you're confusing variant with board_sub.

you could just put the values from boards.txt into your Makefile, for example this compiles for me:

```
ALTERNATE_CORE = MightyCore
BOARD_TAG = 1284
VARIANT = standard
MCU = atmega1284p
F_CPU = 16000000L
```

geosmall, 2017-03-05:
Thanks for the guidance sej7278.  Yes I think I am confusing variant and sub-board.  I will study the documentation and makefiles further.

Your makefile above works for me after changing the include path.

Thanks!

closed 2017-03-19 by sej7278