sudar/Arduino-Makefile #486: Segmentation fault on Serial.begin()

UsoHolger, 2017-03-11:
I am using Arch Linux and everything seemed to work fine. However: When I try to use serial communication I get 
`lto1: internal compiler error: Segmentation fault`
as soon as `Serial.begin(9600);` is uncommented. The same code works fine when compiling with the Arduino IDE.

The makefile:
    ARDUINO_DIR = /usr/share/arduino
    MONITOR_PORT = /dev/ttyACM0
    BOARD_TAG = uno
    ARDUINO_LIBS = SoftwareSerial
    include	/usr/share/arduino/Arduino.mk

sej7278, 2017-03-11:
the arch package seems seriously broken its been noticed before, use git

UsoHolger, 2017-03-11:
Thanks for the reply but that does not resolve the issue. I completely removed the AUR packages and cloned the git repo. The most simplistic case is provided by the "SerialPrint" example in the example folder of the repo. It produces the exact same error while other examples like the "Blink" example compile without any problems: 
```
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [AUTODETECTED]       ARDUINO_DIR = /usr/share/arduino 
- [COMPUTED]           ARDMK_DIR = /home/uso/arduino/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 180 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/uso/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/uso/Arduino (from arduino preferences file)
- [AUTODETECTED]       AVR_TOOLS_DIR = /usr (found in $PATH)
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/share/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/alex/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 9600  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyACM0 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 6.3.0 (avr-gcc)
-------------------------
mkdir -p build-uno
/usr/bin/avr-gcc -mmcu=atmega328p -Wl,--gc-sections -Os -flto -fuse-linker-plugin -o build-uno/SerialPrint.elf build-uno/SerialPrint.ino.o build-uno/libcore.a   -lc -lm 
lto1: internal compiler error: Segmentation fault
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper: fatal error: /usr/bin/avr-gcc returned 1 exit status
compilation terminated.
/usr/bin/avr-ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [../../Arduino.mk:1442: build-uno/SerialPrint.elf] Error 1
```
(That is of course the output of a second `make` without all the successfull preceeding steps)
Might that be a bug in the arch avr-gcc package instead?

sej7278, 2017-03-12:
probably as you're using avr-gcc 6.3 instead of 4.9

ladislas, 2017-03-12:
@UsoHolger how did you install your version of avr-gcc?

UsoHolger, 2017-03-13:
I am using the [avr-gcc](https://www.archlinux.org/packages/community/x86_64/avr-gcc/ package from the official Arch software repository installed via Pacman.

ladislas, 2017-03-13:
well I guess the best way would be to comment the different variables used in bare arduino project and let us know what causes the bug.

UsoHolger, 2017-03-13:
I tested using the simplistic SerialPrint example. What does the trick is setting the 2011 standard flag. So this makefile
```
BOARD_TAG    	  = uno

include ../../Arduino.mk
```
fails with the segmentation fault error while adding the line `CXXFLAGS_STD = -std=gnu++11` leads to successful compilation.

ladislas, 2017-03-13:
can i see your code?

UsoHolger, 2017-03-13:
I used the [SerialPrint](https://github.com/sudar/Arduino-Makefile/blob/master/examples/SerialPrint/SerialPrint.ino) example. So the code is:
```
void setup() {
	Serial.begin(9600);
	Serial.print("Press any key: ");
}

void loop() {
	if (Serial.available()) {
		Serial.println(Serial.read());
	}
}
```
with the makefile 
```
# Arduino Make file. Refer to https://github.com/sudar/Arduino-Makefile

BOARD_TAG    	  = uno
CXXFLAGS_STD      = -std=gnu++11 # if this is missing compilation fails because of the segmentation fault error

include ../../Arduino.mk
```

sej7278, 2017-03-13:
its the LTO parts of CXXFLAGS_STD/CFLAGS_STD, changing them from ```-std=gnu11 -flto -fno-fat-lto-objects``` to anything else e.g. ```-std=gnu11``` removes the LTO flags

https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#cflags_std

i assume your avr-gcc hasn't been built with LTO support

ladislas, 2017-04-25:
Yes, that the issue with AVR. We are currently investigating the homebrew formula. I'll keep you posted.

NVieville, 2017-07-21:
Hello,

Just hitting this issue too, and google led me here. As suggested in https://github.com/osx-cross/homebrew-avr/issues/46 (linked above), I can confirm adding:

CFLAGS_STD       = -std=gnu11 -flto -fno-fat-lto-objects
CXXFLAGS_STD     = -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize

to Makefile

just make arduino-mk (freshly checked-out from github) compiling without segfault on Fedora 26 x86_64 with:
- avr-gcc --version
avr-gcc (Fedora 6.3.0-1.fc26) 6.3.0
- /usr/bin/avr-gcc-ar --version
GNU ar (GNU Binutils) 2.27

Hope this will be helpful.

Cordially,

-- 
NVieville

sej7278, 2017-07-21:
@NVieville i'm a bit confused as these are set by default now in Arduino.mk

```
grep gnu.*11 Arduino.mk 
        CFLAGS_STD      = -std=gnu11 -flto -fno-fat-lto-objects
        CXXFLAGS_STD      = -std=gnu++11 -fno-threadsafe-statics -flto
```

Removing LTO flags fixes things for compilers that don't have LTO support, but i'd have thought the f23 compiler does.....?

NVieville, 2017-07-21:
Hello,

@sej7278 Thanks for your response, and sorry for not being quite explicit.
I saw that CFLAGS_STD and CXXFLAGS_STD were set in Arduino-mk, but my project wasn't compiling on Fedora 26 (lto1: internal compiler error: Segmentation fault). So, after reading this thread and the one cited above, I wanted to had  -fno-devirtualize to CXXFLAGS_STD. The only way I found was to re-define it in my Makefile. Maybe I shouldn't have do it for CFLAGS_STD too, it's unnecessary (I removed it now and only kept CXXFLAGS_STD).

I just tried your suggestion to remove the -flto flag, and compiling goes well. I just noticed that  AVR Memory Usage information displayed at the end of the building process is a bit higher when removing the -flto flag than with keeping it and adding the -fno-devirtualize one.

For information here is the output of the "avr-gcc -v" command on Fedora 26:

Using built-in specs.
Reading specs from /usr/lib/gcc/avr/6.3.0/device-specs/specs-avr2
COLLECT_GCC=/usr/bin/avr-gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/avr/6.3.0/lto-wrapper
Target: avr
Configured with: ../gcc-6.3.0/configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --target=avr --enable-languages=c,c++ --disable-nls --disable-libssp --with-system-zlib --enable-version-specific-runtime-libs --with-pkgversion='Fedora 6.3.0-1.fc26' --with-bugurl=https://bugzilla.redhat.com/
Thread model: single
gcc version 6.3.0 (Fedora 6.3.0-1.fc26) 

My knowledge about this is not enough to say if it's worth removing -flto flag (no link time optimization) or keeping it and adding -fno-devirtualize one in such a configuration. 

Sorry again for not being quite explicit in my first message. 
Cordially,

-- 
NVieville

sej7278, 2017-07-21:
how odd!

so if adding the ```-fno-devirtualize``` flag prevents it from segfaulting, then i assume avr-gcc has LTO support, but perhaps the regular gcc used to compile it has the bug mentioned [here](https://bugzilla.redhat.com/show_bug.cgi?id=1120792) that was in gcc 4.9 - maybe try gcc -v?

you should just be able to add this to your local Makefile:

```CXXFLAGS_STD += -fno-devirtualize```

NVieville, 2017-07-21:
Tha actual version of gcc in Fedora 26 is: gcc (GCC) 7.1.1 20170622 (Red Hat 7.1.1-3)
Output of gcc -v:

```
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,objc,obj-c++,fortran,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 7.1.1 20170622 (Red Hat 7.1.1-3) (GCC) 
```

I tried your suggestion: adding `CXXFLAGS_STD += -fno-devirtualize` to my Makefile, but as Arduino.mk is included at the end of this Makefile, the `+=` only defines the `CXXFLAGS_STD` variable with `-fno-devirtualize` and doesn't append the parameter to the variable (not already defined) that is seen later in Arduino.mk as user defined and then taken as is and not modified.
The only solution (fine to me) is to define entirely the `CXXFLAGS_STD` variable in my local Makefile with the same parameters as the ones used in Arduino.mk, but appending `-fno-devirtualize` at the end.

Thanks for your explanations. 
Cordially,

-- 
NVieville

sej7278, 2017-07-21:
sorry, should be CXXFLAGS, this works for me:

```make
BOARD_TAG = leonardo
MONITOR_PORT = /dev/ttyACM0
CXXFLAGS += -fno-devirtualize
include /usr/share/arduino/Arduino.mk
```

not sure if any other flags need to be changed like CFLAGS or ASFLAGS.

i'd say maybe we ought to add ```-fno-devirtualize``` to Arduino.mk but it seems to affect performance and is only a workaround for a gcc bug (on fedora) and it won't fix the arch/osx issue of having avr-gcc 6.3 without LTO support.

i guess this is why the IDE bundles its own avr toolchain!

maybe we should move from querying the avr-gcc version:
```make
    ifeq ($(shell expr $(CC_VERNUM) '>' 490), 1)
```

to looking for LTO support instead:

```avr-gcc -v | grep COLLECT_LTO_WRAPPER```

sej7278, 2017-07-21:
I made PR #510 which adds ```-fno-devirtualize``` to CXXFLAGS_STD whilst we wait for gcc to fix their compiler bug (good luck with that if it was discovered in 6.3 and not fixed in 7.1.....)

NVieville, 2017-07-22:
@sej7278 Thank you.
I confirm that adding `CXXFLAGS += -fno-devirtualize` to Arduino.mk works on Fedora 26.

About the gcc bug, it seems it has already been reported:

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=80717

and workaround different than this proposed here is available:

https://bbs.archlinux.org/viewtopic.php?pid=1722955#p1722955

Haven't tested it. Workaround proposed here seems fine to me.

Cordially,

-- 
NVieville

sej7278, 2017-07-22:
@NVieville we're already using avr-gcc-ar when we have avr-gcc > 4.9 (when LTO is enabled) i don't believe avr-ar will work with LTO at all, so can't be a workaround for this issue. thanks for testing the workaround.

seems the archlinux guys have reverted a commit in avr-gcc 7.1.0-2 which the fedora guys haven't in 7.1.1-3 meaning that its still going to bite most users of avr-gcc > 6.3 (luckily debian's still on 4.9.2 :smirk:) as upstream haven't fixed it.

@sudar / @ladislas any objections to merging my PR to workaround this?

sudar, 2017-07-25:
@sej7278 No objections for merging the PR.

RicoWarner, 2019-05-28:
Hi All -  Have a simple Arduino Uno project.  Have a simple project that does use the following line and does generate the segmentation fault error as identified in this thread:
Serial.begin(9600);

However, I do not have a makefile system installed and do not have the brain cells nor time to figure out the 5 pages of instructions on the referenced github page for making it work.  Wondering if there is a way to set/unset the necessary flags via environment variables inside the program rather than through makefile?  

Any simple fix without becoming Arduino compiler guru?  Just have a few lines of code to run....

Thanks!

RicoWarner, 2019-05-28:
> Hi All - Have a simple Arduino Uno project. Have a simple project that does use the following line and does generate the segmentation fault error as identified in this thread:
> Serial.begin(9600); [...]

Hmmm maybe found fix?  Included SoftwareSerial.h and it appears to be compiling.