sudar/Arduino-Makefile #489: Disable bootloader

ddavidebor, 2017-03-18:
Is it possible to disable the bootloader?

The bootloader is being automatically bundled in the executable. Since i'm using native USB capabilities on the arduino micro i cannot user the embed bootloader and i'm programming my device only with the ICSP programmer.

Is it possible to disable the bundling of the bootloader into the final executable?

```
AVR Memory Usage
----------------

Device: atmega32u4

Program:    4600 bytes (14.0% Full)
(.text + .data + .bootloader)

Data:        155 bytes (6.1% Full)
(.data + .bss + .noinit
```

sej7278, 2017-03-19:
i don't know how you would do that, when you use an isp it doesn't write the bootloader, but not compiling/linking it i doubt is possible if you're using the arduino core and not just bare avr c/asm.

ddavidebor, 2017-03-19:
Ok if it doesn't write the bootloader it's fine :-)

closed 2017-03-19 by ddavidebor