sudar/Arduino-Makefile #494: ESP8266 NodeMCU Makefile

momiccioli, 2017-04-12:
I added functionality to the Arduino IDE to compile and upload to an ESP8266 dev board. Its becoming very popular and I have a few projects that I will be using it with. It all works fine in the IDE. I have tried to get this working with the Arduino-makefile which work great with all my Arduino sketches, but no luck with the ESP8266. Is it possible to get this to work? I am not strong with make so I have been doing some guess work without much success.

This is the board manager used to install the esp8266 board
http://arduino.esp8266.com/stable/package_esp8266com_index.json

Module I am using is “NodeMCU 1.0 (ESP-12E Module)”

Thanks

sej7278, 2017-04-12:
See also issue #454 

Here's a Makefile i wrote that calls arduino-builder, not sure it'll work with the awful board manager but it does with the [git checkout](https://github.com/esp8266/Arduino#using-git-version) of the esp8266 arduino port.

its not arduino-mk as i think it would need a lot of work to add support for the xtensa toolchain.

```make
ARDUINO_PATH = $(HOME)/arduino-1.8.2
SKETCHBOOK  = $(HOME)/arduino16
ESPTOOL = $(SKETCHBOOK)/hardware/esp8266com/esp8266/tools/esptool/esptool
SKETCH = $(notdir $(CURDIR)).ino
TARGET_DIR = $(CURDIR)/build-esp8266
MONITOR_PORT = /dev/ttyUSB0

all:
	@ mkdir -p $(TARGET_DIR)

	$(ARDUINO_PATH)/arduino-builder -compile -logger=machine \
	-hardware "$(ARDUINO_PATH)/hardware" \
	-hardware "$(SKETCHBOOK)/hardware" \
	-tools "$(ARDUINO_PATH)/tools-builder" \
	-tools "$(ARDUINO_PATH)/hardware/tools/avr" \
	-built-in-libraries "$(ARDUINO_PATH)/libraries" \
	-libraries "$(SKETCHBOOK)/libraries" \
	-fqbn=esp8266com:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M \
	-ide-version=10802 \
	-build-path "$(TARGET_DIR)" \
	-warnings=none \
	-prefs=build.warn_data_percentage=75 \
	-verbose "$(SKETCH)"

flash:
	$(ESPTOOL) -v -cd nodemcu -cb 115200 -cp $(MONITOR_PORT) -ca 0x00000 -cf $(TARGET_DIR)/$(SKETCH).bin

upload: all flash

clean:
	rm -rf $(TARGET_DIR)

monitor:
	screen $(MONITOR_PORT) 9600
```

closed 2017-08-15 by sej7278

chrisspen, 2018-09-12:
I was also curious about this. I love Arduino-Makefile, and recently purchased a ESP8266 ESP-12E NodeMCU, so I'd like to use them together. Is this unsupported?

sej7278, 2018-09-12:
yup, totally unsupported. either use the makefile above or try the new arduino cli tool. we'd welcome a PR to add ESP8266 support though, as neither of those solutions offers the featureset of arduino-mk.