sudar/Arduino-Makefile #495: Unsupported architecture while trying to access local c++ libraries

TPeterW, 2017-04-23:
Installed avr-gcc through home-brew, version 6.2.0

Trying to include/call some standard c++ files that I wrote before, first avr-gcc told me:

    vector: No such file or directory

because I tried to `#include <vector>`, and I added `-I/usr/local/include/c++/6.3.0` flag, then it asks for `<bits/c++config.h>` and `<Availability>`, and I included `/usr/include/c++/4.2.1` (somehow home-brew gcc doesn't have c++config.h) and `/usr/include`, and finally it says 
    
    In file included from /usr/include/sys/_types.h:32:0,
                     from /usr/include/_types.h:27,
                     from /usr/include/unistd.h:71,
                     from /usr/include/c++/4.2.1/bits/os_defines.h:61,
                     from /usr/include/c++/4.2.1/bits/c++config.h:41,
                     from /usr/local/include/c++/6.3.0/bits/stl_algobase.h:59,
                     from /usr/local/include/c++/6.3.0/vector:60,
                     from datatools.cpp:1:
    /usr/include/sys/cdefs.h:761:2: error: #error Unsupported architecture
     #error Unsupported architecture
      ^~~~~
    In file included from /usr/include/sys/_types.h:33:0,
                     from /usr/include/_types.h:27,
                     from /usr/include/unistd.h:71,
                     from /usr/include/c++/4.2.1/bits/os_defines.h:61,
                     from /usr/include/c++/4.2.1/bits/c++config.h:41,
                     from /usr/local/include/c++/6.3.0/bits/stl_algobase.h:59,
                     from /usr/local/include/c++/6.3.0/vector:60,
                     from datatools.cpp:1:
    /usr/include/machine/_types.h:34:2: error: #error architecture not supported
     #error architecture not supported
      ^~~~~
    In file included from /usr/include/c++/4.2.1/bits/c++config.h:41:0,
                     from /usr/local/include/c++/6.3.0/bits/stl_algobase.h:59,
                     from /usr/local/include/c++/6.3.0/vector:60,
                     from datatools.cpp:1:
    /usr/include/c++/4.2.1/bits/os_defines.h:92:27: error: stray '$' in program
     extern void __dtrace_probe$cxa_runtime$cxa_exception_rethrow$v1(void);
                               ^
    /usr/include/c++/4.2.1/bits/os_defines.h:92:39: error: stray '$' in program
     extern void __dtrace_probe$cxa_runtime$cxa_exception_rethrow$v1(void);
                                           ^
    /usr/include/c++/4.2.1/bits/os_defines.h:92:61: error: stray '$' in program
     extern void __dtrace_probe$cxa_runtime$cxa_exception_rethrow$v1(void);
                                                                 ^
    /usr/include/c++/4.2.1/bits/os_defines.h:93:30: error: stray '$' in program
     extern int __dtrace_isenabled$cxa_runtime$cxa_exception_rethrow$v1(void);
                                  ^
    /usr/include/c++/4.2.1/bits/os_defines.h:93:42: error: stray '$' in program
     extern int __dtrace_isenabled$cxa_runtime$cxa_exception_rethrow$v1(void);
                                              ^
    /usr/include/c++/4.2.1/bits/os_defines.h:93:64: error: stray '$' in program
     extern int __dtrace_isenabled$cxa_runtime$cxa_exception_rethrow$v1(void);
                                                                    ^
    /usr/include/c++/4.2.1/bits/os_defines.h:94:27: error: stray '$' in program
     extern void __dtrace_probe$cxa_runtime$cxa_exception_throw$v1$766f6964202a(void *);
                               ^
    /usr/include/c++/4.2.1/bits/os_defines.h:94:39: error: stray '$' in program
     extern void __dtrace_probe$cxa_runtime$cxa_exception_throw$v1$766f6964202a(void *);
                                           ^
    /usr/include/c++/4.2.1/bits/os_defines.h:94:59: error: stray '$' in program
     extern void __dtrace_probe$cxa_runtime$cxa_exception_throw$v1$766f6964202a(void *);
                                                               ^
    /usr/include/c++/4.2.1/bits/os_defines.h:94:62: error: stray '$' in program
     extern void __dtrace_probe$cxa_runtime$cxa_exception_throw$v1$766f6964202a(void *);
                                                                  ^
    /usr/include/c++/4.2.1/bits/os_defines.h:95:30: error: stray '$' in program
     extern int __dtrace_isenabled$cxa_runtime$cxa_exception_throw$v1(void);
                                  ^
    /usr/include/c++/4.2.1/bits/os_defines.h:95:42: error: stray '$' in program
     extern int __dtrace_isenabled$cxa_runtime$cxa_exception_throw$v1(void);
                                              ^
    /usr/include/c++/4.2.1/bits/os_defines.h:95:62: error: stray '$' in program
     extern int __dtrace_isenabled$cxa_runtime$cxa_exception_throw$v1(void);
                                                                  ^
    In file included from /usr/include/sys/signal.h:82:0,
                     from /usr/include/sys/wait.h:109,
                     from /usr/include/stdlib.h:65,
                     from /usr/local/include/c++/6.3.0/cstdlib:75,
                     from /usr/local/include/c++/6.3.0/ext/string_conversions.h:41,
                     from /usr/local/include/c++/6.3.0/bits/basic_string.h:5402,
                     from /usr/local/include/c++/6.3.0/string:52,
                     from /usr/local/include/c++/6.3.0/bits/locale_classes.h:40,
                     from /usr/local/include/c++/6.3.0/bits/ios_base.h:41,
                     from /usr/local/include/c++/6.3.0/ios:42,
                     from /usr/local/include/c++/6.3.0/ostream:38,
                     from /usr/local/include/c++/6.3.0/iostream:39,
                     from datatools.cpp:2:
    /usr/include/machine/signal.h:34:2: error: #error architecture not supported
     #error architecture not supported
      ^~~~~
    In file included from /usr/include/sys/signal.h:146:0,
                     from /usr/include/sys/wait.h:109,
                     from /usr/include/stdlib.h:65,
                     from /usr/local/include/c++/6.3.0/cstdlib:75,
                     from /usr/local/include/c++/6.3.0/ext/string_conversions.h:41,
                     from /usr/local/include/c++/6.3.0/bits/basic_string.h:5402,
                     from /usr/local/include/c++/6.3.0/string:52,
                     from /usr/local/include/c++/6.3.0/bits/locale_classes.h:40,
                     from /usr/local/include/c++/6.3.0/bits/ios_base.h:41,
                     from /usr/local/include/c++/6.3.0/ios:42,
                     from /usr/local/include/c++/6.3.0/ostream:38,
                     from /usr/local/include/c++/6.3.0/iostream:39,
                     from datatools.cpp:2:
    /usr/include/machine/_mcontext.h:31:2: error: #error architecture not supported
     #error architecture not supported
      ^~~~~
    In file included from /usr/include/sys/wait.h:186:0,
                     from /usr/include/stdlib.h:65,
                     from /usr/local/include/c++/6.3.0/cstdlib:75,
                     from /usr/local/include/c++/6.3.0/ext/string_conversions.h:41,
                     from /usr/local/include/c++/6.3.0/bits/basic_string.h:5402,
                     from /usr/local/include/c++/6.3.0/string:52,
                     from /usr/local/include/c++/6.3.0/bits/locale_classes.h:40,
                     from /usr/local/include/c++/6.3.0/bits/ios_base.h:41,
                     from /usr/local/include/c++/6.3.0/ios:42,
                     from /usr/local/include/c++/6.3.0/ostream:38,
                     from /usr/local/include/c++/6.3.0/iostream:39,
                     from datatools.cpp:2:
    /usr/include/machine/endian.h:37:2: error: #error architecture not supported
     #error architecture not supported
      ^~~~~
    In file included from /usr/include/stdlib.h:251:0,
                     from /usr/local/include/c++/6.3.0/cstdlib:75,
                     from /usr/local/include/c++/6.3.0/ext/string_conversions.h:41,
                     from /usr/local/include/c++/6.3.0/bits/basic_string.h:5402,
                     from /usr/local/include/c++/6.3.0/string:52,
                     from /usr/local/include/c++/6.3.0/bits/locale_classes.h:40,
                     from /usr/local/include/c++/6.3.0/bits/ios_base.h:41,
                     from /usr/local/include/c++/6.3.0/ios:42,
                     from /usr/local/include/c++/6.3.0/ostream:38,
                     from /usr/local/include/c++/6.3.0/iostream:39,
                     from datatools.cpp:2:
    /usr/include/machine/types.h:37:2: error: #error architecture not supported
     #error architecture not supported

I didn't change the architecture so it should still be avr, somehow it's not recognising the standard c++ libraries

sej7278, 2017-04-23:
looks like that's gcc not avr-gcc. try using the avr-gcc from the ide instead of homebrew, or install 4.9 not 6.3 or look at [bare-arduino-project](https://github.com/ladislas/Bare-Arduino-Project)

ladislas, 2017-04-23:
vectors are not supported on avr.

you should only use avr-libc.

install all of that following the instructions from https://github.com/osx-cross/homebrew-avr

TPeterW, 2017-04-24:
@sej7278 I did install avr-gcc and related libraries tho, from home-brew but 'include /my/path/to/Arduino.mk, avr-gcc was having trouble finding gcc libraries, is there any way I can use gcc libraries on an Arduino?

ladislas, 2017-04-24:
> is there any way I can use gcc libraries on an Arduino?

@TPeterW - short answer: no

long answer: gcc-libraries are made for computers, not for AVR Embedded Systems.

as explained [here](http://www.nongnu.org/avr-libc/user-manual/overview.html):

> AVR-Libc provides many of the same functions found in a regular Standard C Library and many additional library functions that is specific to an AVR. Some of the Standard C Library functions that are commonly used on a PC environment have limitations or additional issues that a user needs to be aware of when used on an embedded system.

TPeterW, 2017-04-24:
@ladislas So I pretty much cannot use opencv Mat and stuff on the embedded systems right?

I'm just gonna go ahead and close this issue. Thank you for answering.

closed 2017-04-24 by TPeterW