sudar/Arduino-Makefile #497: Not detecting library directory

chrisspen, 2017-05-28:
I have a number of extra libraries installed in `/home/$USER/Arduino/libraries`, which I can compile with from inside the Arduino IDE. However, if I try to compile with the makefile, I get the error:

    The following libraries specified in ARDUINO_LIBS could not be found (searched USER_LIB_PATH and ARDUINO_LIB_PATH)

Please add support for custom library paths.

I tried setting USER_LIB_PATH, but this doesn't appear to support `$USER` or `~`, and my application doesn't allow hard-coding the username in the path, since it's not a static value.

sej7278, 2017-05-28:
Use $HOME instead of ~ in Makefiles

chrisspen, 2017-05-28:
Ok. In that case, the docs should probably be updated. [This](https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#user_lib_path) explicitly says `~` should work. But you're right, it looks like `$(HOME)` and `$(USER)` variables are available.

Is it possible to include relative directories? I have a library in the same directory as my makefile, and to simplify installation and maintainability, I'd prefer to avoid having to copy it into my sketchbook.

However, if I used the setting:

    USER_LIB_PATH = .

it still complains that it can't find the library.

sej7278, 2017-05-28:
have a look through the other issues, there's a few regarding libraries in the sketch directory, don't think its possible

chrisspen, 2017-05-28:
Is appending multiple paths to `USER_LIB_PATH` supported? [This blog post](http://hackaday.com/2015/10/01/arduino-development-theres-a-makefile-for-that/) raves about Arduino-Makefile, claiming it has this ability, but when I use their syntax, e.g.

    USER_LIB_PATH = /home/$(USER)/Arduino/libraries
    USER_LIB_PATH += .

it causes make to not find any libraries.

sej7278, 2017-06-01:
fixed the tilde documentation issue in PR #500 now merged

chrisspen, 2017-06-13:
My end goal is to allow inclusion of a local `./lib` directory, so I don't have to copy every project's libraries into the global sketchbook directory, since that becomes difficult to maintain and inevitably results in conflicts.

Is there any way to currently accomplish this with Arduino-Makefile?

ladislas, 2017-06-14:
@chrisspen - I did something similar with the [bare arduino project](https://github.com/ladislas/Bare-Arduino-Project)

ladislas, 2017-06-14:
@chrisspen  and actually you'll need to take a look at our fork of the arduino-makefile https://github.com/weareleka/Arduino-Makefile

chrisspen, 2017-06-19:
Is there no solution for this?

My code looks like:

    src/
        lib/
            OtherLibrary/
                some_include.h
                some_other_include.h
        main.ino

My makefile looks like:

    BOARD_TAG    = uno
    MONITOR_PORT = /dev/ttyACM0
    AVRDUDE_OPTS = -v
    ARDUINO_LIBS = OtherLibrary
    ARDUINO_SKETCHBOOK = $(realpath .)/lib
    USER_LIB_PATH = $(realpath .)/lib
    include /usr/share/arduino/Arduino.mk

But when I run `make`, I get the error:

    In file included from main.ino:40:0:
    some_include.h:3:45: fatal error: OtherLibrary/some_other_include.h: No such file or directory
     #include "OtherLibrary/some_include.h"

Why isn't it finding secondary includes in either the ARDUINO_SKETCHBOOK or USER_LIB_PATH?

I noticed that, for some odd reason, the g++ command outputted doesn't actually include either of these paths as a `-I` option, and if I manually add that path and call g++ myself, it compiles perfectly.

Am I not configuring arduino-mk correctly, or is this a bug? This seems like a very simple feature that isn't working.

chrisspen, 2017-06-19:
The problem seems to be that arduino-mk completely ignores ARDUINO_SKETCHBOOK, and it doesn't directly include USER_LIB_PATH but instead uses that as a base folder to include the sub-folders that match items in ARDUINO_LIBS. So it's not finding `OtherLibrary/some_include.h` because because it's not using `../OtherLibrary` as a search path.

However, why isn't ARDUINO_SKETCHBOOK being used? If I comment out my custom USER_LIB_PATH, then I get the completely nonsensical error message:

     *** The following libraries specified in ARDUINO_LIBS could not be found (searched USER_LIB_PATH, ARDUINO_LIB_PATH and ARDUINO_PLATFORM_LIB_PATH): OtherLibrary

Yes...it's be found if it checked ARDUINO_SKETCHBOOK. Why isn't that being used as a search path?

sej7278, 2017-06-19:
@ladislas i thought you wrote some sort of library autodetection script but your fork seems identical to upstream now?

@chrisspen  ```ARDUINO_SKETCHBOOK``` is not the right place to search for libraries and shouldn't be dynamic between sketches. you probably want to create another variable to achieve what you want.

ladislas, 2017-06-19:
@sej7278 I'm following upstream (you) closely and then reapplying my changes each time.

you have to checkout the auto-lib branch: https://github.com/weareleka/Arduino-Makefile/tree/auto-lib

here are the differences

https://github.com/weareleka/Arduino-Makefile/commit/42c8bdf6368250f199e3a26a5566370e44bb7786