sudar/Arduino-Makefile #498: avr-g++: error: missing device or architecture after '-mmcu='

chrisspen, 2017-05-28:
I'm attempting to compile a small ROS Arduino project with the Makefile:

    BOARD_TAG    = uno_pro
    MONITOR_PORT = /dev/ttyACM0
    AVRDUDE_OPTS = -v
    ARDUINO_LIBS = ros_lib Wire Adafruit_BNO055
    USER_LIB_PATH = /home/$(USER)/Arduino/libraries
    include /usr/share/arduino/Arduino.mk

I have the head of Arduino-Makefile checked out to `/usr/share/arduino`.

I have Arduino IDE 1.8.2 installed to `/usr/share/arduino`.

I'm compiling for the Arduino Uno*Pro, which requires some additional files installed, as [explained here.](http://www.hobbytronics.co.uk/arduino-uno-pro)

When I run `make`, I get the error:

    -------------------------
    Arduino.mk Configuration:
    - [AUTODETECTED]       CURRENT_OS = LINUX 
    - [AUTODETECTED]       ARDUINO_DIR = /usr/share/arduino 
    - [COMPUTED]           ARDMK_DIR = /usr/share/arduino (relative to Common.mk)
    - [AUTODETECTED]       ARDUINO_VERSION = 105 
    - [DEFAULT]            ARCHITECTURE =  
    - [DEFAULT]            ARDMK_VENDOR = arduino 
    - [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/chris/.arduino/preferences.txt 
    - [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/chris/Arduino
    - [BUNDLED]            AVR_TOOLS_DIR = /usr/share/arduino/hardware/tools/avr (in Arduino distribution)
    - [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
    - [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino//variants (from ARDUINO_DIR)
    - [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino//boards.txt (from ARDUINO_DIR)
    - [USER]               USER_LIB_PATH = /home/chris/Arduino/libraries 
    - [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
    - [USER]               BOARD_TAG = uno_pro 
    - [COMPUTED]           CORE =  (from build.core)
    - [COMPUTED]           VARIANT =  (from build.variant)
    - [COMPUTED]           OBJDIR = build-uno_pro (from BOARD_TAG)
    - [DEFAULT]            ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino//cores/arduino 
    - [ASSUMED]            MONITOR_BAUDRATE = 9600 
    - [DEFAULT]            OPTIMIZATION_LEVEL = s 
    - [DEFAULT]            MCU_FLAG_NAME = mmcu 
    - [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
    - [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
    - [COMPUTED]           DEVICE_PATH = /dev/ttyACM0 (from MONITOR_PORT)
    - [DEFAULT]            FORCE_MONITOR_PORT =  
    - [AUTODETECTED]       Size utility: AVR-aware for enhanced output
    -
    -                      ARDUINO_LIBS =
    - [USER]                 Adafruit_BNO055
    - [USER]                 ros_lib
    - [SYSTEM]               Wire
    - [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino//bootloaders (from ARDUINO_DIR)
    - [COMPUTED]           ARDMK_VERSION = 1.5 
    - [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
    -------------------------
    mkdir -p build-uno_pro
    /usr/share/arduino/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu= -DF_CPU= -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/usr/share/arduino/hardware/arduino//variants/    -I/usr/share/arduino/libraries/Wire -I/usr/share/arduino/libraries/Wire/utility    -I/home/chris/Arduino/libraries/Adafruit_BNO055 -I/home/chris/Arduino/libraries/Adafruit_BNO055/utility   -I/home/chris/Arduino/libraries/ros_lib -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto main.ino -o build-uno_pro/main.ino.o
    avr-g++: error: missing device or architecture after '-mmcu='
    /usr/share/arduino/Arduino.mk:1251: recipe for target 'build-uno_pro/main.ino.o' failed
    make: *** [build-uno_pro/main.ino.o] Error 1

I'm assuming this error is due to the malformed `ARDUINO_CORE_PATH` and `BOARDS_TXT` values, which have a double-slash and appear to be missing the `/avr/` sub-folder. e.g. `BOARDS_TXT` should be `/usr/share/arduino/hardware/arduino/avr/boards.txt`. This is causing the wrong `boards.txt` to be loaded, which contains the `mmcu` value need to compile.

Why are these paths malformed? Does Arduino-Makefile not support Arduino 1.8.2?

sej7278, 2017-05-28:
you don't have 1.8.2 installed in /usr/share/arduino, you have 1.0.5 there.

set ```ARDUINO_DIR``` to the path to 1.8.2 and set ```ALTERNATE_CORE``` to the core's location, as per the [documentation](https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#alternate_core)

chrisspen, 2017-05-31:
Yeah, I had to clear /usr/share/arduino and reinstall 1.8.2. It's frustrating that the official Ubuntu package uses such an ancient version.

closed 2017-05-31 by chrisspen

sej7278, 2017-05-31:
its easier just to untar 1.8.2 to some directory and leave the ubuntu 1.0.5 in /usr/share/arduino, for anything AVR there's nothing much that 1.8.2 offers over 1.0.5

if you want a newer ubuntu (well, debian) package then nag arduino to sort out their licensing:

https://github.com/arduino/Arduino/pull/2703

https://github.com/arduino/Arduino/issues/1117