sudar/Arduino-Makefile #499: Add support of externally supported boards like Pro Micro

MilanV, 2017-05-28:
Some companies provide an official support of their boards, like [Sparkfun does](https://github.com/sparkfun/Arduino_Boards/). Arduino IDE knows to add that as an "addon". How can I add this to the arduino-mk as well, please? Either manually or, better, a script taking that .json file and doing proper actions so, as a result, 'arduino-mk show-boards' shows those newly defined ones among the original ones.

sej7278, 2017-05-29:
See [Blink example](https://github.com/sudar/Arduino-Makefile/blob/master/examples/Blink/Makefile#L41)

You simply download the board's core to your hardware directory instead of using the Boards Manager method. So you end up with something like this in ~/sketchbook/hardware/

```
sparkfun/
└── avr
    ├── boards.txt
    ├── bootloaders
    ├── driver
    ├── platform.txt
    ├── signed_driver
    └── variants
```

And then your Makefile should look something like this:

```make
MONITOR_PORT        = /dev/ttyACM0
BOARD_TAG           = promicro16
ALTERNATE_CORE      = sparkfun

include /usr/share/arduino/Arduino.mk
```

Note with the Pro Micro you can just use ```BOARD_TAG=leonardo``` the sparkfun core doesn't add anything much.

MilanV, 2017-06-01:
Thank you for your fast and detailed answer.

I start with your last note: I used "leonardo" board definition at the beginning, and the sketch was uploaded and worked. But the ttyACM device disappeared and, since then, the only moment to see it was those 3 seconds when bootloader creates it. After that, not even IDE could help (its detection of the connected board is too slow). So it would be more than useful to have Arduino-Makefile working and this case documented in this package so next arduino-greenhorns don't fall into that deep water as I did :) It's a good message everything is already implemented and it's just a matter of the documentation.

Here is what I found when trying your advice:

* the current "package" from sparkfun contains avr/<version>/... drectory tree, I had to remove that <version> subdirectory level (move everything one dir up) because Arduino.mk was confused by that

* there is no "promicro16" or "promicro8" model (any longer), they are defined as "cpu menu" of one "promicro" model. To use that with Arduino.mk, we need this configuration (8MHz variant):
    ALTERNATE_CORE      = sparkfun
    BOARD_TAG           = promicro
    BOARD_SUB           = 8MHzatmega32U4

With this, everything looks configured properly. However, during the compilation of arduino core files, I get an error. It seems some macro(s) are not defined properly. Please, can you help me again?

    /home/milan/arduino-1.8.2/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega32u4 -DF_CPU=8000000L -DARDUINO=182 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/milan/arduino-1.8.2/hardware/arduino/avr/cores/arduino -I/home/milan/sketchbook/hardware/sparkfun/avr/variants/promicro    -Wall -ffunction-sections -fdata-sections -Os -DUSB_VID=0x1b4f -DUSB_PID= -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto /home/milan/arduino-1.8.2/hardware/arduino/avr/cores/arduino/USBCore.cpp -o build-promicro-8MHzatmega32U4/core/USBCore.cpp.o
    In file included from /home/milan/arduino-1.8.2/hardware/arduino/avr/cores/arduino/USBAPI.h:44:0,
                     from /home/milan/arduino-1.8.2/hardware/arduino/avr/cores/arduino/USBCore.cpp:20:
    /home/milan/arduino-1.8.2/hardware/arduino/avr/cores/arduino/USBCore.h:267:70: error: expected primary-expression before ',' token
      { 18, 1, USB_VERSION, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                          ^
    /home/milan/arduino-1.8.2/hardware/arduino/avr/cores/arduino/USBCore.cpp:74:2: note: in expansion of macro 'D_DEVICE'
      D_DEVICE(0xEF,0x02,0x01,64,USB_VID,USB_PID,0x100,IMANUFACTURER,IPRODUCT,ISERIAL,1);
      ^
    /home/milan/Arduino-Makefile/Arduino.mk:1273: recipe for target 'build-promicro-8MHzatmega32U4/core/USBCore.cpp.o' failed

My Makefile:

    ARDUINO_DIR	    = /home/milan/arduino-1.8.2
    
    MONITOR_PORT        = /dev/ttyACM0
    BOARD_TAG           = promicro
    BOARD_SUB	    = 8MHzatmega32U4
    ALTERNATE_CORE      = sparkfun
    
    include ~/Arduino-Makefile/Arduino.mk

MilanV, 2017-06-01:
More details: the error is about USB_PID not defined. This should be in the "build.pid" config item of the board definition. There is the following in boards.txt:

    ################################################################################
    ################################## Pro Micro ###################################
    ################################################################################
    promicro.name=SparkFun Pro Micro
    
    promicro.upload.tool=avrdude
    promicro.upload.protocol=avr109
    promicro.upload.maximum_size=28672
    promicro.upload.maximum_data_size=2560
    promicro.upload.speed=57600
    promicro.upload.disable_flushing=true
    promicro.upload.use_1200bps_touch=true
    promicro.upload.wait_for_upload_port=true
    
    promicro.bootloader.tool=avrdude
    promicro.bootloader.unlock_bits=0x3F
    promicro.bootloader.lock_bits=0x2F
    promicro.bootloader.low_fuses=0xFF
    promicro.bootloader.high_fuses=0xD8
    
    promicro.build.board=AVR_PROMICRO
    promicro.build.core=arduino:arduino
    promicro.build.variant=promicro
    promicro.build.mcu=atmega32u4
    promicro.build.usb_product="SparkFun Pro Micro"
    promicro.build.vid=0x1b4f
    promicro.build.extra_flags={build.usb_flags}
    
    ######################### Pro Micro 3.3V / 8MHz ################################
    promicro.menu.cpu.8MHzatmega32U4=ATmega32U4 (3.3V, 8 MHz)
    
    promicro.menu.cpu.8MHzatmega32U4.build.pid.0=0x9203
    promicro.menu.cpu.8MHzatmega32U4.build.pid.1=0x9204
    promicro.menu.cpu.8MHzatmega32U4.build.pid=0x9204
    promicro.menu.cpu.8MHzatmega32U4.build.f_cpu=8000000L
    
    promicro.menu.cpu.8MHzatmega32U4.bootloader.extended_fuses=0xFE
    promicro.menu.cpu.8MHzatmega32U4.bootloader.file=caterina/Caterina-promicro8.hex
    

I can see "build.vid" set globally for all promicro boards and build.pid set in the local (menu.cpu) section. Is it a problem of those local attributes parsing? Or should I specify something more in the configuration?

sej7278, 2017-06-01:
you shouldn't have removed the avr subdirectory.

you have https://github.com/sparkfun/Arduino_Boards/tree/master/sparkfun/avr

so the directory structure is the same as my example:

```
hardware/
├── sparkfun/
│   └── avr/
│       ├── boards.txt
│       ├── bootloaders/
│       ├── driver/
│       ├── platform.txt
│       ├── signed_driver/
│       └── variants/
```

Which then gives:

```
$ make ARDUINO_QUIET=1 show_boards
atmega128rfa1   SparkFun ATmega128RFA1 Dev Board
digitalsandbox  SparkFun Digital Sandbox
fiov3           SparkFun Fio v3
makeymakey      SparkFun MaKey MaKey
megapro         SparkFun Mega Pro
promicro        SparkFun Pro Micro
qduinomini      Qduino Mini
RedBoard        SparkFun RedBoard
RedBot          SparkFun RedBot
Serial7Seg      SparkFun Serial 7-Segment Display

$ make ARDUINO_QUIET=1 show_submenu
megapro   atmega25603V3    ATmega2560 (3.3V / 8 MHz)
megapro   atmega25605V     ATmega2560 (5V / 16 MHz)
promicro  16MHzatmega32U4  ATmega32U4 (5V, 16 MHz)
promicro  8MHzatmega32U4   ATmega32U4 (3.3V, 8 MHz)

```

So it seems they've changed the tag/sub as you say, and they've messed with the usb vid/pid settings, moving pid into a submenu but not vid which is pretty annoying! this seems to work for me:

```make
MONITOR_PORT = /dev/ttyACM0
BOARD_TAG = promicro
BOARD_SUB = 16MHzatmega32U4
ALTERNATE_CORE = sparkfun
USB_VID = 0x1b4f
USB_PID = 0x9206
```

this patch to Arduino.mk fixes autodetection for me without breaking the 1.0 core or the 1.6ish core i was using that's different to your 1.8ish core. i'll make a PR in a bit, can you try it out?

```diff
diff --git a/Arduino.mk b/Arduino.mk
index 9554f35..5b0af0f 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -628,8 +628,12 @@ ifeq ($(strip $(NO_CORE)),)
             USB_VID = $(call PARSE_BOARD,$(BOARD_TAG),build.vid)
         endif
 
+        # coping with 2-3 methods sparkfun use for usb.pid
         ifndef USB_PID
-            USB_PID = $(call PARSE_BOARD,$(BOARD_TAG),build.pid)
+            USB_PID := $(call PARSE_BOARD,$(BOARD_TAG),build.pid)
+            ifndef USB_PID
+                USB_PID := $(call PARSE_BOARD,$(BOARD_TAG),menu.(chip|cpu).$(BOARD_SUB).build.pid)
+            endif
         endif
     endif
```

MilanV, 2017-06-01:
Thanks, Simon.
The thing about moving subtree one dir up in the sparkfun package was just a misunderstanding, I forgot to escape "\<" and "\>" characters. So we both agree on the same: avr/ dir must be there and the one with some numeric version name not.
I can get the exactly same list of boards as you etc.

About your patch: yes, I may confirm it works. Great, thanks for so fast fix!
Please, can you also add our findings to the documentation so new users of boards defined out of the main boards.txt have a guide? When having also that, we can close my issue. Thanks!

sej7278, 2017-06-01:
huh yeah well i await your documentation PR then ;-) but it would be nice to have more info about alternate cores i guess.

i'll merge my PR and its up to you if you want to close this or not

MilanV, 2017-06-01:
OK, please check my PR #501

closed 2017-06-02 by MilanV