sudar/Arduino-Makefile #504: Dynamically calculating MONITOR_PORT?

chrisspen, 2017-06-19:
Is there any way to dynamically calculate `MONITOR_PORT` while running `make upload`?

I have several Arduino's connected to my host, and I've written a Python script to figure how which ones correspond to each `/dev/ttyACM*` file. I'd like to plug in this script to the makefile so when I run `make upload` it will automatically set `MONITOR_PORT` to the right file. However, I can't find any documentation on how to do this.

sej7278, 2017-06-19:
You could run your python script and pipe the input into MONITOR_PORT in your local Makefile in a similar way as we call shell commands within Arduino.mk, e.g.

```make
MONITOR_PORT = $(shell /path/to/my/script.py)
```

There's also [PRE_BUILD_HOOK](https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#pre_build_hook) which may be of interest.

sej7278, 2017-07-31:
can we close this?

chrisspen, 2017-07-31:
Thanks for the update. It might be helpful if the `make monitor` was explicitly mentioned in the README.

closed 2017-07-31 by chrisspen