sudar/Arduino-Makefile #506: Unable to upload to Leonardo using current bootloader

chrisspen, 2017-06-26:
I burnt the most recent bootloader using the Arduino IDE 1.8.2 onto a Leonardo, and now `make upload` fails with the error:

    mkdir -p build-leonardo
    make reset
    make[1]: Entering directory '/usr/local/myproject/firmware/main'
    /usr/share/arduino/bin/ard-reset-arduino --caterina  /dev/ttyACM0
    make[1]: Leaving directory '/usr/local/myproject/firmware/main'
    make do_upload
    make[1]: Entering directory '/usr/local/myproject/firmware/main'
    /usr/share/arduino/hardware/tools/avr/bin/avrdude -v -p atmega32u4 -C /usr/share/arduino/hardware/tools/avr/etc/avrdude.conf -D -c avr109 -b 57600 -P /dev/ttyACM0 \
		    -U flash:w:build-leonardo/main.hex:i

    avrdude: Version 6.3, compiled on Jan 17 2017 at 11:00:16
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "/usr/share/arduino/hardware/tools/avr/etc/avrdude.conf"
             User configuration file is "/home/chris/.avrduderc"
             User configuration file does not exist or is not a regular file, skipping

             Using Port                    : /dev/ttyACM0
             Using Programmer              : avr109
             Overriding Baud Rate          : 57600
             AVR Part                      : ATmega32U4
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
               flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
               lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : butterfly
             Description     : Atmel AppNote AVR109 Boot Loader

    Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding

    avrdude: butterfly_recv(): programmer is not responding
    avrdude: butterfly_recv(): programmer is not responding
    avrdude: butterfly_recv(): programmer is not responding
    avrdude: butterfly_recv(): programmer is not responding
    avrdude: butterfly_recv(): programmer is not responding
    Found programmer: Id = ""; type = 
        Software Version = .; Hardware Version = �.�
    avrdude: butterfly_recv(): programmer is not responding
    avrdude: butterfly_recv(): programmer is not responding
    avrdude: error: buffered memory access not supported. Maybe it isn't
    a butterfly/AVR109 but a AVR910 device?
    avrdude: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.

    avrdude: butterfly_recv(): programmer is not responding
    avrdude: error: programmer did not respond to command: leave prog mode
    avrdude: butterfly_recv(): programmer is not responding
    avrdude: error: programmer did not respond to command: exit bootloader

    avrdude done.  Thank you.

    /usr/share/arduino/Arduino.mk:1462: recipe for target 'do_upload' failed

It looks like `ard-reset-arduino` works, because the Leonardo's status LED begins to flash, but instead of uploading, it just sits there, and then the Arduino recovers from the reset, and then avrdude starts trying to upload, after which it's too late.

Does Arduino-Makefile support the most recent bootloader?

doronbehar, 2017-09-27:
Hmm, weird. I had the exact same issue after I tried to burn a bootloader with the command line as explained [this manual](https://learn.sparkfun.com/tutorials/installing-an-arduino-bootloader#uploading-code---hard-way). The manual is targeted for `Uno` boards and not `Leonardo` so I believe this is the source of the problem.

As for fixing your problem, are you sure you've selected the correct board in the `Tools` drop-down menu in the IDE? before burning the bootloader?

sej7278, 2017-09-27:
also don't forget that the sf pro micro and arduino leonardo have completely different bootloaders and usb vid/pid. the sparkfun bootloader is very flaky due to their iffy hacks to reduce the boot time

hirikilabs, 2018-05-23:
I have the same problem with the Arduino micro (atmega32u4). IDE uploading works, but arduino-makefile fails with error:

```
mkdir -p build-micro
make reset
make[1]: se entra en el directorio '/home/zako/ownCloud/Devel/Arduino/Prueba_MIDI'
/home/zako/Apps/arduino-makefile/bin/ard-reset-arduino --caterina /dev/ttyACM0
make[1]: se sale del directorio '/home/zako/ownCloud/Devel/Arduino/Prueba_MIDI'
make do_upload
make[1]: se entra en el directorio '/home/zako/ownCloud/Devel/Arduino/Prueba_MIDI'
/home/zako/Apps/arduino-1.8.4/hardware/tools/avr/bin/avrdude -q -V -p atmega32u4 -C /home/zako/Apps/arduino-1.8.4/hardware/tools/avr/etc/avrdude.conf -D -c avr109 -b 57600 -P /dev/ttyACM0 \
                -U flash:w:build-micro/Prueba_MIDI.hex:i

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x3f0d0d
avrdude: Expected signature for ATmega32U4 is 1E 95 87
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

make[1]: *** [/home/zako/Apps/arduino-makefile/Arduino.mk:1571: do_upload] Error 1
make[1]: se sale del directorio '/home/zako/ownCloud/Devel/Arduino/Prueba_MIDI'
make: *** [/home/zako/Apps/arduino-makefile/Arduino.mk:1559: upload] Error 2
```

chrisspen, 2018-05-23:
I ended up abandoning the Leonardo, because of this issue. This can happen with any Arduino, but you can usually fix it by upgrading the IDE and/or writing a blank sketch to the Arduino. I usually have to resort to using the ICSP to re-burn the bootloader and reset fuses.

However, because the Leonardo doesn't use a dedicated chip for the USB/serial, a corrupted sketch or bootloader can prevent you from fixing it. A few times, I was able upload a sketch immediately after pressing reset, but it became such a hassle, and this issue seemed to happen once out of every 10 uploads, that it just became easier to ditch the Leonardo and stick with the Uno, Nano or Mega. In general, I recommend avoiding any Arduino using the m32u4 chip, which fortunately isn't very many.

I'm closing my question since this issue is an Arduino problem, not a Makefile problem.

closed 2018-05-23 by chrisspen