sudar/Arduino-Makefile #513: the macro ARDUINO have a wrong value

YruamaLairba, 2017-08-03:
When compiling with arduino-makefile, the macro ARDUINO doesn't expand correctly inside a .ino . For example, when compiling with arduino 1.8.3, ARDUINO expand to 10803 with the IDE but i get 183 with the arduino-makefile. In my case, this prevent me to use the MIDIUSB librairy because it test the value of ARDUINO and throw an error if this value is too low.

A quick workaround is set manually ARDUINO_VERSION, but i get other problems with the arduino micro and arduino 1.6.9. The workaround work well with arduino 1.8.3

sej7278, 2017-08-03:
i'm not aware there is an ARDUINO macro, is this documented somewhere?

YruamaLairba, 2017-08-04:
i can't find any documentation about the ARDUINO macro or runtime.ide.version, but the value of ARDUINO macro is zero-padded at least since 1.6.9.

After little more test, my problem with arduino-makefile+arduino-1.6.9+micro doesn't seems related to the current problem as it don't work with basic sketch, no matter of ARDUINO values, so don't care about it for the current issue.

sej7278, 2017-08-04:
"micro" works for me with ide 1.8.3, been quite a while since i tried 1.6.9 but i know 1.6.10 worked fine, i suspect its a bug with a library or your setup is odd - i assume you're not trying to use a board installed using boards manager, as that's not supported. use the hardware directory inside your sketchbook.

```-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/user/arduino-1.8.3 
- [COMPUTED]           ARDMK_DIR = /home/user/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 183 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/user/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/user/arduino16 (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /home/user/arduino-1.8.3/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /home/user/arduino-1.8.3/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/user/arduino-1.8.3/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/user/arduino-1.8.3/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /home/user/arduino-1.8.3/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/user/arduino16/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = micro 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = micro (from build.variant)
- [COMPUTED]           OBJDIR = build-micro (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /home/user/arduino-1.8.3/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects -fdiagnostics-color 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /home/user/arduino-1.8.3/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6 
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
-------------------------
```

faxm0dem, 2020-02-06:
I can confirm this. It prevents for instance some libraries to build e.g. https://github.com/MHeironimus/ArduinoJoystickLibrary

As a workarount, you can force the version in your Makefile:

```makefile
BOARD_TAG    = micro
ARDUINO_LIBS = Joystick
ARDUINO_VERSION = 10810
ARDUINO_DIR = /usr/local/arduino-1.8.10
include /path/to/Arduino.mk
```