sudar/Arduino-Makefile #515: USER, SYSTEM and PLATFORM libs on teensy

eeweegh, 2017-08-06:
Apologies if this has come up in another thread, I looked around but could not find a similar issue.
I was trying to build the Chord Organ code (https://github.com/TomWhitwell/Chord-Organ) on OSX with v1.5.2, but got an error when compiling the Teensy SD library (a mismatch between a .h declaration and a .cpp definition). That turned out to be because the compiler was using the wrong (SYSTEM) include directory, rather than the right (PLATFORM) directory.

I hacked around that like this (moving the SYS_LIBS calculation below the PLATFORM_LIBS calculation, and adding an extra variable to the filter-out):
```
SYS_LIBS       := $(sort $(wildcard $(patsubst %,$(ARDUINO_LIB_PATH)/%,$(filter-out $(USER_LIB_NAMES) $(PLATFORM_LIB_NAMES),$(ARDUINO_LIBS)))))
```

Not very clean, as it assumes that PLATFORM_LIB_NAMES will always exist - but it works, and I'd like your comment before I do any further work on this.

FYI, my Makefile is this:
```
BOARD_TAG = teensy31
ARDUINO_LIBS += Bounce2 Audio SPI SD SerialFlash Wire EEPROM
include /usr/local/opt/arduino-mk/Teensy.mk
```

make output looks like this (note the double SD)
```
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Java 
- [USER]               ARDMK_DIR = /usr/local/Cellar/arduino-mk/1.5.2 
- [AUTODETECTED]       ARDUINO_VERSION = 183 
- [USER]               ARCHITECTURE = arm 
- [USER]               ARDMK_VENDOR = teensy 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/erik/Library/Arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/erik/Documents/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /Applications/Arduino.app/Contents/Java/hardware/tools/arm 
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [USER]               ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries 
- [COMPUTED]           ARDUINO_VAR_PATH = /Applications/Arduino.app/Contents/Java/hardware/teensy/arm/variants (from ARDUINO_DIR)
- [USER]               BOARDS_TXT = /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/boards.txt 
- [DEFAULT]            USER_LIB_PATH = /Users/erik/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = teensy31 
- [COMPUTED]           CORE = teensy3 (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-teensy31 (from BOARD_TAG)
- [USER]               ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [USER]               MCU_FLAG_NAME = mcpu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
-
-                      ARDUINO_LIBS =
- [SYSTEM]               SD
- [PLATFORM]             Audio
- [PLATFORM]             Bounce2
- [PLATFORM]             EEPROM
- [PLATFORM]             SD
- [PLATFORM]             SPI
- [PLATFORM]             SerialFlash
- [PLATFORM]             Wire
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/teensy/arm/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 5.4.1 (arm-none-eabi-gcc)
```

Thanks for looking int this!

sej7278, 2017-08-06:
have you tried ```ARDUINO_LIBS=``` rather than appending with += ? or even letting it auto-detect?

eeweegh, 2017-08-07:
yes. I started off with auto-detection, but that didn't work (compiles
would complain of unknown include files).
The += was a late change (I used = before), but should not really make a
difference, as ARDUINO_LIBS is empty anyway before the include.