sudar/Arduino-Makefile #517: Missing Nano Support

chrisspen, 2017-08-15:
How do you compile for the Arduino Nano? Running `make showboards` shows the Nano is supported, and that its `BOARD_TAG` setting is simply `nano`, but when I use this and run `make`, I get the error:

    mkdir -p build-nano
    /usr/share/arduino/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu= -DF_CPU=16000000L -DARDUINO=182 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/eightanaloginputs    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto main.ino -o build-nano/main.ino.o
    avr-g++: error: missing device or architecture after '-mmcu='
    /usr/share/arduino/Arduino.mk:1251: recipe for target 'build-nano/main.ino.o' failed
    make: *** [build-nano/main.ino.o] Error 1

This is my makefile:

    BOARD_TAG    = nano
    MONITOR_PORT = /dev/ttyACM0
    MONITOR_BAUDRATE = 115200
    AVRDUDE_OPTS = -v
    ARDUINO_SKETCHBOOK = $(realpath .)/lib
    USER_LIB_PATH = $(realpath .)/lib
    include /usr/share/arduino/Arduino.mk

What am I doing wrong?

sej7278, 2017-08-15:
you can't keep opening issues like this without doing some basic investigation yourself.

you know that arduino 1.8 needs a ```BOARD_SUB```, maybe you should just use arduino 1.0.5?

```
$ ARDUINO_QUIET=1  make show_submenu|grep nano
nano       atmega168       ATmega168
nano       atmega328       ATmega328
```

chrisspen, 2017-08-15:
Sorry, thanks for the clarification. A lot of these questions would be clarified if these subtle commands were explicitly mentioned in the README. I did some additional Googling after I posted this, and I also found the answer...not in any formal documentation but in a issue asking almost the same question.

Again, sorry for the noise.

closed 2017-08-15 by chrisspen

sej7278, 2017-08-16:
no problem, if you think of things that could be documented better, we always welcome a PR

Neumi, 2018-01-23:
@chrisspen would you be so kind and share your answer? Thanks! :)

sej7278, 2018-01-23:
https://github.com/sudar/Arduino-Makefile/blob/master/examples/Blink/Makefile#L25