sudar/Arduino-Makefile #519: (Windows) Avoid having to specify ARDUINO_DIR as a relative path

birgersp, 2017-08-28:
Hi,

According to the docs, `ARDUINO_DIR` has to be a relative path when using Windows. This is quite unfortunate, in my opinion. Especially when dealing with collaborative projects, and/or developing the project(s) on multiple computers or installations.

- Why does the path have to be relative?
- Is there really **no** workaround?

tuna-f1sh, 2017-09-04:
Have you tried using non-relative paths? I made a symbolic link on the root of my drive then set these as defines. Works fine for me using cygwin. Not sure if something has been fixed since this advice.

```
export ARDUINO_DIR=C:/Arduino
export ARDMK_DIR=C:/Arduino-Makefile
```

birgersp, 2017-09-04:
Thanks for the tip, I'll try some more to get it working. Using relative paths to the directories isn't really an "advice", it's stated pretty clear in the README:

>> In Cygwin in Windows this path must be relative, not absolute (e.g. "../../arduino" and not "/c/cygwin/Arduino").

birgersp, 2017-09-04:
Non-relative paths works, thank you very much @tuna-f1sh !

Then I guess the statements in the README are false

sej7278, 2017-09-04:
do non-relative paths only work with a symlink?

tuna-f1sh, 2017-09-05:
Doesn't seem like it, no. One can't use the default Arduino install directory due to the space in "Program Files (x86)" (escape charactors don't work and all my tests it didn't resolve correctly) but all the other defines work for me as direct paths or symbolic links. It's worth noting that the path must be defined with **/** and not Windows **\\**, so it's a sort of hybrid!

sej7278, 2017-09-05:
@tuna-f1sh any chance you could do a PR to update the docs as you seem to understand what's involved in windows more than myself/sudar?

perhaps including limitations such as spaces and remove references to requiring symlinks and relative paths.

tuna-f1sh, 2017-09-05:
I've created https://github.com/sudar/Arduino-Makefile/pull/525

Would be good to get your input too @birgersp.

birgersp, 2017-09-06:
No objections, PR resolves what I asked for ++

Thanks!

closed 2017-09-26 by birgersp