sudar/Arduino-Makefile #521: Search for headers and source files in subdirectories other than utility

ladislas, 2017-08-28:
Currently, [`get_library_includes`](https://github.com/sudar/Arduino-Makefile/blob/master/Arduino.mk#L927) and [`get_library_files`](https://github.com/sudar/Arduino-Makefile/blob/master/Arduino.mk#L934) are only searching the the `utility` subfolder.

But it sometimes happen that directories have different names like [ChRt](https://github.com/greiman/ChRt/tree/master/src).

Adding the following lines fixes everything! 👍  And we could even remove the `utility` part as the wildcard takes care of that.

```diff
diff --git a/Arduino.mk b/Arduino.mk
index 50d60ee..78d1eac 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -926,14 +926,15 @@ rwildcard=$(foreach d,$(wildcard $1*),$(call rwildcard,$d/,$2) $(filter $(subst
 # Gets include flags for library
 get_library_includes = $(if $(and $(wildcard $(1)/src), $(wildcard $(1)/library.properties)), \
                            -I$(1)/src, \
-                           $(addprefix -I,$(1) $(wildcard $(1)/utility)))
+                           $(addprefix -I,$(1) $(wildcard $(1)/utility)) \
+                           $(addprefix -I,$(1) $(dir $(wildcard $(1)/*/))))

 # Gets all sources with given extension (param2) for library (path = param1)
 # for old (1.0.x) layout looks in . and "utility" directories
 # for new (1.5.x) layout looks in src and recursively its subdirectories
 get_library_files  = $(if $(and $(wildcard $(1)/src), $(wildcard $(1)/library.properties)), \
                         $(call rwildcard,$(1)/src/,*.$(2)), \
-                        $(wildcard $(1)/*.$(2) $(1)/utility/*.$(2)))
+                        $(wildcard $(1)/*.$(2) $(1)/*/*.$(2) $(1)/utility/*.$(2)))

 # General arguments
 USER_LIBS      := $(sort $(wildcard $(patsubst %,$(USER_LIB_PATH)/%,$(ARDUINO_LIBS))))```

sej7278, 2017-08-28:
yup, makes sense. PR?

ladislas, 2017-08-28:
sure, just a sec

ladislas, 2017-08-28:
#522 here you go