sudar/Arduino-Makefile #523: User and system flags

ladislas, 2017-09-01:
Hi there! --

As of now, it seems like setting `CFLAGS_STD` or `CXXFLAGS_STD` removes a lot of important flags.

This can be confusing for users who just might want to use `-std=gnu++14` but keep the other important flags as is.

I think we should have different variables defined:

- `CFLAGS_STD` & `CXXFLAGS_STD` must be only for `-std=xxx` and provide a sane default
- `CFLAGS_CORE` for `-flto -fno-fat-lto-objects -fdiagnostics-color`
- `CXXFLAGS_CORE` for `-fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color`
- `CFLAGS_USER` & `CXXFLAGS_USER` - for the user to add more flags and still keep the core ones

the `*_USER` flags can be kept empty and the `*_CORE` can also be set to empty if the user wants to provide only his own flags.

What do you think?

sej7278, 2017-09-02:
yes the *_STD variables got abused a bit recently in the rush to fix LTO and g++ issues, they should revert to just setting the standard e.g. gnu++11

ok so CFLAGS_CORE must not be user-modifiable.

if gcc >4.9 is used, then CFLAGS_CORE must be set with the LTO etc. flags.

if gcc <4.9 is used, then CFLAGS_CORE must be empty.

CFLAGS_USER is down to the user to know what they're doing.

the simpler way of doing it is probably to append them to CFLAGS and not bother with CFLAGS_CORE (or CFLAGS_USER as the user can simply append to CFLAGS too):

```diff
diff --git a/Arduino.mk b/Arduino.mk
index 50d60ee..77313c2 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -1045,7 +1045,7 @@ endif
 
 ifndef CFLAGS_STD
     ifeq ($(shell expr $(CC_VERNUM) '>' 490), 1)
-        CFLAGS_STD      = -std=gnu11 -flto -fno-fat-lto-objects -fdiagnostics-color
+        CFLAGS_STD      = -std=gnu11
     else
         CFLAGS_STD        =
     endif
@@ -1056,7 +1056,7 @@ endif
 
 ifndef CXXFLAGS_STD
     ifeq ($(shell expr $(CC_VERNUM) '>' 490), 1)
-        CXXFLAGS_STD      = -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color
+        CXXFLAGS_STD      = -std=gnu++11
     else
         CXXFLAGS_STD      =
     endif
@@ -1069,7 +1069,9 @@ CFLAGS        += $(CFLAGS_STD)
 CXXFLAGS      += -fpermissive -fno-exceptions $(CXXFLAGS_STD)
 ASFLAGS       += -x assembler-with-cpp
 ifeq ($(shell expr $(CC_VERNUM) '>' 490), 1)
-    ASFLAGS += -flto
+    ASFLAGS  += -flto
+    CXXFLAGS += -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color
+    CFLAGS   += -flto -fno-fat-lto-objects -fdiagnostics-color
 endif
 LDFLAGS       += -$(MCU_FLAG_NAME)=$(MCU) -Wl,--gc-sections -O$(OPTIMIZATION_LEVEL)
 ifeq ($(shell expr $(CC_VERNUM) '>' 490), 1)
```

ladislas, 2017-09-02:
> ok so CFLAGS_CORE must not be user-modifiable.

They should not be advertised as user modifiable. But they could be turned off for the user to provide his own flags.

sej7278, 2017-09-04:
can we close this now @ladislas - even though the merge didn't really do it how you suggested? ;-)

ladislas, 2017-09-04:
yes we can. Do you want me to make a PR with my suggestion?

sej7278, 2017-09-04:
no i meant that the PR we merged covers this - users can append their own variables to CFLAGS instead of your CFLAGS_USER, and CFLAGS_STD is cleansed now. CFLAGS_CORE isn't needed

ladislas, 2017-09-04:
true but there is no way for the moment to turn off the provided flags and only add the one you want.

sej7278, 2017-09-05:
oh yes, CFLAGS and CXXFLAGS are always appended to unlike CFLAGS_STD and CXXFLAGS_STD which can be user-overridden, so you can't disable LTO for example

ladislas, 2017-09-05:
yes, that's exactly why I posted in the first place. To change this behavior and have the option to not append to CFLAGS and CXXFLAGS but override them if needed.

I think it's something that could be handy for power users and absolutely transparent for the regular users.

ladislas, 2017-09-07:
@sej7278 I've done some testing with your PR.

We need to add the `-fdiagnostics-color` flag to `LDFLAGS` as well.

sej7278, 2017-09-07:
we didn't before, and its not something the PR removed, but it can be added sure for consistency.

ladislas, 2017-09-07:
that's interesting -- I used to have the output in colors without setting any flags and after upgrading to the latest commit it went missing. I'll investigate a bit more.
I also have new warnings I did not have before...

sej7278, 2017-09-07:
depends which PR you're looking at - d1156e8fdf9c49cfd0dace56b8a467b7bbbe798c added color to CXXFLAGS_STD, fa82c3a9dbad998fd391dc9c2c4ca913744cc26d moved it to CXXFLAGS, but we never had it set on LDFLAGS.

perhaps its on by default in the gcc you're using? in all cases we check for avr-gcc 4.9+ (i know you're using 7.2 or something bleeding edge)

Astro-Johnny, 2019-01-23:
Hello! Sorry for the resurrecting an old thread, but this appears to be somewhat on-topic. Is there an "official way" of getting rid of the `-fpermissive` from CXXFLAGS?

For the record, my current approach that works is adding:
```Makefile
CXXFLAGS := $(filter-out -fpermissive,$(CXXFLAGS))
```
**after** the include of `Arduino.mk` (also, the `:=` operator is mandatory to avoid recursion).