sudar/Arduino-Makefile #526: Diferent behaviour of arduino nano using arduino-Makefile and Windows Arduino IDE

juampe, 2017-09-05:
First I had to say, that Ardunio-Makefile is a great tool for my development workflow. Thanks to all, to made this possible.
Second I trying to explain the issue as far as i can. I think that the AVR behaviour is diferent, so the compilation and upload made by Windows IDE(arduino-1.8.3) and Arduino-Makefile (arduino-1.8.3,arduino-1.8.4,both).

Escenery : some simple ADS1115 with I2C wiring and 3.3v test from nano

[image]

The sketch involved
`#include <Wire.h>
#include <Adafruit_ADS1015.h>

Adafruit_ADS1115 ads;

void setup(void)
{
  Serial.begin(115200);
  ads.setGain(GAIN_ONE);       // +/- 4.096V  1 bit = 0.125mV
  ads.begin();
}

void loop(void)
{
  float adc0, adc1, adc2, adc3,adc01;

  adc0 = ads.readADC_SingleEnded(0)*0.125F/1000;
  adc1 = ads.readADC_SingleEnded(1)*0.125F/1000;
  adc2 = ads.readADC_SingleEnded(2)*0.125F/1000;
  adc3 = ads.readADC_SingleEnded(3)*0.125F/1000;
  Serial.print("AIN0: "); Serial.println(adc0);
  Serial.print("AIN1: "); Serial.println(adc1);
  Serial.print("AIN2: "); Serial.println(adc2);
  Serial.print("AIN3: "); Serial.println(adc3);
  adc01=ads.readADC_Differential_0_1()*0.125F/1000;
  Serial.print("DIFF01: "); Serial.println(adc01);
  Serial.println(" ");

  delay(1000);
}
`

In Windows flavour 
The result works as expected ADC1115 and the I2C revelals coherent data  0 0 0 3.3 0
`AIN0: 0.80
AIN1: 0.33
AIN2: 0.65
AIN3: 3.22
DIFF01: -0.00
`

In Rpi flavour Ardunio-Makefile the output is full of repeated ilegal values (for ads1115 posibilities) and does not fit the reality with the 3.3 pin in A3
`AIN0: 8.19
AIN1: 8.19
AIN2: 8.19
AIN3: 8.19
DIFF01: 0.51

AIN0: 0.00
AIN1: 0.00
AIN2: 0.00
AIN3: 0.00
DIFF01: 0.56
`
The Makefile involved
`root@kelsier:~/arduino/ads1115# cat Makefile
ARDUINO_DIR = /opt/arduino-1.8.4
USER_LIB_PATH = /root/arduino/libraries
ARDUINO_LIBS = Wire Adafruit_ADS1015
BOARD_TAG = nano
BOARD_SUB = atmega328
MONITOR_PORT  = /dev/ttyUSB0

include /opt/Arduino-Makefile/Arduino.mk
`
Some ouput from make upload
`root@kelsier:~/arduino/ads1115# make upload
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX
- [USER]               ARDUINO_DIR = /opt/arduino-1.8.4
- [COMPUTED]           ARDMK_DIR = /opt/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 184
- [DEFAULT]            ARCHITECTURE = avr
- [DEFAULT]            ARDMK_VENDOR = arduino
- [DEFAULT]            ARDUINO_SKETCHBOOK =
- [BUNDLED]            AVR_TOOLS_DIR = /opt/arduino-1.8.4/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /opt/arduino-1.8.4/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /opt/arduino-1.8.4/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /opt/arduino-1.8.4/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /opt/arduino-1.8.4/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [USER]               USER_LIB_PATH = /root/arduino/libraries
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_SUB = atmega328
- [USER]               BOARD_TAG = nano
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = eightanaloginputs (from build.variant)
- [COMPUTED]           OBJDIR = build-nano-atmega328 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [DETECTED]           MONITOR_BAUDRATE = 115200  (in sketch)
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [DEFAULT]            CFLAGS_STD = -std=gnu11
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11
- [COMPUTED]           DEVICE_PATH = /dev/ttyUSB0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 Adafruit_ADS1015
- [PLATFORM]             Wire
- [COMPUTED]           BOOTLOADER_PARENT = /opt/arduino-1.8.4/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
-------------------------
mkdir -p build-nano-atmega328
/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=184 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/opt/arduino-1.8.4/hardware/arduino/avr/cores/arduino -I/opt/arduino-1.8.4/hardware/arduino/avr/variants/eightanaloginputs    -I/opt/arduino-1.8.4/hardware/arduino/avr/libraries/Wire/src   -I/root/arduino/libraries/Adafruit_ADS1015 -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color ads1115.ino -o build-nano-atmega328/ads1115.ino.o
/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-gcc -mmcu=atmega328p -Wl,--gc-sections -Os -flto -fuse-linker-plugin -o build-nano-atmega328/ads1115.elf build-nano-atmega328/ads1115.ino.o build-nano-atmega328/libcore.a   -lc -lm
/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom build-nano-atmega328/ads1115.elf build-nano-atmega328/ads1115.hex

/opt/arduino-1.8.4/hardware/tools/avr/bin/avr-size --mcu=atmega328p -C --format=avr build-nano-atmega328/ads1115.elf
AVR Memory Usage
----------------
Device: atmega328p

Program:    5774 bytes (17.6% Full)
(.text + .data + .bootloader)

Data:        451 bytes (22.0% Full)
(.data + .bss + .noinit)

make reset
make[1]: Entering directory '/root/arduino/ads1115'
/opt/Arduino-Makefile/bin/ard-reset-arduino  /dev/ttyUSB0
make[1]: Leaving directory '/root/arduino/ads1115'
make do_upload
make[1]: Entering directory '/root/arduino/ads1115'
/opt/arduino-1.8.4/hardware/tools/avr/bin/avrdude -q -V -p atmega328p -C /opt/arduino-1.8.4/hardware/tools/avr/etc/avrdude.conf -D -c arduino -b 57600 -P /dev/ttyUSB0 \
                -U flash:w:build-nano-atmega328/ads1115.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "build-nano-atmega328/ads1115.hex"
avrdude: writing flash (5774 bytes):
avrdude: 5774 bytes of flash written

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

make[1]: Leaving directory '/root/arduino/ads1115'
`
Third and last, I think is posible I forgot a "Anduino Nano Makefile rare flag" to allow a better AVR firmware, but in don't found it anyway.
Any ideas to workaround this?

Thanks.

sej7278, 2017-09-05:
no idea, if it compiles and uploads its usually fine and not a makefile issue.

have you tried the IDE on the raspberry pi (is that even possible?) to rule out something like the library containing windows line endings or only being tested on windows (not the first time i've seen that from adafruit)?

we probably need the full output from the ide and the full output from the makefile to see if there's any new compiler flags they've silently added lately.

i doubt it'd make any difference, but i really don't recommend running make as root. you might want to define ARDUINO_SKETCHBOOK too (usually read from the ide config, but if you haven't run it.....)

juampe, 2017-09-06:
Many Thanks,
Good  ideas!,

I can discard the avrdude/upload part I got the hex from Windows I flashed it to Nano and It works as expected
`/opt/arduino-1.8.4/hardware/tools/avr/bin/avrdude -q -V -p atmega328p -C /opt/arduino-1.8.4/hardware/tools/avr/etc/avrdude.conf -D -c arduino -b 57600 -P /dev/ttyUSB0 -U flash:w:arduino_build_75435/ads1115.ino.hex:i`

I ran the arduino IDE in Rpi , and... It works great too.
```
root@kelsier:/opt/arduino-1.8.3# ./arduino
Picked up JAVA_TOOL_OPTIONS:
/opt/arduino-1.8.3/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.3/hardware -tools /opt/arduino-1.8.3/tools-builder -tools /opt/arduino-1.8.3/hardware/tools/avr -built-in-libraries /opt/arduino-1.8.3/libraries -libraries /root/arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10803 -build-path /tmp/arduino_build_16526 -warnings=none -build-cache /tmp/arduino_cache_542615 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/opt/arduino-1.8.3/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/opt/arduino-1.8.3/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/opt/arduino-1.8.3/hardware/tools/avr -verbose /root/arduino/ads1115/ads1115.ino
/opt/arduino-1.8.3/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.3/hardware -tools /opt/arduino-1.8.3/tools-builder -tools /opt/arduino-1.8.3/hardware/tools/avr -built-in-libraries /opt/arduino-1.8.3/libraries -libraries /root/arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10803 -build-path /tmp/arduino_build_16526 -warnings=none -build-cache /tmp/arduino_cache_542615 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/opt/arduino-1.8.3/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/opt/arduino-1.8.3/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/opt/arduino-1.8.3/hardware/tools/avr -verbose /root/arduino/ads1115/ads1115.ino
Using board 'nano' from platform in folder: /opt/arduino-1.8.3/hardware/arduino/avr
Using core 'arduino' from platform in folder: /opt/arduino-1.8.3/hardware/arduino/avr
ATENCIÓN:  .mystools Espúrias carpetas en librería 'MySensors'
Detecting libraries used...
"/opt/arduino-1.8.3/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/opt/arduino-1.8.3/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.3/hardware/arduino/avr/variants/eightanaloginputs" "/tmp/arduino_build_16526/sketch/ads1115.ino.cpp" -o "/dev/null"
"/opt/arduino-1.8.3/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/opt/arduino-1.8.3/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.3/hardware/arduino/avr/variants/eightanaloginputs" "-I/opt/arduino-1.8.3/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_16526/sketch/ads1115.ino.cpp" -o "/dev/null"
"/opt/arduino-1.8.3/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/opt/arduino-1.8.3/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.3/hardware/arduino/avr/variants/eightanaloginputs" "-I/opt/arduino-1.8.3/hardware/arduino/avr/libraries/Wire/src" "-I/root/arduino/libraries/Adafruit_ADS1015" "/tmp/arduino_build_16526/sketch/ads1115.ino.cpp" -o "/dev/null"
"/opt/arduino-1.8.3/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/opt/arduino-1.8.3/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.3/hardware/arduino/avr/variants/eightanaloginputs" "-I/opt/arduino-1.8.3/hardware/arduino/avr/libraries/Wire/src" "-I/root/arduino/libraries/Adafruit_ADS1015" "/opt/arduino-1.8.3/hardware/arduino/avr/libraries/Wire/src/Wire.cpp" -o "/dev/null"
"/opt/arduino-1.8.3/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/opt/arduino-1.8.3/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.3/hardware/arduino/avr/variants/eightanaloginputs" "-I/opt/arduino-1.8.3/hardware/arduino/avr/libraries/Wire/src" "-I/root/arduino/libraries/Adafruit_ADS1015" "/opt/arduino-1.8.3/hardware/arduino/avr/libraries/Wire/src/utility/twi.c" -o "/dev/null"
"/opt/arduino-1.8.3/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/opt/arduino-1.8.3/hardware/arduino/avr/cores/arduino" "-I/opt/arduino-1.8.3/hardware/arduino/avr/variants/eightanaloginputs" "-I/opt/arduino-1.8.3/hardware/arduino/avr/libraries/Wire/src" "-I/root/arduino/libraries/Adafr [... cut ...]
                -U flash:w:build-nano-atmega328/ads1115.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "build-nano-atmega328/ads1115.hex"
avrdude: writing flash (5774 bytes):
avrdude: 5774 bytes of flash written

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

make[1]: Leaving directory '/root/arduino/ads1115'

```

Sorry for the inconvenience, and many, many thanks.

sej7278, 2017-09-06:
no problem, pretty bizarre but i'm glad you got it working.

code size is the same, might have been something in your environment or stale temp files i guess.

can you close the issue if you're happy please?

juampe, 2017-09-06:
Now I see some extra AUTODETECTED values, I supose due to IDE config file preferences.txt

- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /root/.arduino15/preferences.txt
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /root/arduino (from arduino preferences file)

I close the issue , thanks again.

closed 2017-09-06 by juampe