sudar/Arduino-Makefile #527: avr-gcc command not found

concavegit, 2017-09-24:
It seems that avr-gcc is looked for in $AVR_TOOLS_DIR/bin/avr-gcc. However, in my system, avr-gcc is in $AVR_TOOLS_DIR/4.9.4/avr-gcc, so `make` fails. I could make a sym link, however I want to see if there is a cleaner solution. In case I have made a misdiagnosis, here is the `make` output from the blink example:

```
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [AUTODETECTED]       ARDUINO_DIR = /usr/share/arduino 
- [COMPUTED]           ARDMK_DIR = /usr/share/arduino (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 105 
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/concaveusr/.arduino/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/concaveusr/sketchbook (from arduino preferences file)
- [AUTODETECTED]       AVR_TOOLS_DIR = /usr/x86_64-pc-linux-gnu/avr/gcc-bin (found in $PATH)
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino//variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino//boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/concaveusr/sketchbook/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino//cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
/bin/sh: /usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin/avr-gcc: No such file or directory
expr: syntax error
/bin/sh: /usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin/avr-gcc: No such file or directory
expr: syntax error
- [DEFAULT]            CFLAGS_STD =  
/bin/sh: /usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin/avr-gcc: No such file or directory
expr: syntax error
- [DEFAULT]            CXXFLAGS_STD =  
/bin/sh: /usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin/avr-gcc: No such file or directory
expr: syntax error
/bin/sh: /usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin/avr-gcc: No such file or directory
expr: syntax error
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
make: /usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin/avr-size: Command not found
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino//bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
make: /usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin/avr-gcc: Command not found
- [COMPUTED]           CC_VERSION =  (avr-gcc)
-------------------------
mkdir -p build-uno
/usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/usr/share/arduino/hardware/arduino//variants/standard    -Wall -ffunction-sections -fdata-sections -Os  Blink.c -o build-uno/Blink.c.o
make: /usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin/avr-gcc: Command not found
make: *** [/usr/share/arduino/Arduino.mk:1230: build-uno/Blink.c.o] Error 127
```

and the output of `which avr-gcc:`
```
/usr/x86_64-pc-linux-gnu/avr/gcc-bin/4.9.4/avr-gcc
```

sej7278, 2017-09-25:
its because you're not using the "standard" directory structure (what is that, LFS or gentoo or something?)

so the path to avr-gcc is detected from "which" as /usr/x86_64-pc-linux-gnu/avr/gcc-bin/4.9.4/avr-gcc

the parent directory of that is then fed into AVR_TOOLS_DIR, and /bin is appended to make AVR_TOOLS_PATH - essentially /usr/x86_64-pc-linux-gnu/avr/gcc-bin/4.9.4/../bin which becomes /usr/x86_64-pc-linux-gnu/avr/gcc-bin/bin

its just unfortunate that you have "4.9.4" instead of "bin" that the makefile is expecting at the same level, a symlink is the easy fix, otherwise you'd have to manually define AVR_TOOLS_DIR and AVR_TOOLS_PATH and i'm not sure if that would break detection of avrdude too.

concavegit, 2017-09-25:
Gentoo, nailed it. It seems like sym linking will be the best general solution. Thanks.

closed 2017-09-25 by concavegit

sej7278, 2017-09-25:
yeah or maybe fix the gentoo ebuild ;-)