sudar/Arduino-Makefile #534: Makefile from esp8266

renat2985, 2017-10-13:
Help me please configure Makefile file for ESP8266.
Is this possible?
[image]

sej7278, 2017-10-13:
arduino-mk doesn't support the esp8266 but i use this makefile which supports flashing normally or via ota, serial debugging etc; and has similar recipes to arduino-mk:

```make
ARDUINO_PATH = $(HOME)/arduino-1.8.5
SKETCHBOOK = $(HOME)/arduino16
ESPTOOL = $(SKETCHBOOK)/hardware/esp8266com/esp8266/tools/esptool/esptool
ESPOTA = $(SKETCHBOOK)/hardware/esp8266com/esp8266/tools/espota.py
SKETCH = $(notdir $(CURDIR)).ino
TARGET_DIR = $(CURDIR)/build-esp8266
MONITOR_PORT = /dev/ttyUSB0
ESP_IP = 192.168.1.3
OTA_SERVER = 192.168.1.2
OTA_PASSWD = aplaintextpassword
#DEBUG = ,Debug=Serial,DebugLevel=all_____
DEBUG = 

all:
	@ mkdir -p $(TARGET_DIR)

	$(ARDUINO_PATH)/arduino-builder -compile -logger=machine \
	-hardware "$(ARDUINO_PATH)/hardware" \
	-hardware "$(SKETCHBOOK)/hardware" \
	-tools "$(ARDUINO_PATH)/tools-builder" \
	-tools "$(ARDUINO_PATH)/hardware/tools/avr" \
	-built-in-libraries "$(ARDUINO_PATH)/libraries" \
	-libraries "$(SKETCHBOOK)/libraries" \
	-fqbn=esp8266com:esp8266:generic:CpuFrequency=80,CrystalFreq=26,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M3M,ResetMethod=nodemcu$(DEBUG) \
	-ide-version=10805 \
	-build-path "$(TARGET_DIR)" \
	-warnings=none \
	-prefs=build.warn_data_percentage=75 \
	-verbose "$(SKETCH)"

upload:
	$(ESPTOOL) -v -cd nodemcu -cb 115200 -cp $(MONITOR_PORT) -ca 0x00000 -cf $(TARGET_DIR)/$(SKETCH).bin

ota:
	$(ESPOTA) -d -r -i $(ESP_IP) -I $(OTA_SERVER) -p 8266 -P 8266 -a $(OTA_PASSWD) -f $(TARGET_DIR)/$(SKETCH).bin

clean:
	rm -rf $(TARGET_DIR)

monitor:
	screen $(MONITOR_PORT) 115200
```

closed 2018-06-27 by sej7278

scottchiefbaker, 2019-01-11:
@sej7278 is the above script maintained anywhere? I'd like to get involved in updating and helping grow this. Right now I just copy and paste from this comment and that's not ideal :)

sej7278, 2019-01-11:
its not really complex enough to put on github really, only changes i ever make are the debug flags when they change in the core and the version when there's a new ide. you're free to do so if you like.

there is [this](https://github.com/plerup/makeEspArduino) which the arduino core mentions, but i find it overly complicated, its a bit like the one from the espressif idk.

here's my one for esp32 which builds fine, but i don't know if uploading works as i don't have one to try (and not sure it even uses esptool?):

```make
ARDUINO_PATH = $(HOME)/arduino-1.8.8
SKETCHBOOK = $(HOME)/arduino16
ESPTOOL = $(SKETCHBOOK)/hardware/espressif/esp32/tools/esptool/esptool.py
SKETCH = $(notdir $(CURDIR)).ino
TARGET_DIR = $(CURDIR)/build-esp32
MONITOR_PORT = /dev/ttyUSB0

all:
	@ mkdir -p $(TARGET_DIR)

	$(ARDUINO_PATH)/arduino-builder -compile -logger=machine \
	-hardware "$(ARDUINO_PATH)/hardware" \
	-hardware "$(SKETCHBOOK)/hardware" \
	-tools "$(ARDUINO_PATH)/tools-builder" \
	-tools "$(ARDUINO_PATH)/hardware/tools/avr" \
	-built-in-libraries "$(ARDUINO_PATH)/libraries" \
	-libraries "$(SKETCHBOOK)/libraries" \
	-fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none \
	-ide-version=10808 \
	-build-path "$(TARGET_DIR)" \
	-warnings=none \
	-prefs=build.warn_data_percentage=75 \
	-verbose "$(SKETCH)"

upload:
	$(ESPTOOL) -v -cd nodemcu -cb 115200 -cp $(MONITOR_PORT) -ca 0x00000 -cf $(TARGET_DIR)/$(SKETCH).bin

clean:
	rm -rf $(TARGET_DIR)

monitor:
	screen $(MONITOR_PORT) 115200
```