sudar/Arduino-Makefile #535: Problem building for adafruit Feather 32u4

Badane, 2017-10-16:
Hi, 
I'm trying to move my usual flashing routine from Arduino IDE to Makefile. 
I use a feather32u4 with embedded Lora module. 
For the moment I'm just trying to upload a simple blink code. 

To build I tried to use the ALTERNATE_CORE way. I got the Adafruit-1.4.9-avr.
I put it in my sketch_folder/hardware/avr. 

When building I can see that everything looks have correct settings : 
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [USER]               ARDUINO_DIR = /Applications/Arduino.app/Contents/Java 
- [COMPUTED]           ARDMK_DIR = /usr/local/Cellar/arduino-mk/1.5.2 (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 184 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /Users/dann/Library/Arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /Users/dann/Documents/Arduino (from arduino preferences file)
- [BUNDLED]            AVR_TOOLS_DIR = /Applications/Arduino.app/Contents/Java/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /Applications/Arduino.app/Contents/Java/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [USER]               ALTERNATE_CORE = adafruit 
- [COMPUTED]           ALTERNATE_CORE_PATH = /Users/dann/Documents/Arduino/hardware/adafruit/avr  (from ARDUINO_SKETCHBOOK and ALTERNATE_CORE)
- [COMPUTED]           ARDUINO_VAR_PATH = /Users/dann/Documents/Arduino/hardware/adafruit/avr/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = /Users/dann/Documents/Arduino/hardware/adafruit/avr/boards.txt (from ALTERNATE_CORE_PATH)
- [DEFAULT]            USER_LIB_PATH = /Users/dann/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = atmega32u4 
- [USER]               BOARD_TAG = Adafruit Feather 32u4 
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-Adafruit Feather 32u4-atmega32u4 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH = /dev/tty.usbmodem14341 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
 
But I'm stuck when error "make: *** No rule to make target `build-Adafruit', needed by `Feather'.  Stop." pops... 
Just starting learning Makefiles. 
Any help please ? 
Thanks :)

sej7278, 2017-10-16:
i'll have to look into the new core, the one i have doesn't even list the feather, but this works:

```make
ARDUINO_DIR = $(HOME)/arduino-1.8.5
ALTERNATE_CORE = adafruit
BOARD_TAG = trinket3
include /usr/share/arduino/Arduino.mk
```

your problem is the spaces in the board tag, don't tell me they've actually done that?!:

```
[USER] BOARD_TAG = Adafruit Feather 32u4
[COMPUTED] OBJDIR = build-Adafruit Feather 32u4-atmega32u4 (from BOARD_TAG)
```

Badane, 2017-10-16:
Yeeah ! Works fine 
Thank you :D 
Let's implement my radio libs now

closed 2017-10-16 by Badane